aaronwmorris/indi-allsky #593: RPi4 8GB, Raspi OS Bullseye 64bit, RPi HQ Camera - issues - indi-allsky keeps stopping by itself as well as when reloading config from system->services. Also focus mode not working

baryah, 2023-02-27:
I happened to be using Thomas Jacquin Allsky from last couple of years and recently gave the indi-allsky a try. I have had following issues with my configuration -

1. I had to choose no GPS during installation to be able to set the latitude and longitude manually (i don't have any GPS module on my RPi).
2. Most of the times when I make changes to the configuration from the webpage and choose to Reload on save, the indi-allsky service crashes.
3. When I enable the Focus mode in the configurations, I constantly get 'No image for 5 minutes' and no image is shot.
4. The indi-allsky service keeps on stopping by itself randomly.

During installation I chose the libcamera and the imx447.

[indi-allsky.log](https://github.com/aaronwmorris/indi-allsky/files/10842724/indi-allsky.log)

aaronwmorris, 2023-02-27:
> I had to choose no GPS during installation to be able to set the latitude and longitude manually (i don't have any GPS module on my RPi).

This is expected.  Do not select GPS support if you do not have a GPS adapter.

> Most of the times when I make changes to the configuration from the webpage and choose to Reload on save, the indi-allsky service crashes.

Investigating...

> When I enable the Focus mode in the configurations, I constantly get 'No image for 5 minutes' and no image is shot.

Investigating...

> The indi-allsky service keeps on stopping by itself randomly.

I see the sqlite database is being locked, which does not make any sense.  How recently did you pull down this code?

> Another Issue I am having is that the web interface shows allsky service to be 'DOWN' while the service is running; even the system->services page shows it to be active.

This could be a timing issue.  It could take 1-2 minutes before the DOWN message show RUNNING, but I am investigating.

aaronwmorris, 2023-02-27:
Regarding focus mode.  If focus mode is enabled, due to a recent change, it would only show the image in the Focus Mode view.  I merged #595 to show the image on the Latest Image view even when focus mode is enabled.

aaronwmorris, 2023-02-27:
Regarding the crashes due to the database being locked, it is acting like the database journal is not in WAL mode.  Can you show me the output of the following?

```
sqlite3 /var/lib/indi-allsky/indi-allsky.sqlite "PRAGMA journal_mode;"
```

baryah, 2023-02-28:
[image]

aaronwmorris, 2023-02-28:
Issues 2, 3, and 5 (possibly 4) all point to issues with the database.  With the database journal in WAL mode, it should not be locking since this allows multiple readers and writers.

In the past, I have experienced the database locking issue which is why the WAL journal is utilized.

Is this a new SDcard you are using?  I do not like to cavalierly blame the card, but bad storage is the only thing makes sense to me.

baryah, 2023-03-01:
I ll change the card and see how it behaves

baryah, 2023-03-01:
> Issues 2, 3, and 5 (possibly 4) all point to issues with the database. With the database journal in WAL mode, it should not be locking since this allows multiple readers and writers.
>
> In the past, I [...]

Hard to believe, but changing the SD card actually did the trick. Everything is working fine now.

baryah, 2023-03-02:
Still after running perfectly for sometime, the status becomes 'Down' while the services are active.

[indi-allsky (2).log](https://github.com/aaronwmorris/indi-allsky/files/10874175/indi-allsky.2.log)

baryah, 2023-03-02:
[indi-allsky.log](https://github.com/aaronwmorris/indi-allsky/files/10874300/indi-allsky.log)

aaronwmorris, 2023-03-02:
The database is experiencing locks for some reason.

Every 15 seconds or indi-allsky updates the timestamp of the WATCHDOG state in the database.  If that timestamp exceeds a certain threshold, it is assumed indi-allsky is not running.

```
Mar  2 23:13:41 indi-allsky [ERROR] MainProcess allsky.unhandled_exception() #34: An uncaught exception occurred:
Mar  2 23:13:41 indi-allsky [ERROR] MainProcess allsky.unhandled_exception() #35: Type: <class 'sqlalchemy.exc.OperationalError'>
Mar  2 23:13:41 indi-allsky [ERROR] MainProcess allsky.unhandled_exception() #36: Value: (sqlite3.OperationalError) database is locked#012[SQL: UPDATE state SET "createDate"=?, value=? WHERE state."key" = ?]#012[parameters: ('2023-03-02 23:13:30.760751', '1677779010', 'WATCHDOG')]#012(Background on this error at: https://sqlalche.me/e/20/e3q8)
```

The bizarre thing is that this only seems to be affecting updating the WATCHDOG database change.  There are other database transactions still running all the time.  The only thing this would really affect is the Status message, it does not affect any other functions.

You can try to flip over to mysql/mariadb if you want.  https://github.com/aaronwmorris/indi-allsky/wiki/MySQL-MariaDB-Information

aaronwmorris, 2023-03-02:
Addendum:  Something is still suspect with the sdcard.  You may want to see if the following wiki page helps to reduce disk IO on your system:

https://github.com/aaronwmorris/indi-allsky/wiki/Disk-Optimizations

aaronwmorris, 2023-03-04:
In light of the report of database locking in #502 , I guess there is an underlying issue here.  I think its journald causing the disk I/O to slow down, but I think I found a solution to that.

Merged #601 to extend the I/O timeout for sqlite.

baryah, 2023-03-04:
Thanks,
Will update

> On 04-Mar-2023, at 10:25 PM, Aaron W Morris ***@***.***> wrote:
> 
> Merged #601 <https://github.com/aaronwmorris/indi-allsky/pull/601> to extend the I/O timeout for sqlite.
> 
>

baryah, 2023-03-05:
> Addendum: Something is still suspect with the sdcard. You may want to see if the following wiki page helps to reduce disk IO on your system:
> 
> https://github.com/aaronwmorris/indi-allsky/wiki/Disk-Optimizations

Does these settings plus the latest update - Its been one whole night without a single database locking ..

baryah, 2023-03-05:
I have had no database locking since yesterday night.
Today I tried to generate the darks, I got - 

(indi-allsky) pi@indi-allsky:~/indi-allsky $ ./darks.py sigmaclip
2023-03-05 14:15:34,118 [INFO] MainProcess indi.__init__() #135: creating an instance of IndiClient
2023-03-05 14:15:34,119 [INFO] MainProcess indi.__init__() #142: PyIndi version: 1.9.7
2023-03-05 14:15:34,120 [INFO] MainProcess darks._initialize() #182: Connecting to indiserver
INDI::BaseClient::connectServer: creating new connection...
INDI::BaseClient::connectServer: Already connected.
2023-03-05 14:15:34,120 [INFO] MainProcess indi.serverConnected() #328: Server connected (localhost:7624)
2023-03-05 14:15:34,126 [INFO] MainProcess indi.newDevice() #198: new device CCD Simulator
2023-03-05 14:15:34,136 [INFO] MainProcess indi.newDevice() #198: new device Telescope Simulator
2023-03-05 14:15:42,129 [WARNING] MainProcess darks._initialize() #205: Connecting to device libcamera_imx477
INDI::BaseClient: Error. Unable to find driver libcamera_imx477
2023-03-05 14:15:42,190 [INFO] MainProcess miscDb.addCamera() #71: Camera DB ID: 2
2023-03-05 14:15:42,192 [INFO] MainProcess indi.updateCcdBlobMode() #390: Set BLOB mode
2023-03-05 14:15:42,192 [ERROR] MainProcess darks._initialize() #254: CCD night gain above maximum, changing to 16
2023-03-05 14:15:45,196 [ERROR] MainProcess darks._initialize() #263: CCD moon mode gain above maximum, changing to 16
2023-03-05 14:15:48,199 [ERROR] MainProcess darks._initialize() #268: CCD day gain below minimum, changing to 1
2023-03-05 14:15:51,203 [WARNING] MainProcess darks._run() #572: ****** IF THE CCD COOLER WAS ENABLED, YOU MAY CONSIDER STOPPING THIS UNTIL THE SENSOR HAS WARMED ******
2023-03-05 14:15:59,214 [INFO] MainProcess darks._take_exposures() #648: Temp folder: /tmp/tmpv7iack15
2023-03-05 14:15:59,215 [INFO] MainProcess darks.shoot() #278: Taking 60.00000000 s exposure (gain 1)
2023-03-05 14:15:59,217 [INFO] MainProcess libcamera.setCcdExposure() #181: image command: libcamera-still --immediate --nopreview --raw --denoise off --awbgains 1,1 --gain 1 --shutter 60000000 --metadata /tmp/tmpxprjygvm.json --metadata-format json --output /tmp/tmpim1laxxc.dng
2023-03-05 14:17:00,655 [INFO] MainProcess darks._take_exposures() #664: Exposure received in 61.4402 s
2023-03-05 14:17:00,995 [INFO] MainProcess darks._take_exposures() #680: Image average adu: 257.88
2023-03-05 14:17:00,996 [INFO] MainProcess darks.shoot() #278: Taking 60.00000000 s exposure (gain 1)
2023-03-05 14:17:00,997 [INFO] MainProcess libcamera.setCcdExposure() #181: image command: libcamera-still --immediate --nopreview --raw --denoise off --awbgains 1,1 --gain 1 --shutter 60000000 --metadata /tmp/tmpfawidsq1.json --metadata-format json --output /tmp/tmp16wrbm5j.dng
2023-03-05 14:18:02,376 [INFO] MainProcess darks._take_exposures() #664: Exposure received in 61.3799 s
2023-03-05 14:18:02,724 [INFO] MainProcess darks._take_exposures() #680: Image average adu: 256.96
2023-03-05 14:18:02,724 [INFO] MainProcess darks.shoot() #278: Taking 60.00000000 s exposure (gain 1)
2023-03-05 14:18:02,725 [INFO] MainProcess libcamera.setCcdExposure() #181: image command: libcamera-still --immediate --nopreview --raw --denoise off --awbgains 1,1 --gain 1 --shutter 60000000 --metadata /tmp/tmpjmwa8902.json --metadata-format json --output /tmp/tmp__clhl51.dng
2023-03-05 14:19:04,094 [INFO] MainProcess darks._take_exposures() #664: Exposure received in 61.3697 s
2023-03-05 14:19:04,438 [INFO] MainProcess darks._take_exposures() #680: Image average adu: 256.92
2023-03-05 14:19:04,439 [INFO] MainProcess darks.shoot() #278: Taking 60.00000000 s exposure (gain 1)
2023-03-05 14:19:04,440 [INFO] MainProcess libcamera.setCcdExposure() #181: image command: libcamera-still --immediate --nopreview --raw --denoise off --awbgains 1,1 --gain 1 --shutter 60000000 --metadata /tmp/tmpbq74wsez.json --metadata-format json --output /tmp/tmpd38g30aq.dng
2023-03-05 14:20:05,809 [INFO] MainProcess darks._take_exposures() #664: Exposure received in 61.3703 s
2023-03-05 14:20:06,128 [INFO] MainProcess darks._take_exposures() #680: Image average adu: 257.91
2023-03-05 14:20:06,128 [INFO] MainProcess darks.shoot() #278: Taking 60.00000000 s exposure (gain 1)
2023-03-05 14:20:06,129 [INFO] MainProcess libcamera.setCcdExposure() #181: image command: libcamera-still --immediate --nopreview --raw --denoise off --awbgains 1,1 --gain 1 --shutter 60000000 --metadata /tmp/tmpn0zen3u2.json --metadata-format json --output /tmp/tmpsssjrdl1.dng
2023-03-05 14:21:07,505 [INFO] MainProcess darks._take_exposures() #664: Exposure received in 61.3769 s
2023-03-05 14:21:07,827 [INFO] MainProcess darks._take_exposures() #680: Image average adu: 257.94
2023-03-05 14:21:07,828 [INFO] MainProcess darks.shoot() #278: Taking 60.00000000 s exposure (gain 1)
2023-03-05 14:21:07,829 [INFO] MainProcess libcamera.setCcdExposure() #181: image command: libcamera-still --immediate --nopreview --raw --denoise off --awbgains 1,1 --gain 1 --shutter 60000000 --metadata /tmp/tmplvn3i0aa.json --metadata-format json --output /tmp/tmp_6llz7c5.dng
2023-03-05 14:22:09,247 [INFO] MainProcess darks._take_exposures() #664: Exposure received in 61.4191 s
2023-03-05 14:22:09,597 [INFO] MainProcess darks._take_exposures() #680: Image average adu: 256.93
2023-03-05 14:22:09,597 [INFO] MainProcess darks.shoot() #278: Taking 60.00000000 s exposure (gain 1)
[... cut ...]
  File "/home/pi/indi-allsky/indi_allsky/darks.py", line 330, in _wait_for_image
    raw = rawpy.imread(str(filename_p))
  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/rawpy/__init__.py", line 20, in imread
    d.open_file(pathOrFile)
  File "rawpy/_rawpy.pyx", line 409, in rawpy._rawpy.RawPy.open_file
  File "rawpy/_rawpy.pyx", line 936, in rawpy._rawpy.RawPy.handle_error
rawpy._rawpy.LibRawIOError: b'Input/output error'
BaseClient::~BaseClient: Probability of detecting a deadlock.

baryah, 2023-03-05:
> > Addendum: Something is still suspect with the sdcard. You may want to see if the following wiki page helps to reduce disk IO on your system:
> > https://github.com/aaronwmorris/indi-allsky/wiki/Disk-Optimizations
> 
> Did these optimisations and installed the latest update - Its been one whole night without a single database locking ..

Update on the issue of database locking -- 

After more than 36 hours of flawless running - the status wend 'Down'. When I checked the status in the terminal with 'systemctl --user status indi-allsky', I got the message that the storage space is all full.
I flushed the images and timelapses and restarted the service. it again started working but now the database episodes are back  - 

Mar  5 21:24:52 indi-allsky [INFO] MainProcess allsky.run() #907: Exposure state: BUSY
Mar  5 21:24:52 indi-allsky [INFO] MainProcess allsky.detectNight() #1436: Sun altitude: -39:12:06.3
Mar  5 21:24:52 indi-allsky [INFO] MainProcess allsky.detectMoonMode() #1453: Moon altitude: 59:57:19.0, phase 96.6%
Mar  5 21:24:52 indi-allsky [INFO] MainProcess allsky.detectMoonMode() #1457: Moon Mode conditions detected
Mar  5 21:24:52 indi-allsky [INFO] ImageWorker001 image.export_raw_image() #899: Raw image written in 2.4703 s
Mar  5 21:24:52 indi-allsky [INFO] ImageWorker001 miscDb.addRawImage() #435: Adding raw image /var/www/html/allsky/images/export/20230305/night/05_21/raw_ccd2_20230305_212345.tif to DB
Mar  5 21:24:53 indi-allsky [INFO] MainProcess indi.getTelescopeRaDec() #834: Telescope Coord: RA 7.91, Dec 30.74
Mar  5 21:24:53 indi-allsky [INFO] MainProcess allsky._queueManualTasks() #2234: Checking for manually submitted tasks
Mar  5 21:24:56 indi-allsky [ERROR] MainProcess allsky.unhandled_exception() #34: An uncaught exception occurred:
Mar  5 21:24:56 indi-allsky [ERROR] MainProcess allsky.unhandled_exception() #35: Type: <class 'sqlalchemy.exc.OperationalError'>
Mar  5 21:24:56 indi-allsky [ERROR] MainProcess allsky.unhandled_exception() #36: Value: (sqlite3.OperationalError) database is locked#012[SQL: UPDATE state SET "createDate"=?, value=? WHERE state."key" = ?]#012[parameters: ('2023-03-05 21:24:53.084412', '1678031692', 'WATCHDOG')]#012(Background on this error at: https://sqlalche.me/e/20/e3q8)
Mar  5 21:24:56 indi-allsky [INFO] ImageWorker001 image.export_raw_image() #947: RAW filename: /var/www/html/allsky/images/export/20230305/night/05_21/raw_ccd2_20230305_212345.tif
Mar  5 21:24:56 indi-allsky [INFO] ImageWorker001 image._convert_16bit_to_8bit() #1922: Resampling image from 16 to 8 bits
Mar  5 21:24:56 indi-allsky [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/home/pi/indi-allsky/allsky.py", line 119, in <module>\n    action_func(*args_list, **kwargs_dict)\n'
Mar  5 21:24:56 indi-allsky [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/home/pi/indi-allsky/indi_allsky/allsky.py", line 967, in run\n    self._miscDb.setState(\'WATCHDOG\', int(loop_start_time))\n'
Mar  5 21:24:56 indi-allsky [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/home/pi/indi-allsky/indi_allsky/flask/miscDb.py", line 527, in setState\n    db.session.commit()\n'
Mar  5 21:24:56 indi-allsky [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/sqlalchemy/orm/scoping.py", line 538, in commit\n    return self._proxied.commit()\n'
Mar  5 21:24:56 indi-allsky [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/sqlalchemy/orm/session.py", line 1903, in commit\n    trans.commit(_to_root=True)\n'
Mar  5 21:24:56 indi-allsky [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "<string>", line 2, in commit\n'
Mar  5 21:24:56 indi-allsky [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/sqlalchemy/orm/state_changes.py", line 137, in _go\n    ret_value = fn(self, *arg, **kw)\n'
Mar  5 21:24:56 indi-allsky [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/sqlalchemy/orm/session.py", line 1218, in commit\n    self._prepare_impl()\n'
Mar  5 21:24:56 indi-allsky [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "<string>", line 2, in _prepare_impl\n'
Mar  5 21:24:56 indi-allsky [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/sqlalchemy/orm/state_changes.py", line 137, in _go\n    ret_value = fn(self, *arg, **kw)\n'
Mar  5 21:24:56 indi-allsky [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/sqlalchemy/orm/session.py", line 1193, in _prepare_impl\n    self.session.flush()\n'
Mar  5 21:24:56 indi-allsky [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/sqlalchemy/orm/session.py", line 4153, in flush\n    self._flush(objects)\n'
Mar  5 21:24:56 indi-allsky [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/sqlalchemy/orm/session.py", line 4290, in _flush\n    transaction.rollback(_capture_exception=True)\n'
[... cut ...]
Mar  5 21:46:06 indi-allsky [INFO] ImageWorker001 image.white_balance_manual_bgr() #2025: Applying manual color balance settings
Mar  5 21:46:06 indi-allsky [INFO] ImageWorker001 image.processImage() #395: Image processed in 6.4575 s
Mar  5 21:46:07 indi-allsky [INFO] ImageWorker001 image.write_img() #982: Image compressed in 0.3696 s
Mar  5 21:46:07 indi-allsky [INFO] ImageWorker001 miscDb.addImage() #107: Adding image /var/www/html/allsky/images/20230305/night/05_21/ccd2_20230305_214536.jpg to DB

StarGeezerPhil, 2023-03-05:
Just to add to this, rather than in the other discussion.

`sqlite3 /var/lib/indi-allsky/indi-allsky.sqlite "PRAGMA journal_mode;"`

Also reports I'm in `WAL `mode.

I'm having the same random service outage/down time after months/year of stable running - since installing the update in the last week/two.

aaronwmorris, 2023-03-05:
Thanks, I'm investigating.

The service is not actually going down, it's just the timestamp of the watchdog is not being updated.

aaronwmorris, 2023-03-05:
Merged #603 

I found another SQLite PRAGMA that might resolve this issue.  https://www.sqlite.org/pragma.html#pragma_synchronous  This setting is recommended to be set to NORMAL when the journal is in WAL mode.  The default is FULL, which performs a filesystem sync after every commit.

I have elected to set synchronous to OFF and just let the OS perform its own syncs.  This should be advantageous for SDcards, although there is a slight chance of DB corruption of power is lost.

aaronwmorris, 2023-03-06:
This resulted in a nice reduction in I/O on my system.  I am going to run some tests with NORMAL, well.

[image]

StarGeezerPhil, 2023-03-06:
Thanks for the update Aaron. Let us know if we can apply changes or you're releasing an update 👍

aaronwmorris, 2023-03-06:
Go ahead and apply the update.

baryah, 2023-03-06:
Thanks Aaron.

If I got it right adding " dbapi_con.execute('PRAGMA synchronous=OFF') " to  " ~/indi-allsky/indi_allsky/flask/__init__.py " under " def _sqlite_pragma_on_connect(dbapi_con, con_record): " should do the trick.

What would be the symptom of corrupted database. I had a power loss while the indi-allsky  was running. After that I was not able to have a look at the "Images" - Clicking on 'Images' got me to an error describing page that the page could not be loaded due to something something.

aaronwmorris, 2023-03-06:
I would have to see the error.

If you were editing the python files, the gunicorn process might have crashed if there was a typo at any point.  You can try to restart the gunicorn process with `systemctl --user restart gunicorn-indi-allsky`

baryah, 2023-03-07:
> I would have to see the error.
> 
> If you were editing the python files, the gunicorn process might have crashed if there was a typo at any point. You can try to restart the gunicorn process with `systemctl --user restart gunicorn-indi-allsky`
> 
> The log for the gunicorn is at `/var/log/indi-allsky/webapp-indi-allsky.log`

[webapp-indi-allsky.log](https://github.com/aaronwmorris/indi-allsky/files/10908658/webapp-indi-allsky.log)

StarGeezerPhil, 2023-03-07:
I'm not familiar with your scripts, so not confident making any edits. 

Watching for any breakthroughs here...

aaronwmorris, 2023-03-07:
I think I might have been too cavalier by setting synchronous to OFF.  I am going to set this to NORMAL and have you guys test this.  None of my cameras are exhibiting the DB locking.  I am reasonably confident, based on what I read that NORMAL should resolve the locking as well.

aaronwmorris, 2023-03-07:
@baryah You had a typo in your edit.  dbasi_con should have been `dbapi_con`

> dbasi_con.execute('PRAGMA synchronous=0')#012NameError: name 'dbasi_con' is not defined

baryah, 2023-03-07:
> @baryah You had a typo in your edit. dbasi_con should have been `dbapi_con`
> 
> > dbasi_con.execute('PRAGMA synchronous=0')#012NameError: name 'dbasi_con' is not defined

yes I corrected it

aaronwmorris, 2023-03-07:
Merged #606 for the image viewer when RAW export files are placed outside the document root.

StarGeezerPhil, 2023-03-07:
@aaronwmorris please let me know what changes to make/where if you'd like us to test (for the synchronous db settings).

aaronwmorris, 2023-03-07:
I have already committed the change and merge it.  Just pull down the latest code and restart.

baryah, 2023-03-08:
> I have had no database locking since yesterday night. Today I tried to generate the darks, I got -
> 
> (indi-allsky) pi@indi-allsky:~/indi-allsky $ ./darks.py sigmaclip 2023-03-05 14:15:34,118 [INFO] MainProcess indi.**init**() #135: creating an instance of IndiClient 2023-03-05 14:15:34,119 [INFO] MainProcess indi.**init**() #142: PyIndi version: 1.9.7 2023-03-05 14:15:34,120 [INFO] MainProcess darks._initialize() #182: Connecting to indiserver INDI::BaseClient::connectServer: creating new connection... INDI::BaseClient::connectServer: Already connected. 2023-03-05 14:15:34,120 [INFO] MainProcess indi.serverConnected() #328: Server connected (localhost:7624) 2023-03-05 14:15:34,126 [INFO] MainProcess indi.newDevice() #198: new device CCD Simulator 2023-03-05 14:15:34,136 [INFO] MainProcess indi.newDevice() #198: new device Telescope Simulator 2023-03-05 14:15:42,129 [WARNING] MainProcess darks._initialize() #205: Connecting to device libcamera_imx477 INDI::BaseClient: Error. Unable to find driver libcamera_imx477 2023-03-05 14:15:42,190 [INFO] MainProcess miscDb.addCamera() #71: Camera DB ID: 2 2023-03-05 14:15:42,192 [INFO] MainProcess indi.updateCcdBlobMode() #390: Set BLOB mode 2023-03-05 14:15:42,192 [ERROR] MainProcess darks._initialize() #254: CCD night gain above maximum, changing to 16 2023-03-05 14:15:45,196 [ERROR] MainProcess darks._initialize() #263: CCD moon mode gain above maximum, changing to 16 2023-03-05 14:15:48,199 [ERROR] MainProcess darks._initialize() #268: CCD day gain below minimum, changing to 1 2023-03-05 14:15:51,203 [WARNING] MainProcess darks._run() #572: ****** IF THE CCD COOLER WAS ENABLED, YOU MAY CONSIDER STOPPING THIS UNTIL THE SENSOR HAS WARMED ****** 2023-03-05 14:15:59,214 [INFO] MainProcess darks._take_exposures() #648: Temp folder: /tmp/tmpv7iack15 2023-03-05 14:15:59,215 [INFO] MainProcess darks.shoot() #278: Taking 60.00000000 s exposure (gain 1) 2023-03-05 14:15:59,217 [INFO] MainProcess libcamera.setCcdExposure() #181: image command: libcamera-still --immediate --nopreview --raw --denoise off --awbgains 1,1 --gain 1 --shutter 60000000 --metadata /tmp/tmpxprjygvm.json --metadata-format json --output /tmp/tmpim1laxxc.dng 2023-03-05 14:17:00,655 [INFO] MainProcess darks._take_exposures() #664: Exposure received in 61.4402 s 2023-03-05 14:17:00,995 [INFO] MainProcess darks._take_exposures() #680: Image average adu: 257.88 2023-03-05 14:17:00,996 [INFO] MainProcess darks.shoot() #278: Taking 60.00000000 s exposure (gain 1) 2023-03-05 14:17:00,997 [INFO] MainProcess libcamera.setCcdExposure() #181: image command: libcamera-still --immediate --nopreview --raw --denoise off --awbgains 1,1 --gain 1 --shutter 60000000 --metadata /tmp/tmpfawidsq1.json --metadata-format json --output /tmp/tmp16wrbm5j.dng 2023-03-05 14:18:02,376 [INFO] MainProcess darks._take_exposures() #664: Exposure received in 61.3799 s 2023-03-05 14:18:02,724 [INFO] MainProcess darks._take_exposures() #680: Image average adu: 256.96 2023-03-05 14:18:02,724 [INFO] MainProcess darks.shoot() #278: Taking 60.00000000 s exposure (gain 1) 2023-03-05 14:18:02,725 [INFO] MainProcess libcamera.setCcdExposure() #181: image command: libcamera-still --immediate --nopreview --raw --denoise off --awbgains 1,1 --gain 1 --shutter 60000000 --metadata /tmp/tmpjmwa8902.json --metadata-format json --output /tmp/tmp__clhl51.dng 2023-03-05 14:19:04,094 [INFO] MainProcess darks._take_exposures() #664: Exposure received in 61.3697 s 2023-03-05 14:19:04,438 [INFO] MainProcess darks._take_exposures() #680: Image average adu: 256.92 2023-03-05 14:19:04,439 [INFO] MainProcess darks.shoot() #278: Taking 60.00000000 s exposure (gain 1) 2023-03-05 14:19:04,440 [INFO] MainProcess libcamera.setCcdExposure() #181: image command: libcamera-still --immediate --nopreview --raw --denoise off --awbgains 1,1 --gain 1 --shutter 60000000 --metadata /tmp/tmpbq74wsez.json --metadata-format json --output /tmp/tmpd38g30aq.dng 2023-03-05 14:20:05,809 [INFO] MainProcess darks._take_exposures() #664: Exposure received in 61.3703 s 2023-03-05 14:20:06,128 [INFO] MainProcess darks._take_exposures() #680: Image average adu: 257.91 2023-03-05 14:20:06,128 [INFO] MainProcess darks.shoot() #278: Taking 60.00000000 s exposure (gain 1) 2023-03-05 14:20:06,129 [INFO] MainProcess libcamera.setCcdExposure() #181: image command: libcamera-still --immediate --nopreview --raw --denoise off --awbgains 1,1 --gain 1 --shutter 60000000 --metadata /tmp/tmpn0zen3u2.json --metadata-format json --output /tmp/tmpsssjrdl1.dng 2023-03-05 14:21:07,505 [INFO] MainProcess darks._take_exposures() #664: Exposure received in 61.3769 s 2023-03-05 14:21:07,827 [INFO] MainProcess darks._take_exposures() #680: Image average adu: 257.94 2023-03-05 14:21:07,828 [INFO] MainProcess darks.shoot() #278: Taking 60.00000000 s exposure (gain 1) 2023-03-05 14:21:07,829 [INFO] MainProcess libcamera.setCcdExposure() #181: image command: libcamera-still --immediate --nopreview --raw --denoise off --awbgains 1,1 --gain 1 --shutter 60000000 --metadata /tmp/tmplvn3i0aa.json --metadata-format json --output /tmp/tmp_6llz7c5.dng 2023-03-05 14:22:09,247 [INFO] MainProcess darks._take_exposures() #664: Exposure received in 61.4191 s 2023-03-05 14:22:09,597 [INFO] MainProcess darks._take_exposures() #680: Image average adu: 256.93 2023-03-05 14:22:09,597 [INFO] MainProcess darks.shoot() #278: Taking 60.00000000 s exposure (gain 1) 2023-03-05 14:22:09,598 [INFO] MainProcess libcamera.setCcdExposur [... cut ...] wait_for_image raw = rawpy.imread(str(filename_p)) File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/rawpy/**init**.py", line 20, in imread d.open_file(pathOrFile) File "rawpy/_rawpy.pyx", line 409, in rawpy._rawpy.RawPy.open_file File "rawpy/_rawpy.pyx", line 936, in rawpy._rawpy.RawPy.handle_error rawpy._rawpy.LibRawIOError: b'Input/output error' BaseClient::~BaseClient: Probability of detecting a deadlock.

Not able to make out why am I not able to generate darks??

aaronwmorris, 2023-03-08:
Are you running out of space?

baryah, 2023-03-08:
No have plenty of space - some 90GB or so

aaronwmorris, 2023-03-08:
I am reviewing my code and I already catch conditions if the image file is missing or if the size is 0.

I can only assume the file is either corrupt or the rawpy module does not like the format.  Can you try reseating your CSI cable in both the raspi and the camera module?

StarGeezerPhil, 2023-03-09:
Hi @aaronwmorris, despite a promising start after the update I'm afraid it was still going down.

Just checked and it appears there's another updated package - installed it, but I'm now getting _**Internal Server Error**_.

About to try a cold power off/on cycle.

aaronwmorris, 2023-03-09:
What is the exception in `/var/log/indi-allsky/webapp-indi-allsky.log`?

StarGeezerPhil, 2023-03-09:
Sorry - just doing the update again (I didn't replace INDI for the first time that last time - don't know if that's likely to have made any difference).

StarGeezerPhil, 2023-03-09:
Same after trying the update again.

Exception from:` /var/log/indi-allsky/webapp-indi-allsky.log` as below:

```
Mar  9 21:44:53 SkyCam [ERROR] MainProcess app.log_exception() #1741: Exception on /indi-allsky/ [GET]#012Traceback (most recent call last):#012  File "/home/StarGeezer/indi-allsky/virtualenv/indi-allsky/lib/python3.9//site-packages/sqlalchemy/engine/result.py", line 614, in _only_one_row#012    raise exc.MultipleResultsFound(#012sqlalchemy.exc.MultipleResultsFound: Multiple rows were found when exactly one was required
```

aaronwmorris, 2023-03-09:
I will need a little more of that exception to see where is was generated.

aaronwmorris, 2023-03-09:
I think I found it.  One of my test cameras is generating the exception.

StarGeezerPhil, 2023-03-10:
Awesome, thanks Aaron. Updated and the webUI is back online. 

However, new imageworker exception:

```
Mar 10 00:23:50 SkyCam [ERROR] MainProcess allsky._startImageWorker() #734: Image worker exception: 
Mar 10 00:23:50 SkyCam [ERROR] MainProcess allsky._startImageWorker() #734: Image worker exception: KeyError: 'CCD_INFO'
Mar 10 00:23:50 SkyCam [ERROR] MainProcess allsky._startImageWorker() #734: Image worker exception:     elif self.config['CCD_INFO']['CCD_CFA']['CFA_TYPE'].get('text'):
Mar 10 00:23:50 SkyCam [ERROR] MainProcess allsky._startImageWorker() #734: Image worker exception:   File "/home/pi/indi-allsky/indi_allsky/image.py", line 1570, in add
Mar 10 00:23:50 SkyCam [ERROR] MainProcess allsky._startImageWorker() #734: Image worker exception:     self.image_processor.add(filename_p, exposure, exp_date, exp_elapsed, camera)
Mar 10 00:23:50 SkyCam [ERROR] MainProcess allsky._startImageWorker() #734: Image worker exception:   File "/home/pi/indi-allsky/indi_allsky/image.py", line 288, in processImage
Mar 10 00:23:50 SkyCam [ERROR] MainProcess allsky._startImageWorker() #734: Image worker exception:     self.processImage(i_dict)
Mar 10 00:23:50 SkyCam [ERROR] MainProcess allsky._startImageWorker() #734: Image worker exception:   File "/home/pi/indi-allsky/indi_allsky/image.py", line 224, in saferun
Mar 10 00:23:50 SkyCam [ERROR] MainProcess allsky._startImageWorker() #734: Image worker exception:     self.saferun()
Mar 10 00:23:50 SkyCam [ERROR] MainProcess allsky._startImageWorker() #734: Image worker exception:   File "/home/pi/indi-allsky/indi_allsky/image.py", line 195, in run
Mar 10 00:23:50 SkyCam [ERROR] MainProcess allsky._startImageWorker() #734: Image worker exception: Traceback (most recent call last):
Mar 10 00:23:50 SkyCam [INFO] MainProcess allsky.run() #997: Exposure state: READY
Mar 10 00:23:50 SkyCam [INFO] MainProcess allsky.run() #996: Camera last ready: 9.5s
Mar 10 00:23:40 SkyCam [INFO] MainProcess allsky.run() #1142: Exposure received in 1.4525 s (1.4515)
Mar 10 00:23:39 SkyCam [INFO] MainProcess allsky.run() #1211: Total time since last exposure 45.1407 s
Mar 10 00:23:39 SkyCam [INFO] MainProcess libcamera.setCcdExposure() #192: image command: libcamera-still --immediate --nopreview --raw --denoise off --awbgains 1,1 --gain 16 --shutter 1000 --metadata /tmp/tmpa3iadv4d.json --metadata-format json --output /tmp/tmpsajhurw5.dng
```

aaronwmorris, 2023-03-10:
Merged #620 to pass the CFA pattern between processes.  Unfortunately, this required a database change so it will be necessary to re-run setup.sh.

baryah, 2023-03-10:
Hi Aaron. I am still not able to capture Darks. Previously, the error would arise after around 9 to 10 exposures, now it is happening after the first one - Timeout INDI::BaseClient: Error. Unable to find driver libcamera_imx477
2023-03-10 13:06:50,223 [INFO] MainProcess miscDb.addCamera() #80: Camera DB ID: 1
2023-03-10 13:06:50,223 [INFO] MainProcess indi.updateCcdBlobMode() #400: Set BLOB mode
2023-03-10 13:06:50,224 [ERROR] MainProcess darks._initialize() #282: CCD day gain below minimum, changing to 1
2023-03-10 13:06:53,227 [WARNING] MainProcess darks._run() #586: ****** IF THE CCD COOLER WAS ENABLED, YOU MAY CONSIDER STOPPING THIS UNTIL THE SENSOR HAS WARMED ******
2023-03-10 13:07:01,239 [INFO] MainProcess darks._take_exposures() #662: Temp folder: /tmp/tmpbhab4xsy
2023-03-10 13:07:01,240 [INFO] MainProcess darks.shoot() #292: Taking 15.00000000 s exposure (gain 1)
2023-03-10 13:07:01,242 [INFO] MainProcess libcamera.setCcdExposure() #192: image command: libcamera-still --immediate --nopreview --encoding jpg --quality 100 --gain 1 --shutter 15000000 --metadata /tmp/tmpe7tk3vu6.json --metadata-format json --output /tmp/tmptt7bydh5.jpg
2023-03-10 13:07:41,283 [ERROR] MainProcess libcamera.setCcdExposure() #209: Exposure timeout
Traceback (most recent call last):
  File "/home/pi/indi-allsky/indi_allsky/camera/libcamera.py", line 207, in setCcdExposure
    self.libcamera_process.wait(timeout=timeout)
  File "/usr/lib/python3.9/subprocess.py", line 1189, in wait
    return self._wait(timeout=timeout)
  File "/usr/lib/python3.9/subprocess.py", line 1911, in _wait
    raise TimeoutExpired(self.args, timeout)
subprocess.TimeoutExpired: Command '['libcamera-still', '--immediate', '--nopreview', '--encoding', 'jpg', '--quality', '100', '--gain', '1', '--shutter', '15000000', '--metadata', '/tmp/tmpe7tk3vu6.json', '--metadata-format', 'json', '--output', '/tmp/tmptt7bydh5.jpg']' timed out after 40.0 seconds

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/indi-allsky/./darks.py", line 90, in <module>
    action_func()
  File "/home/pi/indi-allsky/indi_allsky/darks.py", line 432, in sigmaclip
    self._sigmaclip()
  File "/home/pi/indi-allsky/indi_allsky/darks.py", line 439, in _sigmaclip
    self._run(IndiAllSkyDarksSigmaClip)
  File "/home/pi/indi-allsky/indi_allsky/darks.py", line 598, in _run
    self._take_exposures(exposure, dark_filename_t, bpm_filename_t, ccd_bits, stacking_class)
  File "/home/pi/indi-allsky/indi_allsky/darks.py", line 674, in _take_exposures
    self.shoot(exposure_f, sync=True, timeout=timeout)
  File "/home/pi/indi-allsky/indi_allsky/darks.py", line 294, in shoot
    self.indiclient.setCcdExposure(exposure, sync=sync, timeout=timeout)
  File "/home/pi/indi-allsky/indi_allsky/camera/libcamera.py", line 210, in setCcdExposure
    raise TimeOutException('Timeout waiting for exposure')
indi_allsky.exceptions.TimeOutException: Timeout waiting for exposure
BaseClient::~BaseClient: Probability of detecting a deadlock.

StarGeezerPhil, 2023-03-10:
> Merged #620 to pass the CFA pattern between processes. Unfortunately, this required a database change so it will be necessary to re-run setup.sh.

Thanks @aaronwmorris, just to confirm this has worked for me. I'll let you know if it stays up or if it continues to go down.

baryah, 2023-03-10:
> Hi Aaron. I am still not able to capture Darks. Previously, the error would arise after around 9 to 10 exposures, now it is happening after the first one - Timeout
>
> ```
>   INDI::BaseClient: Error. [...]

I found out that the 'metadata file' in /tmp folder is not being generated.

StarGeezerPhil, 2023-03-10:
> > Merged #620 to pass the CFA pattern between processes. Unfortunately, this required a database change so it will be necessary to re-run setup.sh.
> 
> Thanks @aaronwmorris, just to confirm this has worked for me. I'll let you know if it stays up or if it continues to go down.

Update: down again:

```
Mar 10 10:11:43 SkyCam [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/sqlalchemy/engine/default.py", line 736, in do_execute\n    cursor.execute(statement, parameters)\n'
Mar 10 10:11:43 SkyCam [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/sqlalchemy/engine/base.py", line 1900, in _execute_context\n    self.dialect.do_execute(\n'
Mar 10 10:11:43 SkyCam [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/sqlalchemy/util/compat.py", line 210, in raise_\n    raise exception\n'
Mar 10 10:11:43 SkyCam [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/sqlalchemy/engine/base.py", line 2124, in _handle_dbapi_exception\n    util.raise_(\n'
Mar 10 10:11:43 SkyCam [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/sqlalchemy/engine/base.py", line 1943, in _execute_context\n    self._handle_dbapi_exception(\n'
Mar 10 10:11:43 SkyCam [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/sqlalchemy/engine/base.py", line 1572, in _execute_clauseelement\n    ret = self._execute_context(\n'
Mar 10 10:11:43 SkyCam [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/sqlalchemy/sql/elements.py", line 334, in _execute_on_connection\n    return connection._execute_clauseelement(\n'
Mar 10 10:11:43 SkyCam [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/sqlalchemy/engine/base.py", line 1705, in _execute_20\n    return meth(self, args_10style, kwargs_10style, execution_options)\n'
Mar 10 10:11:43 SkyCam [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/sqlalchemy/orm/persistence.py", line 1001, in _emit_update_statements\n    c = connection._execute_20(\n'
Mar 10 10:11:43 SkyCam [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/sqlalchemy/orm/persistence.py", line 237, in save_obj\n    _emit_update_statements(\n'
Mar 10 10:11:43 SkyCam [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/sqlalchemy/orm/unitofwork.py", line 630, in execute\n    util.preloaded.orm_persistence.save_obj(\n'
Mar 10 10:11:43 SkyCam [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/sqlalchemy/orm/unitofwork.py", line 456, in execute\n    rec.execute(self)\n'
Mar 10 10:11:43 SkyCam [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/sqlalchemy/orm/session.py", line 3544, in _flush\n    flush_context.execute()\n'
Mar 10 10:11:43 SkyCam [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/sqlalchemy/util/compat.py", line 210, in raise_\n    raise exception\n'
Mar 10 10:11:43 SkyCam [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/sqlalchemy/util/langhelpers.py", line 70, in __exit__\n    compat.raise_(\n'
Mar 10 10:11:43 SkyCam [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/sqlalchemy/orm/session.py", line 3584, in _flush\n    transaction.rollback(_capture_exception=True)\n'
Mar 10 10:11:43 SkyCam [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/sqlalchemy/orm/session.py", line 3444, in flush\n    self._flush(objects)\n'
Mar 10 10:11:43 SkyCam [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/sqlalchemy/orm/session.py", line 808, in _prepare_impl\n    self.session.flush()\n'
Mar 10 10:11:43 SkyCam [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/sqlalchemy/orm/session.py", line 829, in commit\n    self._prepare_impl()\n'
Mar 10 10:11:43 SkyCam [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/sqlalchemy/orm/session.py", line 1451, in commit\n    self._transaction.commit(_to_root=self.future)\n'
Mar 10 10:11:43 SkyCam [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "<string>", line 2, in commit\n'
[... cut ...]
Mar 10 10:11:40 SkyCam [ERROR] MainProcess allsky.unhandled_exception() #35: Type: <class 'sqlalchemy.exc.OperationalError'>
Mar 10 10:11:40 SkyCam [ERROR] MainProcess allsky.unhandled_exception() #34: An uncaught exception occurred:
Mar 10 10:11:37 SkyCam [INFO] MainProcess allsky._queueManualTasks() #2379: Checking for manually submitted tasks
Mar 10 10:11:37 SkyCam [INFO] MainProcess indi.getTelescopeRaDec() #844: Telescope Coord: RA 21.29, Dec 53.47
Mar 10 10:11:37 SkyCam [INFO] MainProcess allsky.detectMoonMode() #1557: Moon altitude: -23:33:02.3, phase 91.9%
```

aaronwmorris, 2023-03-10:
Okay... well I tried NORMAL mode for synchronous. I have changed it back to OFF in https://github.com/aaronwmorris/indi-allsky/pull/621 .

As for the darks, this will not work in JPEG mode, you will have to switch to DNG raw mode.

baryah, 2023-03-10:
> Okay... well I tried NORMAL mode for synchronous. I have changed it back to OFF in #621 .
> 
> As for the darks, this will not work in JPEG mode, you will have to switch to DNG raw mode.

Thanks @aaronwmorris
Now the Darks are being taken and stacked. Was it always the mode (DNG) that was required.

Regarding mode for synchronous - did you face any issues with NORMAL mode. I have been running it in NORMAL mode for more than 24 hours now, with no database locks or any other errors!!

aaronwmorris, 2023-03-10:
@baryah @StarGeezerPhil reported that he received a database lock in his last post.

My takeaway from this is that the situation with SDcards is they are just crap.  I will continue to think about this.

aaronwmorris, 2023-03-10:
@StarGeezerPhil setup.sh only needs to be run if there are database changes or possibly python module updates (more rare).  When either of these conditions occur, I increment the __config_level__ in ./indi_allsky/version.py which will cause indi-allsky to not start until you re-run setup.sh.  If indi-allsky starts, then you should not need to re-run setup.sh

As for recent changes... over the past month or two, yes, there have been significant changes.  I have been working toward making indi-allsky into a more "cloud-native" product.  As part of this effort, I have been leaning more heavily on the database.

The web interface previously used the indi-allsky PID to determine if the process is running.  This is really easy since the web app and the capture process run on the same device, just checking on a PID is very simple.  However, if you run these components on separate devices (or containers), it is not as easy.  I am using a simple state table in the database and updating one row every 10 seconds with a timestamp as a replacement.

Updating a row every 10 seconds should not cause this much of a problem, but here we are.

Another change that might be having an effect is establishing new flask application contexts for database transactions.  Previously, I was using a shared flask context across all of the multiprocessing processes.  The shared context worked likely because sqlite was being used, but caused problems with mysql connections with TLS.  Establishing multiple contexts could be causing more IOs with sqlite.  Maybe, maybe not.

aaronwmorris, 2023-03-10:
I did some quick and dirty work to revert my changes before having all of the separate contexts.  The separate contexts appear to have no effect on the I/O.

Removing the watchdog commits has minor impact, maybe -0.5 tps IO.

I feel like setting synchronous to off probably presents very little danger.  It is not like there are extensive complex transactions.... the SQL insert queries are just simple single row inserts.  Long term, I think moving to mysql is probably a good move.

aaronwmorris, 2023-03-10:
@baryah Can you see if you have any libcamera related updates?  I have never experienced a corrupt frame or one that rawpy was unable to parse.

If you could run it again and perhaps send the bad DNG to me.  I would like to see what is going on.

```
sudo apt-get update
sudo apt-get upgrade
```

StarGeezerPhil, 2023-03-11:
It ran really nicely for 10-12 hours and then went down this morning (at around the night-day threshold):

```
Mar 11 06:33:09 SkyCam [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/sqlalchemy/engine/default.py", line 736, in do_execute\n    cursor.execute(statement, parameters)\n'
Mar 11 06:33:09 SkyCam [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/sqlalchemy/engine/base.py", line 1900, in _execute_context\n    self.dialect.do_execute(\n'
Mar 11 06:33:09 SkyCam [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/sqlalchemy/util/compat.py", line 210, in raise_\n    raise exception\n'
Mar 11 06:33:09 SkyCam [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/sqlalchemy/engine/base.py", line 2124, in _handle_dbapi_exception\n    util.raise_(\n'
Mar 11 06:33:09 SkyCam [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/sqlalchemy/engine/base.py", line 1943, in _execute_context\n    self._handle_dbapi_exception(\n'
Mar 11 06:33:09 SkyCam [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/sqlalchemy/engine/base.py", line 1572, in _execute_clauseelement\n    ret = self._execute_context(\n'
Mar 11 06:33:09 SkyCam [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/sqlalchemy/sql/elements.py", line 334, in _execute_on_connection\n    return connection._execute_clauseelement(\n'
Mar 11 06:33:09 SkyCam [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/sqlalchemy/engine/base.py", line 1705, in _execute_20\n    return meth(self, args_10style, kwargs_10style, execution_options)\n'
Mar 11 06:33:09 SkyCam [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/sqlalchemy/orm/persistence.py", line 1001, in _emit_update_statements\n    c = connection._execute_20(\n'
Mar 11 06:33:09 SkyCam [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/sqlalchemy/orm/persistence.py", line 237, in save_obj\n    _emit_update_statements(\n'
Mar 11 06:33:09 SkyCam [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/sqlalchemy/orm/unitofwork.py", line 630, in execute\n    util.preloaded.orm_persistence.save_obj(\n'
Mar 11 06:33:09 SkyCam [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/sqlalchemy/orm/unitofwork.py", line 456, in execute\n    rec.execute(self)\n'
Mar 11 06:33:09 SkyCam [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/sqlalchemy/orm/session.py", line 3544, in _flush\n    flush_context.execute()\n'
Mar 11 06:33:09 SkyCam [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/sqlalchemy/util/compat.py", line 210, in raise_\n    raise exception\n'
Mar 11 06:33:09 SkyCam [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/sqlalchemy/util/langhelpers.py", line 70, in __exit__\n    compat.raise_(\n'
Mar 11 06:33:09 SkyCam [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/sqlalchemy/orm/session.py", line 3584, in _flush\n    transaction.rollback(_capture_exception=True)\n'
Mar 11 06:33:09 SkyCam [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/sqlalchemy/orm/session.py", line 3444, in flush\n    self._flush(objects)\n'
Mar 11 06:33:09 SkyCam [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/sqlalchemy/orm/session.py", line 808, in _prepare_impl\n    self.session.flush()\n'
Mar 11 06:33:09 SkyCam [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/sqlalchemy/orm/session.py", line 829, in commit\n    self._prepare_impl()\n'
Mar 11 06:33:09 SkyCam [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/sqlalchemy/orm/session.py", line 1451, in commit\n    self._transaction.commit(_to_root=self.future)\n'
Mar 11 06:33:09 SkyCam [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "<string>", line 2, in commit\n'
Mar 11 06:33:09 SkyCam [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/home/pi/indi-allsky/indi_allsky/flask/miscDb.py", line 697, in setState\n    db.session.commit()\n'
[... cut ...]
Mar 11 06:33:06 SkyCam [ERROR] MainProcess allsky.unhandled_exception() #36: Value: (sqlite3.OperationalError) database is locked#012[SQL: UPDATE state SET "createDate"=?, value=? WHERE state."key" = ?]#012[parameters: ('2023-03-11 06:33:02.729769', '1678516382', 'WATCHDOG')]#012(Background on this error at: https://sqlalche.me/e/14/e3q8)
Mar 11 06:33:06 SkyCam [ERROR] MainProcess allsky.unhandled_exception() #35: Type: <class 'sqlalchemy.exc.OperationalError'>
Mar 11 06:33:06 SkyCam [ERROR] MainProcess allsky.unhandled_exception() #34: An uncaught exception occurred:
Mar 11 06:33:02 SkyCam [INFO] MainProcess allsky._queueManualTasks() #2379: Checking for manually submitted tasks
```

baryah, 2023-03-11:
> @baryah Can you see if you have any libcamera related updates? I have never experienced a corrupt frame or one that rawpy was unable to parse.
> 
> If you could run it again and perhaps send the bad DNG to me. I would like to see what is going on.
> 
> ```
> sudo apt-get update
> sudo apt-get upgrade
> ```

```
pi@indi-allsky:~ $ sudo apt update
Hit:1 http://deb.debian.org/debian bullseye InRelease
Get:2 http://security.debian.org/debian-security bullseye-security InRelease [48.4 kB]
Get:3 http://deb.debian.org/debian bullseye-updates InRelease [44.1 kB]
Get:4 http://security.debian.org/debian-security bullseye-security/main armhf Packages [230 kB]
Get:5 http://archive.raspberrypi.org/debian bullseye InRelease [23.6 kB]          
Get:6 http://security.debian.org/debian-security bullseye-security/main arm64 Packages [230 kB]
Hit:7 https://www.astroberry.io/repo bullseye InRelease                           
Get:8 http://archive.raspberrypi.org/debian bullseye/main armhf Packages [315 kB]
Get:9 http://archive.raspberrypi.org/debian bullseye/main arm64 Packages [307 kB]
Fetched 1,104 kB in 2s (453 kB/s)
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
1 package can be upgraded. Run 'apt list --upgradable' to see it.
pi@indi-allsky:~ $ sudo apt list --upgradable 
Listing... Done
rp-prefapps/stable 0.46 arm64 [upgradable from: 0.45]
N: There is 1 additional version. Please use the '-a' switch to see it
```

I am getting the same error - again and again - 

```
2023-03-11 19:00:05,645 [INFO] MainProcess libcamera.setCcdExposure() #181: image command: libcamera-still --immediate --nopreview --raw --denoise off --awbgains 1,1 --gain 1 --shutter 60000000 --metadata /tmp/tmpnyaqrhkm.json --metadata-format json --output /tmp/tmp8lyz0uod.dng
2023-03-11 19:01:07,044 [INFO] MainProcess darks._take_exposures() #664: Exposure received in 61.4007 s
Traceback (most recent call last):
  File "/home/pi/indi-allsky/./darks.py", line 90, in <module>
    action_func()
  File "/home/pi/indi-allsky/indi_allsky/darks.py", line 418, in sigmaclip
    self._sigmaclip()
  File "/home/pi/indi-allsky/indi_allsky/darks.py", line 425, in _sigmaclip
    self._run(IndiAllSkyDarksSigmaClip)
  File "/home/pi/indi-allsky/indi_allsky/darks.py", line 584, in _run
    self._take_exposures(exposure, dark_filename_t, bpm_filename_t, ccd_bits, stacking_class)
  File "/home/pi/indi-allsky/indi_allsky/darks.py", line 667, in _take_exposures
    hdulist = self._wait_for_image(exposure_f)
  File "/home/pi/indi-allsky/indi_allsky/darks.py", line 330, in _wait_for_image
    raw = rawpy.imread(str(filename_p))
  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/rawpy/__init__.py", line 20, in imread
    d.open_file(pathOrFile)
  File "rawpy/_rawpy.pyx", line 409, in rawpy._rawpy.RawPy.open_file
  File "rawpy/_rawpy.pyx", line 936, in rawpy._rawpy.RawPy.handle_error
rawpy._rawpy.LibRawIOError: b'Input/output error'
BaseClient::~BaseClient: Probability of detecting a deadlock.
```

I am attaching the generated DNG file
[tmp8lyz0uod.dng.zip](https://github.com/aaronwmorris/indi-allsky/files/10948635/tmp8lyz0uod.dng.zip)

baryah, 2023-03-11:
> It ran really nicely for 10-12 hours and then went down this morning (at around the night-day threshold):
>
> ```
> Mar 11 06:33:09 SkyCam [ERROR] MainProcess allsky.unhandled_exception() #41: '  File [...]

Hi @StarGeezerPhil 
When I first installed Indi-allsky I was having these database lockdowns once every few hours and at times even earlier. Changing the SDCard sorted it out. Could your issue also be related to that. 

> > Issues 2, 3, and 5 (possibly 4) all point to issues with the database. With the database journal in WAL mode, it should not be locking since this allows multiple readers and writers.
> > In the past, I have experienced the database locking issue which is why the WAL journal is utilized.
> > Is this a new SDcard you are using? I do not like to cavalierly blame the card, but bad storage is the only thing makes sense to me.
> 
> Hard to believe, but changing the SD card actually did the trick. Everything is working fine now. Thanks. a lot for your guidance. I couldn't have ever guessed this to be the issue.

StarGeezerPhil, 2023-03-11:
@baryah I don't believe so, as the issue started immediately upon installing the update Aaron suggested last week.

I've had the device running flawlessly for ~year or so and only recently replaced the microSD (Sandisk Extreme Pro), the SD model I reliably use in numerous dev/production Pi devices as well as my astronomy controller and DSLR.

It's the timing - I don't think this is due to the SD failing, though could be I/O limitations (but you're not going to find many faster cards than that model red/black SanDisk).

baryah, 2023-03-11:
> > @baryah Can you see if you have any libcamera related updates? I have never experienced a corrupt frame or one that rawpy was unable to parse.
> > If you could run it again and perhaps send the bad DNG to me. I would like to see what is going on.
> > ```
> > sudo apt-get update
> > sudo apt-get upgrade
> > ```
> 
> ```
> pi@indi-allsky:~ $ sudo apt update
> Hit:1 http://deb.debian.org/debian bullseye InRelease
> Get:2 http://security.debian.org/debian-security bullseye-security InRelease [48.4 kB]
> Get:3 http://deb.debian.org/debian bullseye-updates InRelease [44.1 kB]
> Get:4 http://security.debian.org/debian-security bullseye-security/main armhf Packages [230 kB]
> Get:5 http://archive.raspberrypi.org/debian bullseye InRelease [23.6 kB]          
> Get:6 http://security.debian.org/debian-security bullseye-security/main arm64 Packages [230 kB]
> Hit:7 https://www.astroberry.io/repo bullseye InRelease                           
> Get:8 http://archive.raspberrypi.org/debian bullseye/main armhf Packages [315 kB]
> Get:9 http://archive.raspberrypi.org/debian bullseye/main arm64 Packages [307 kB]
> Fetched 1,104 kB in 2s (453 kB/s)
> Reading package lists... Done
> Building dependency tree... Done
> Reading state information... Done
> 1 package can be upgraded. Run 'apt list --upgradable' to see it.
> pi@indi-allsky:~ $ sudo apt list --upgradable 
> Listing... Done
> rp-prefapps/stable 0.46 arm64 [upgradable from: 0.45]
> N: There is 1 additional version. Please use the '-a' switch to see it
> ```
> 
> Hi @aaronwmorris I am getting the same error - again and again -
> 
> ```
> 2023-03-11 19:00:05,645 [INFO] MainProcess libcamera.setCcdExposure() #181: image command: libcamera-still --immediate --nopreview --raw --denoise off --awbgains 1,1 --gain 1 --shutter 60000000 --metadata /tmp/tmpnyaqrhkm.json --metadata-format json --output /tmp/tmp8lyz0uod.dng
> 2023-03-11 19:01:07,044 [INFO] MainProcess darks._take_exposures() #664: Exposure received in 61.4007 s
> Traceback (most recent call last):
>   File "/home/pi/indi-allsky/./darks.py", line 90, in <module>
>     action_func()
>   File "/home/pi/indi-allsky/indi_allsky/darks.py", line 418, in sigmaclip
>     self._sigmaclip()
>   File "/home/pi/indi-allsky/indi_allsky/darks.py", line 425, in _sigmaclip
>     self._run(IndiAllSkyDarksSigmaClip)
>   File "/home/pi/indi-allsky/indi_allsky/darks.py", line 584, in _run
>     self._take_exposures(exposure, dark_filename_t, bpm_filename_t, ccd_bits, stacking_class)
>   File "/home/pi/indi-allsky/indi_allsky/darks.py", line 667, in _take_exposures
>     hdulist = self._wait_for_image(exposure_f)
>   File "/home/pi/indi-allsky/indi_allsky/darks.py", line 330, in _wait_for_image
>     raw = rawpy.imread(str(filename_p))
>   File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/rawpy/__init__.py", line 20, in imread
>     d.open_file(pathOrFile)
>   File "rawpy/_rawpy.pyx", line 409, in rawpy._rawpy.RawPy.open_file
>   File "rawpy/_rawpy.pyx", line 936, in rawpy._rawpy.RawPy.handle_error
> rawpy._rawpy.LibRawIOError: b'Input/output error'
> BaseClient::~BaseClient: Probability of detecting a deadlock.
> ```
> 
> I am attaching the generated DNG file [tmp8lyz0uod.dng.zip](https://github.com/aaronwmorris/indi-allsky/files/10948635/tmp8lyz0uod.dng.zip)

Update -- the dark frame capture works fine with a '--count 5', returns same error with '--count 9'. Haven't tested other 'counts'.

aaronwmorris, 2023-03-11:
Merged #625 to set the watchdog every 3 minutes instead of 10 seconds.

[30 more replies]