aaronwmorris/indi-allsky #600: Feature Request: INDI GPS USB and NTP Server

HxPI, 2023-03-04:
Is it possible to connect a USB GPS dongle, like the u-Blox7 USB GPS variants, and allow INDI to connect to it for location and time information? Would it also be possible to setup an internal NTP server to use this time information or can INDI provide the time information to indi-allsky? There are devices like the LeoNTP that will provide a portable NTP service but was wondering if this can be done with the RPi using a simple GPS USB dongle.

HxPI, 2023-03-04:
Never mind. I see there is a GPS time sync option in configuration webpage. Not sure about the GPSd install requirement. Is that something I have to do manually or does it happen automatically when this feature is enabled and/or a GPS device is detected?

aaronwmorris, 2023-03-04:
The script `./misc/setup_gpsd.sh` that should do most of the gpsd install and setup for you.

HxPI, 2023-03-04:
I’ll try that. Thanks.

aaronwmorris, 2023-03-04:
Yes, it will disable the normal systemd timesyncd.

HxPI, 2023-03-04:
So there’s no way to restore the normal systems timesyncd after enabling GPSd? I’d rather leave the system the way it is than permanently change it.

aaronwmorris, 2023-03-04:
It is better to just sync time using ntp and gpsd directly than let indi-allsky do it.

https://github.com/aaronwmorris/indi-allsky/wiki/GPSd-NTP-Time-Sync

HxPI, 2023-03-04:
OK thanks. I’ll close this.

closed 2023-03-04 by HxPI