aaronwmorris/indi-allsky #655: New issue on image rotate?

StarGeezerPhil, 2023-03-20:
I've just turned on image rotation for the first time and getting the image worker failing.

Not the end of the world - I can disable it again, but log file below:

```
Mar 20 18:22:00 SkyCam [ERROR] MainProcess allsky._startImageWorker() #758: Image worker exception: 
Mar 20 18:22:00 SkyCam [ERROR] MainProcess allsky._startImageWorker() #758: Image worker exception: 
Mar 20 18:22:00 SkyCam [ERROR] MainProcess allsky._startImageWorker() #758: Image worker exception: cv2.error: OpenCV(4.6.0) /io/opencv/modules/core/src/matrix_iterator.cpp:71: error: (-215:Assertion failed) A.size == arrays[i0]->size in function 'init'
Mar 20 18:22:00 SkyCam [ERROR] MainProcess allsky._startImageWorker() #758: Image worker exception:     adu = cv2.mean(src=data_mono, mask=self._adu_mask)[0]
Mar 20 18:22:00 SkyCam [ERROR] MainProcess allsky._startImageWorker() #758: Image worker exception:   File "/home/StarGeezer/indi-allsky/indi_allsky/image.py", line 1204, in calculate_histogram
Mar 20 18:22:00 SkyCam [ERROR] MainProcess allsky._startImageWorker() #758: Image worker exception:     adu, adu_average = self.calculate_histogram(self.image_processor.image, exposure)
Mar 20 18:22:00 SkyCam [ERROR] MainProcess allsky._startImageWorker() #758: Image worker exception:   File "/home/StarGeezer/indi-allsky/indi_allsky/image.py", line 362, in processImage
Mar 20 18:22:00 SkyCam [ERROR] MainProcess allsky._startImageWorker() #758: Image worker exception:     self.processImage(i_dict)
Mar 20 18:22:00 SkyCam [ERROR] MainProcess allsky._startImageWorker() #758: Image worker exception:   File "/home/StarGeezer/indi-allsky/indi_allsky/image.py", line 235, in saferun
Mar 20 18:22:00 SkyCam [ERROR] MainProcess allsky._startImageWorker() #758: Image worker exception:     self.saferun()
Mar 20 18:22:00 SkyCam [ERROR] MainProcess allsky._startImageWorker() #758: Image worker exception:   File "/home/StarGeezer/indi-allsky/indi_allsky/image.py", line 206, in run
Mar 20 18:22:00 SkyCam [ERROR] MainProcess allsky._startImageWorker() #758: Image worker exception: Traceback (most recent call last):
```

Also, I'm using an image mask, but noticed this in the log:
`Mar 20 18:23:50 SkyCam [WARNING] ImageWorker004 sqm._generateSqmMask() #63: Using central 20% ROI for SQM calculations`
Is this correct?

aaronwmorris, 2023-03-20:
Did you rotate your mask?

aaronwmorris, 2023-03-20:
Unfortunately, yes.  I am not particularly happy with this order of operations, but I do not want to break anyone's setup by changing this.

https://github.com/aaronwmorris/indi-allsky/wiki/Image-order-of-operations

I have considered having automatic rotation, etc of the mask, but one would have to create a mask for the image orientation that comes directly out of the camera, which could be very confusing.

StarGeezerPhil, 2023-03-20:
Yeah, that makes sense. But anything calibration-wise like darks/bpm wise should be raw before any processing.

Not sure what's going on here though...

[image]

Had this ages back when I first got the IMX. I'll give it a reboot

StarGeezerPhil, 2023-03-20:
Ok, rebooting had no effect, but the Day <> Night change fixed the image.

closed 2023-03-20 by StarGeezerPhil

StarGeezerPhil, 2023-03-20:
Brilliant. Thanks @aaronwmorris 

One other quick question: using libcamera as you know, and wanting HQ videos, so I'm not compressing my JPEG output. However, the upload script (FTP) is using the full size JPEGs rather than the 'latest.jpg' image that is used within INDI-allsky for the "Latest" page.

Could it use this - or could we have an option whether to upload full-resolution or the compressed version used locally?

I hadn't realised until recently - as it's quite a lot of bandwidth.

aaronwmorris, 2023-03-20:
Not easily... let me think about it.