aaronwmorris/indi-allsky #656: Recent update: INDI-allsky confusion over image location / camera name

StarGeezerPhil, 2023-03-21:
After a recent update images started to get stored in a new location (/images/ccd_49e24657-c476-4488-85ff-91bed9eae2e2) rather than just /images.

This hasn't seemed to be a problem until today when the timelapse generation script seems to be getting confused:

Yesterday's timelapse generated fine, but the night timelapse (etc.) failed due to the image folder:
[image]

aaronwmorris, 2023-03-21:
I noticed that too.  Let me know what happens tonight.  My cameras had a similar issue when I upgraded the code, but it was only the one time.

StarGeezerPhil, 2023-03-21:
Did yours revert back to saving into /images or has yours continued within the subfolder?

All images since I updated on 10 March have been within the /ccd_(guid)/ subfolder.

I'll let you know what happens with the timelapse for today/tonight 👍

aaronwmorris, 2023-03-21:
The GUID name is the new standard.

The tasks in the screenshot, did you initiate those manually or were those automatic tasks?

StarGeezerPhil, 2023-03-21:
I *think* they were initialized automatically. All the manual tasks I tried after failed.

Today's daytime timelapse has generated - but only the first few hours (been running ok all day).

StarGeezerPhil, 2023-03-22:
Update @aaronwmorris: manually re-generating yesterday's daytime timelapse worked, and the night time one worked automatically this morning.

Seems hit and miss - I'll let you know this evening.

aaronwmorris, 2023-03-22:
After re-reading the screenshot above, I think I understand the error.  For some reason a task to create the night timelapse for 3/21, but it was 8:23am.  Night had not started, therefore the destination folder did not yet exist.

StarGeezerPhil, 2023-03-22:
Ah - sorry yes, that must have been a manual one (and I must have selected the wrong date), but the issue does stay the same.

A subsequent one for the same period:
[image]

Hopefully it's just a one-off - I still can't get 20 March to generate, but they have done since.

aaronwmorris, 2023-03-22:
It could be a corrupted image file.  You would have to find the exception.

aaronwmorris, 2023-04-30:
Closing this issue for now.  Reopen if there are any other issues.

closed 2023-04-30 by aaronwmorris