aaronwmorris/indi-allsky #666: Install Issue

mshetzer, 2023-04-01:
When installing on an Odroid N2+ 4GB, Ubuntu mate odroid, the install does not complete.

All goes well until:

./setup.sh 
runs for a while, then aborts with 

"Errors were encountered while processing:/tmp/apt-dpkg-install-HegNO3/019-ffmpeg_7%3a4.2.7-0ubuntu0.1_arm64.deb"
E: Sub-process /usr/bin/dpkg returned an error code (1)
The script exited abnormally, please try to run again.

Any suggestions greatly appreciated.  Looks like a great program!!!
Matt

aaronwmorris, 2023-04-01:
You got the lucky issue 666 :-)

I do not see the exact error here, but my best guess is you are out of space on one of the file systems.  If you send more of the error, we can get to the bottom of it.

mshetzer, 2023-04-01:
I'm a lucky guy !!!  It's a brand new build on a 64g mini SD card.  What can I supply to help.

mshetzer, 2023-04-01:
That was the whole error, but it was unpacking proj-bin (6.3.1-1).  Please keep in mind, I'm a Linux moron !
Matt

aaronwmorris, 2023-04-01:
No worries.

Try running this and give the full error:

```
sudo apt-get -f install
```

mshetzer, 2023-04-01:
Looks like it complete ran without any errors

aaronwmorris, 2023-04-01:
Alrighty then.  Try re-running setup.sh

mshetzer, 2023-04-01:
indi: For astro/planetary cameras normally connected via USB
libcamera: supports cameras connected via CSI interface on Raspberry Pi SoCs

1) indi
2) libcamera
Select a camera interface: 1

Fixing git checkout permissions
**** Installing packages... ****

Current default time zone: 'US/Mountain'
Local time is now:      Sat Apr  1 13:36:10 MDT 2023.
Universal Time is now:  Sat Apr  1 19:36:10 UTC 2023.

Hit:1 http://ports.ubuntu.com/ubuntu-ports focal InRelease
Hit:2 http://archive.canonical.com/ubuntu focal InRelease                                                                          
Get:3 http://ports.ubuntu.com/ubuntu-ports focal-updates InRelease [114 kB]                                                        
Hit:4 http://ppa.launchpad.net/hardkernel/ppa/ubuntu focal InRelease                                         
Hit:5 http://deb.odroid.in/n2 focal InRelease                                                           
Get:6 http://ports.ubuntu.com/ubuntu-ports focal-backports InRelease [108 kB]
Get:7 http://ports.ubuntu.com/ubuntu-ports focal-security InRelease [114 kB]
Get:8 http://ports.ubuntu.com/ubuntu-ports focal-updates/main arm64 Packages [1790 kB]
Fetched 2126 kB in 3s (669 kB/s)   
Reading package lists... Done
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'libxslt1-dev' instead of 'libxslt-dev'
cron is already the newest version (3.0pl1-136ubuntu1).
default-libmysqlclient-dev is already the newest version (1.0.5ubuntu2).
gfortran is already the newest version (4:9.3.0-1ubuntu2).
libffi-dev is already the newest version (3.3-4).
pkg-config is already the newest version (0.29.1-0ubuntu4).
whiptail is already the newest version (0.52.21-4ubuntu2).
gifsicle is already the newest version (1.92-2).
libatlas-base-dev is already the newest version (3.10.3-8ubuntu7).
libcfitsio-dev is already the newest version (3.470-3).
libilmbase-dev is already the newest version (2.3.0-6build1).
libnova-dev is already the newest version (0.16-4).
libopencv-dev is already the newest version (4.2.0+dfsg-5).
swig is already the newest version (4.0.1-5build1).
apache2 is already the newest version (2.4.41-4ubuntu3.14).
avahi-daemon is already the newest version (0.7-4ubuntu7.1).
build-essential is already the newest version (12.8ubuntu1.1).
ca-certificates is already the newest version (20211016ubuntu0.20.04.1).
cmake is already the newest version (3.16.3-1ubuntu1.20.04.1).
cpio is already the newest version (2.13+dfsg-2ubuntu0.3).
dbus-user-session is already the newest version (1.12.16-2ubuntu2.3).
git is already the newest version (1:2.25.1-1ubuntu3.10).
libcurl4-gnutls-dev is already the newest version (7.68.0-1ubuntu2.18).
libdbus-1-dev is already the newest version (1.12.16-2ubuntu2.3).
libglib2.0-dev is already the newest version (2.64.6-1~ubuntu20.04.4).
libgnutls28-dev is already the newest version (3.6.13-2ubuntu1.8).
libgtk-3-0 is already the newest version (3.24.20-0ubuntu1.1).
libssl-dev is already the newest version (1.1.1f-1ubuntu2.17).
libxml2-dev is already the newest version (2.9.10+dfsg-5ubuntu0.20.04.5).
libxslt1-dev is already the newest version (1.1.34-4ubuntu0.20.04.1).
policykit-1 is already the newest version (0.105-26ubuntu1.3).
rsyslog is already the newest version (8.2001.0-1ubuntu1.3).
sqlite3 is already the newest version (3.31.1-4ubuntu0.5).
tzdata is already the newest version (2022g-0ubuntu0.20.04.1).
cargo is already the newest version (0.66.0+ds0ubuntu0.libgit2-0ubuntu0.20.04).
jq is already the newest version (1.6-1ubuntu0.20.04.1).
libopenblas-dev is already the newest version (0.3.8+ds-1ubuntu0.20.04.1).
libopenexr-dev is already the newest version (2.3.0-6ubuntu0.5).
python3-pip is already the newest version (20.0.2-5ubuntu1.8).
python3.9 is already the newest version (3.9.5-3ubuntu0~20.04.1).
python3.9-dev is already the newest version (3.9.5-3ubuntu0~20.04.1).
python3.9-venv is already the newest version (3.9.5-3ubuntu0~20.04.1).
rustc is already the newest version (1.65.0+dfsg0ubuntu1~llvm2-0ubuntu0.20.04).
virtualenv is already the newest version (20.0.17-1ubuntu0.4).
The following packages were automatically installed and are no longer required:
  libfwupdplugin1 libxmlb1 ubuntu-advantage-desktop-daemon
Use 'sudo apt autoremove' to remove them.
Suggested packages:
  ffmpeg-doc
The following NEW packages will be installed:
  ffmpeg
0 upgraded, 1 newly installed, 0 to remove and 3 not upgraded.
Need to get 0 B/1440 kB of archives.
After this operation, 2021 kB of additional disk space will be used.
(Reading database ... 219602 files and directories currently installed.)
Preparing to unpack .../ffmpeg_7%3a4.2.7-0ubuntu0.1_arm64.deb ...
Unpacking ffmpeg (7:4.2.7-0ubuntu0.1) ...
dpkg: error processing archive /var/cache/apt/archives/ffmpeg_7%3a4.2.7-0ubuntu0.1_arm64.deb (--unpack):
 trying to overwrite '/usr/bin/ffprobe', which is also in package aml-libs 20200708-14f646a-1
dpkg-deb: error: paste subprocess was killed by signal (Broken pipe)
Errors were encountered while processing:
 /var/cache/apt/archives/ffmpeg_7%3a4.2.7-0ubuntu0.1_arm64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)

The script exited abnormally, please try to run again...

aaronwmorris, 2023-04-01:
Okay, try this and re-run setup.sh

```
sudo apt-get -o Dpkg::Options::="--force-overwrite" install ffmpeg
```

I pulled this from https://forum.odroid.com/viewtopic.php?t=39663

mshetzer, 2023-04-01:
That did the trick.  Thank you so much for your help !!!

mshetzer, 2023-04-01:
The server is up and running, and I'm able to access the web site to mange everything, however I don't see any images, nor the camera (ZWO ASI224MC).  I have tried to run ./allsky.sh cameraReport from the indi-allsky directory, however receive 'No such file or directory'.  I'm sure the error is on my part as its been 20 years since I've been in Linux.

mshetzer, 2023-04-01:
Nevermind.  Rebooted and everything started to work.  Thanks again for all your help !!!

aaronwmorris, 2023-04-02:
Great!

aaronwmorris, 2023-04-30:
Closing this issue for now.

closed 2023-04-30 by aaronwmorris