aaronwmorris/indi-allsky #671: SFTP not uploading every image - despite config saying upload every image

vastmeridian, 2023-04-04:
I'm not seeing every file transferred to my upstream server overnight.  I'm running Release 6.1 on Ubuntu 20.04

Looking thriough the logs I see this from time to time

```
Apr  4 20:30:52 allsky [ERROR] MainProcess allsky._startFileUploadWorker() #812: Upload worker exception: Traceback (most recent call last):
Apr  4 20:30:52 allsky [ERROR] MainProcess allsky._startFileUploadWorker() #812: Upload worker exception:   File "/home/username/indi-allsky/indi_allsky/uploader.py", line 93, in run
Apr  4 20:30:52 allsky [ERROR] MainProcess allsky._startFileUploadWorker() #812: Upload worker exception:     self.saferun()
Apr  4 20:30:52 allsky [ERROR] MainProcess allsky._startFileUploadWorker() #812: Upload worker exception:   File "/home/username/indi-allsky/indi_allsky/uploader.py", line 254, in saferun
Apr  4 20:30:52 allsky [ERROR] MainProcess allsky._startFileUploadWorker() #812: Upload worker exception:     client.put(**put_kwargs)
Apr  4 20:30:52 allsky [ERROR] MainProcess allsky._startFileUploadWorker() #812: Upload worker exception:   File "/home/username/indi-allsky/indi_allsky/filetransfer/pycurl_sftp.py", line 152, in put
Apr  4 20:30:52 allsky [ERROR] MainProcess allsky._startFileUploadWorker() #812: Upload worker exception:     raise e from e
Apr  4 20:30:52 allsky [ERROR] MainProcess allsky._startFileUploadWorker() #812: Upload worker exception:   File "/home/username/indi-allsky/indi_allsky/filetransfer/pycurl_sftp.py", line 131, in put
Apr  4 20:30:52 allsky [ERROR] MainProcess allsky._startFileUploadWorker() #812: Upload worker exception:     self.client.perform()
Apr  4 20:30:52 allsky [ERROR] MainProcess allsky._startFileUploadWorker() #812: Upload worker exception: pycurl.error: (79, '')
Apr  4 20:30:52 allsky [ERROR] MainProcess allsky._startFileUploadWorker() #812: Upload worker exception:
```

I've added the FILETRANSFER section from the config and an extract from the log file:
[FILETRANSFER.txt](https://github.com/aaronwmorris/indi-allsky/files/11153394/FILETRANSFER.txt)
[logfile_extract.txt](https://github.com/aaronwmorris/indi-allsky/files/11153398/logfile_extract.txt)

Thanks

aaronwmorris, 2023-04-04:
Curl error 79 is

> CURLE_SSH (79)
> An unspecified error occurred during the SSH session. 

Try adding the following option to the PyCURL options in the file transfer config and restart.
```
{
    "VERBOSE": 1
}
```

I am not sure where the log entries will appear.  They may appear in indi-allsky.log or they might be in /var/log/syslog.log

vastmeridian, 2023-04-05:
Thanks.  I've added that so will restart and see what happens.
I think I may have solved it.  I found a runaway process on the destination host (not the allsky) that had consumed all disk space.  That's now corrected and I *think* all is well.  I'll know after tonight.  I'll also confirm which log file is used for the logging.

vastmeridian, 2023-04-23:
This can be closed. I think the root cause of the issue was the runaway process on the remote fileserver.

closed 2023-04-23 by vastmeridian