aaronwmorris/indi-allsky #686: Stacking info on image not correct

RoySalisbury, 2023-04-09:
The label template has the "Stacking {stack_method:s}" but even when stacking is enabled, this always says "off"

I have included screen shots to show the configuration and result.

[image]
[image]
[image]

aaronwmorris, 2023-04-09:
Merged #687.  This one was tough to spot.

aaronwmorris, 2023-04-30:
Closing this for now.  Let me know if it does not work for you.

closed 2023-04-30 by aaronwmorris