aaronwmorris/indi-allsky #692: Is IMX219 based camera out of the question?

kbrown73, 2023-04-14:
Hi,

I have an old Sony IMX219 based Arducam camera module on a Raspberry Pi 3B running Debian GNU/Linux 11 (bullseye) (aarch64). It works with libcamera (tested with libcamera-still).  I had high hopes on getting indi-allsky running with it but the setup.sh doesn't list imx219. Is this camera module a no go for indi-allsky?

aaronwmorris, 2023-04-16:
The IMX219 camera should work fine.  You can pick any of the existing libcamera profiles as a test.  The profiles really only tell indi-allsky which debayering algorithm to use for the images.  You can override this manually in the indi-allsky config.

If you could let me know which debayering algorithm is correct for the IMX219, I can get the profile added when I get back.  The libcamera-still command should report it.  I am traveling for work this weekend and I will be able to submit changes on Wednesday or Thursday.

kbrown73, 2023-04-16:
Brilliant. I just went with imx477 and all seems to work as expected. Colours are correct too.

aaronwmorris, 2023-04-16:
Are you using DNG (raw) more?

kbrown73, 2023-04-17:
Apologies. I seem to have misjudged the images I was getting. The colours don't seem quite right after all and changing the bayer pattern doesn't change it much with any setting. My images always have a strong magenta tint to them. Tried with DNG and JPG.

Initially I thought this was because my IR/UV cut filter was off but turning that on doesn't get rid of the problem.

Here's what libcamera-still --list-cameras reports:

```
Available cameras
-----------------
0 : imx219 [3280x2464] (/base/soc/i2c0mux/i2c@1/imx219@10)
    Modes: 'SBGGR10_CSI2P' : 640x480 [30.00 fps - (0, 0)/0x0 crop]
                             1640x1232 [30.00 fps - (0, 0)/0x0 crop]
                             1920x1080 [30.00 fps - (0, 0)/0x0 crop]
                             3280x2464 [30.00 fps - (0, 0)/0x0 crop]
           'SBGGR8' : 640x480 [30.00 fps - (0, 0)/0x0 crop]
                      1640x1232 [30.00 fps - (0, 0)/0x0 crop]
                      1920x1080 [30.00 fps - (0, 0)/0x0 crop]
                      3280x2464 [30.00 fps - (0, 0)/0x0 crop]

```

Apologies if this is something trivial. I'm a pretty new to indi-allsky.

kbrown73, 2023-04-17:
Update. I had a look what kind of Red and Blue gain values libcamera-vid was coming up with and normalised those between 0 and 1 and put those as the Red, Green and Blue balance factors in indi-allsky config. That seems to give me much better results.

Not sure how you really should go about setting these values. Here's what I entered:
R: 0.558139535
G: 0.465116279
B: 1.0

I left the bayer pattern as auto detect...

aaronwmorris, 2023-04-19:
If you have a pink bias, it is probably because of the lack of the IR/Cut filter.  There is a separate tuning file for the imx219 without the IR filter.  I removed the IR/Cut filter on my imx477 and switching to the noir specific tuning file fixed the color balance issues.

Try adding this to the additional libcamera options.

```
--tuning-file /usr/share/libcamera/ipa/raspberrypi/imx219_noir.json
```

kbrown73, 2023-04-20:
Thanks for the suggestion. My camera is actually an Arducam module with motorised IR cut filter which I can control on and off with GPIO. It was pink even with the filter in front of the sensor. Anyway, all is good. The camera works with indi-allsky which was the main issue. Suppose you can close this now if you wish.

I'll take a look at the json file as well, thanks!

aaronwmorris, 2023-04-30:
Closing this for now.  Let me know if any issues pop up.

closed 2023-04-30 by aaronwmorris