aaronwmorris/indi-allsky #693: Timelapse Generation

StarGeezerPhil, 2023-04-17:
I'm not sure if this issue is isolated to me, but when regenerating timelapses/keograms, the process never seems to complete - following the steps:

1. Delete Keogram/Startrails
2. Generate Keogram/Startrails

All images appear to generate successfully during the normal daily day/night routines.

After Deleting and then starting the "Generate Keogram/Startrail" routine I have noticed that the images do not appear to disappear from the Timelapses page, however a task does show in the Timelapse > Generate page.

I have left the process as shown "RUNNING" multiple times for more than 3 hours with no completion - and seemingly no images appearing or disappearing from the timelapse page or file system. I have also rebooted and cold-booted the device several times during testing.

E.g.
[image]

I'm going to try Delete All / Generate All to see if this has better success.

StarGeezerPhil, 2023-04-17:
Just an update to confirm - "Delete All" does see the images disappearing from the Timelapses page and from the filesystem.

Is there a bug in the "Delete Keogram/Startrail" script?

aaronwmorris, 2023-04-20:
I have run some tests in one of my test cameras and deleting and generating all seem to work.  If there is a problem with deleting keograms, the error would be in the webapp logs in `/var/log/indi-allsky/webapp-indi-allsky.log`

aaronwmorris, 2023-05-10:
Closing this for now.  Just re-open if you still experience issues.

closed 2023-05-10 by aaronwmorris