aaronwmorris/indi-allsky #694: allow upload of the generated startrail video

xsnrg, 2023-04-18:
On the videoviewer page, when the startrail image thumbnail is clicked, a generated video timelapse of the startrail can be viewed. It would be nice to have this video as an optional upload. The video is named something like `allsky-startrail_timelapse_ccd1_20230417_night.mp4"

aaronwmorris, 2023-04-19:
The startrail video uploading is currently tied with the "Transfer video" option.  I will create a separate option for the startrail video.

xsnrg, 2023-04-19:
Hi Aaron, for the videos, with everything checked to transfer except the metadata, I end up with startrail JPG images and timelapse night images, but I am not seeing the mp4 video with startrail in the in the name, just the day and night timelapse. If it is supposed to be included with the "Transfer video" option, perhaps I have something configured incorrectly.

Transferred with the below settings in the image are:

allsky-startrail_ccd_<date>_night.jpg
allsky-timelapse_ccd1_<date>_(day and night).mp4
allsky-keogram_ccd1_<date>_(day and night).jpg

not transferred:

allsky-startrail_timelapse_ccd1_<date>_night.mp4

[image]

aaronwmorris, 2023-04-20:
Well, I may have to perform some testing then.

aaronwmorris, 2023-04-21:
Merged #699 .  There was indeed a code problem preventing the startrail video upload.

I am going to proceed with adding separate upload options for the startrail videos.

aaronwmorris, 2023-04-21:
Merged #700 with separate startrail video upload configuration options.

xsnrg, 2023-04-21:
Thank you. Pulling now and will close this as complete after the first transfer. I see the setting.

closed 2023-06-21 by xsnrg