aaronwmorris/indi-allsky #696: ASI178MC exposure failure

GeremiaForino, 2023-04-19:
Hello all,

I installed yesterday Indi-Allsky and from this morning, worked well all night, have "Exposure state: BUSY" and "Exposure state: ALERT" with the relative hang-up of the reception of images, I reread the Issues present here but I did not find relative response and need your help, attached from log the two periods of the event. 

First event:

Apr 19 06:16:52 raspberrypi [INFO] MainProcess allsky.run() #1039: Exposure state: OK
Apr 19 06:16:52 raspberrypi [INFO] MainProcess allsky.detectNight() #1587: Sun altitude: -0:18:29.6
Apr 19 06:16:52 raspberrypi [INFO] MainProcess allsky.detectMoonMode() #1604: Moon altitude: 2:42:24.5, phase 1.3%
Apr 19 06:16:52 raspberrypi [INFO] MainProcess indi.getCcdTemperature() #848: Sensor temperature: 37.8
Apr 19 06:16:52 raspberrypi [INFO] MainProcess indi.getTelescopeRaDec() #833: Telescope Coord: RA 19.05, Dec 40.96
Apr 19 06:16:52 raspberrypi [INFO] MainProcess allsky._queueManualTasks() #2422: Checking for manually submitted tasks
Apr 19 06:16:58 raspberrypi [INFO] MainProcess allsky.shoot() #1824: Taking 0.02924824 s exposure (gain 0)
Apr 19 06:16:58 raspberrypi [INFO] MainProcess indi.set_number() #1215: Setting Duration (s) = 0.02924823760986328
Apr 19 06:16:58 raspberrypi [INFO] MainProcess allsky.run() #1272: Total time since last exposure 15.0457 s
Apr 19 06:17:03 raspberrypi [INFO] MainProcess allsky.run() #1038: Camera last ready: 18.1s
Apr 19 06:17:03 raspberrypi [INFO] MainProcess allsky.run() #1039: Exposure state: BUSY
Apr 19 06:17:03 raspberrypi [INFO] MainProcess allsky.detectNight() #1587: Sun altitude: -0:16:51.5
Apr 19 06:17:03 raspberrypi [INFO] MainProcess allsky.detectMoonMode() #1604: Moon altitude: 2:44:22.9, phase 1.3%
Apr 19 06:17:03 raspberrypi [INFO] MainProcess indi.getCcdTemperature() #848: Sensor temperature: 37.8
Apr 19 06:17:03 raspberrypi [INFO] MainProcess indi.getTelescopeRaDec() #833: Telescope Coord: RA 19.05, Dec 40.96
Apr 19 06:17:03 raspberrypi [INFO] MainProcess allsky._queueManualTasks() #2422: Checking for manually submitted tasks
Apr 19 06:17:05 raspberrypi [INFO] MainProcess indi.newMessage() #324: new Message 2023-04-19T04:17:05: [ERROR] Exposure failed after 3 attempts. 
Apr 19 06:17:14 raspberrypi [INFO] MainProcess allsky.run() #1038: Camera last ready: 29.4s
Apr 19 06:17:14 raspberrypi [INFO] MainProcess allsky.run() #1039: Exposure state: ALERT

second event:

Apr 19 13:01:19 raspberrypi [INFO] MainProcess allsky.run() #1039: Exposure state: OK
Apr 19 13:01:19 raspberrypi [INFO] MainProcess allsky.detectNight() #1587: Sun altitude: 60:12:53.0
Apr 19 13:01:19 raspberrypi [INFO] MainProcess allsky.detectMoonMode() #1604: Moon altitude: 54:17:20.9, phase 0.6%
Apr 19 13:01:19 raspberrypi [INFO] MainProcess indi.getCcdTemperature() #848: Sensor temperature: 58.2
Apr 19 13:01:19 raspberrypi [INFO] MainProcess indi.getTelescopeRaDec() #833: Telescope Coord: RA 1.81, Dec 40.96
Apr 19 13:01:19 raspberrypi [INFO] MainProcess allsky._queueManualTasks() #2422: Checking for manually submitted tasks
Apr 19 13:01:29 raspberrypi [INFO] MainProcess allsky.shoot() #1824: Taking 0.00003201 s exposure (gain 0)
Apr 19 13:01:29 raspberrypi [INFO] MainProcess indi.set_number() #1215: Setting Duration (s) = 3.201000072294846e-05
Apr 19 13:01:29 raspberrypi [INFO] MainProcess allsky.run() #1272: Total time since last exposure 15.0252 s
Apr 19 13:01:30 raspberrypi [INFO] MainProcess allsky.run() #1038: Camera last ready: 12.5s
Apr 19 13:01:30 raspberrypi [INFO] MainProcess allsky.run() #1039: Exposure state: BUSY
Apr 19 13:01:30 raspberrypi [INFO] MainProcess allsky.detectNight() #1587: Sun altitude: 60:12:52.7
Apr 19 13:01:30 raspberrypi [INFO] MainProcess allsky.detectMoonMode() #1604: Moon altitude: 54:16:52.8, phase 0.6%
Apr 19 13:01:30 raspberrypi [INFO] MainProcess indi.getCcdTemperature() #848: Sensor temperature: 58.2
Apr 19 13:01:30 raspberrypi [INFO] MainProcess indi.getTelescopeRaDec() #833: Telescope Coord: RA 1.81, Dec 40.96
Apr 19 13:01:30 raspberrypi [INFO] MainProcess allsky._queueManualTasks() #2422: Checking for manually submitted tasks
Apr 19 13:01:35 raspberrypi [INFO] MainProcess indi.newMessage() #324: new Message 2023-04-19T11:01:35: [ERROR] Exposure failed after 3 attempts. 
Apr 19 13:01:41 raspberrypi [INFO] MainProcess allsky.run() #1038: Camera last ready: 23.7s
Apr 19 13:01:41 raspberrypi [INFO] MainProcess allsky.run() #1039: Exposure state: ALERT

To continue I have to stop and restart the services
thanks in advance for your help, Thanks especially to Aaron Morris for the commitment in this project.

regards
Geremia

aaronwmorris, 2023-04-19:
Are you using a USB3 or USB2 cable?  Have you tried another cable?

GeremiaForino, 2023-04-19:
I'm using a USB2 cable on USB2 port, I didn't use a USB3 cable on USB3 port because I seem to have read that it was not supported. If yes, I can try

aaronwmorris, 2023-04-19:
USB2 cables are generally less prone to problems, however, bad cables are a very common source of problems like this.  If you have a different cable to try, I would do that first as a test.

GeremiaForino, 2023-04-19:
yes I have some to try, tomorrow in the day I do some tests and I update you

Thanks for now

GeremiaForino, 2023-04-20:
I tried to lower the band width of the USB from 80 to 40 and the software is running since last night without interruption, I have not changed the cable, I'm curious to see if this was the problem, I continue the analysis

aaronwmorris, 2023-04-20:
That was going to be my next suggestion.  Glad to hear it might be helping.

closed 2023-04-21 by GeremiaForino