aaronwmorris/indi-allsky #701: Alternate location for tmp writes?

kbrown73, 2023-04-21:
My old Pi 3B was struggling a bit with IO wait so I went through the suggestions in the [Disk Optimizations](https://github.com/aaronwmorris/indi-allsky/wiki/Disk-Optimizations) page. I wouldn't really want to convert the whole system to use the RAM based tmpfs as for example some g++ compilations use it a lot and I simply would run out of RAM or deadlock the Pi when the tmpfs gets full.

So my question is: Is it possible to tell indi-allsky to use an alternate location for its temporary files? I could then have a separate smaller tmpfs mounted somewhere in my home directory just for indi-allsky purposes.

aaronwmorris, 2023-04-21:
Merged #702 to add an environment file.  You will have to pull down the latest code and re-run setup.sh.

The environment file will be `/etc/indi-allsky/indi-allsky.env`

Within this file, you can set something like this:
```
TMPDIR=/path/to/tmp
```

Python uses TMPDIR, TEMP, or TMP environment variables.  The directory must exist, otherwise a default is used.

kbrown73, 2023-04-21:
Brilliant. That works!

Any suggestions for the minimum size for it?

aaronwmorris, 2023-04-21:
100-200MB should be sufficient

kbrown73, 2023-04-21:
Perfect. Thank you very much!

closed 2023-04-30 by aaronwmorris