aaronwmorris/indi-allsky #708: Pure RGB noise frames

WarrenSchultz, 2023-04-24:
Odd issue. It'll cruise along for a long while (day+), and then I'll get periodic images of RGB noise. Previously, only a single frame and then it would resume normally, but this evening, all recent images are pure RGB noise. I have dark frames/BPM enabled.

While testing inside before deployment, I've got the camera covered still for testing long exposures when this came up. 

Only warnings in the log:
`ImageWorker001 image._calibrate() #1832: Temperature matched dark not found: 31.20c`

What's the temperature delta range for dark frames being able to be matched?
If I run another batch of darks generation, does it add to the existing library or replace it?

[image]

Camera config:
```
{
    "PROPERTIES": {
        "CCD_CONTROLS": {
            "BandWidth": 40
        }
    },
    "SWITCHES": {
        "CCD_VIDEO_FORMAT": {
            "on": [
                "ASI_IMG_RAW16"
            ],
            "off": [
                "ASI_IMG_RAW8"
            ]
        }
    }
}
```

aaronwmorris, 2023-04-24:
I have not seen that previously.  Perhaps you do have a bad dark frame.

The temperature matching warning is a red herring.  indi-allsky will first look for a dark frame within 5 degrees C, if one is not found, it will look for any dark frame with a similar exposure.  You should see a later message where a dark frame was either found or not found again.

If you generate new dark frames, the newer ones should have preference to the older ones (they are sorted by date).  However, the darks module has a flush command to remove all existing dark frames.

WarrenSchultz, 2023-04-28:
@aaronwmorris Ok, I ran the script and generated another set of dark frames, and things seem to be stable now for the last four days.

Of course, running that long, I'm now running low on swap, but I think I can sort that out from the wiki. :)

Thanks!

closed 2023-04-28 by WarrenSchultz