aaronwmorris/indi-allsky #709: Connection refused MQTT

ispybadguys, 2023-04-24:
What am I doing wrong? I've tried MQTT port - 8883 = TLS, 1883 = standard both with TLS on and off. The password I set is correct but I get 
 FileUploader006 uploader.processUpload() #382: Connection failure: [Errno 111] Connection refused

aaronwmorris, 2023-04-24:
Have you installed mosquitto?  There is a script to install and setup, `./misc/setup_mosquitto.sh`

ispybadguys, 2023-04-24:
Yes. I used the script.

aaronwmorris, 2023-04-24:
It sounds like mosquitto is not running.  Try starting it....

`sudo systemctl start mosquitto`

ispybadguys, 2023-04-24:
If I execute:
 mosquitto_sub -d -t indi-allsky/stars 
Client mosqsub|16771-astroberr sending CONNECT
1682367962: New connection from ::1 on port 1883.
1682367962: New client connected from ::1 as mosqsub|16771-astroberr (c1, k60).
1682367962: No will message specified.
1682367962: Sending CONNACK to mosqsub|16771-astroberr (0, 0)
Client mosqsub|16771-astroberr received CONNACK (0)
Client mosqsub|16771-astroberr sending SUBSCRIBE (Mid: 1, Topic: indi-allsky/stars, QoS: 0)
1682367962: Received SUBSCRIBE from mosqsub|16771-astroberr
1682367962: 	indi-allsky/stars (QoS 0)
1682367962: mosqsub|16771-astroberr 0 indi-allsky/stars
1682367962: Sending SUBACK to mosqsub|16771-astroberr
Client mosqsub|16771-astroberr received SUBACK
Subscribed (mid: 1): 0
Client mosqsub|16771-astroberr sending PINGREQ
1682368022: Received PINGREQ from mosqsub|16771-astroberr
1682368022: Sending PINGRESP to mosqsub|16771-astroberr
Client mosqsub|16771-astroberr received PINGRESP

ispybadguys, 2023-04-24:
astroberry@astroberry:~ $ sudo systemctl start mosquitto
Job for mosquitto.service failed because the control process exited with error code.
See "systemctl status mosquitto.service" and "journalctl -xe" for details.

ps -elf| grep mosquito
0 S astrobe+ 19569 17227  0  80   0 -  1836 pipe_r 14:37 pts/1    00:00:00 grep --color=auto mosquito

ispybadguys, 2023-04-24:
I have never used MQTT but I changed the port to 1883 and turned TLS off and I am getting results now. Using command:
astroberry@astroberry:~ $ mosquitto_sub -d -t indi-allsky/longitude
Client mosqsub|21753-astroberr sending CONNECT
1682369127: New connection from ::1 on port 1883.
1682369127: New client connected from ::1 as mosqsub|21753-astroberr (c1, k60).
1682369127: No will message specified.
1682369127: Sending CONNACK to mosqsub|21753-astroberr (0, 0)
Client mosqsub|21753-astroberr received CONNACK (0)
Client mosqsub|21753-astroberr sending SUBSCRIBE (Mid: 1, Topic: indi-allsky/longitude, QoS: 0)
1682369127: Received SUBSCRIBE from mosqsub|21753-astroberr
1682369127: 	indi-allsky/longitude (QoS 0)
1682369127: mosqsub|21753-astroberr 0 indi-allsky/longitude
1682369127: Sending SUBACK to mosqsub|21753-astroberr
1682369127: Sending PUBLISH to mosqsub|21753-astroberr (d0, q0, r1, m0, 'indi-allsky/longitude', ... (5 bytes))
Client mosqsub|21753-astroberr received SUBACK
Subscribed (mid: 1): 0
Client mosqsub|21753-astroberr received PUBLISH (d0, q0, r1, m0, 'indi-allsky/longitude', ... (5 bytes))
-84.0
1682369129: New connection from ::1 on port 1883.
1682369129: New client connected from ::1 as a3f20e84-f6a0-43f5-8b11-5dd741fc0a99 (c1, k60, u'indi-allsky').
1682369129: No will message specified.
1682369129: Sending CONNACK to a3f20e84-f6a0-43f5-8b11-5dd741fc0a99 (0, 0)
1682369129: Received PUBLISH from a3f20e84-f6a0-43f5-8b11-5dd741fc0a99 (d0, q0, r1, m0, 'indi-allsky/latest', ... (145445 bytes))
1682369129: Received PUBLISH from a3f20e84-f6a0-43f5-8b11-5dd741fc0a99 (d0, q0, r1, m0, 'indi-allsky/exposure', ... (7 bytes))
1682369129: Received PUBLISH from a3f20e84-f6a0-43f5-8b11-5dd741fc0a99 (d0, q0, r1, m0, 'indi-allsky/gain', ... (1 bytes))
1682369129: Received PUBLISH from a3f20e84-f6a0-43f5-8b11-5dd741fc0a99 (d0, q0, r1, m0, 'indi-allsky/bin', ... (1 bytes))
1682369129: Received PUBLISH from a3f20e84-f6a0-43f5-8b11-5dd741fc0a99 (d0, q0, r1, m0, 'indi-allsky/temp', ... (4 bytes))
1682369129: Received PUBLISH from a3f20e84-f6a0-43f5-8b11-5dd741fc0a99 (d0, q0, r1, m0, 'indi-allsky/sunalt', ... (4 bytes))
1682369129: Received PUBLISH from a3f20e84-f6a0-43f5-8b11-5dd741fc0a99 (d0, q0, r1, m0, 'indi-allsky/moonalt', ... (4 bytes))
1682369129: Received PUBLISH from a3f20e84-f6a0-43f5-8b11-5dd741fc0a99 (d0, q0, r1, m0, 'indi-allsky/moonphase', ... (4 bytes))
1682369129: Received PUBLISH from a3f20e84-f6a0-43f5-8b11-5dd741fc0a99 (d0, q0, r1, m0, 'indi-allsky/moonmode', ... (5 bytes))
1682369129: Received PUBLISH from a3f20e84-f6a0-43f5-8b11-5dd741fc0a99 (d0, q0, r1, m0, 'indi-allsky/night', ... (5 bytes))
1682369129: Received PUBLISH from a3f20e84-f6a0-43f5-8b11-5dd741fc0a99 (d0, q0, r1, m0, 'indi-allsky/sqm', ... (8 bytes))
1682369129: Received PUBLISH from a3f20e84-f6a0-43f5-8b11-5dd741fc0a99 (d0, q0, r1, m0, 'indi-allsky/stars', ... (1 bytes))
1682369129: Received PUBLISH from a3f20e84-f6a0-43f5-8b11-5dd741fc0a99 (d0, q0, r1, m0, 'indi-allsky/latitude', ... (4 bytes))
1682369129: Received PUBLISH from a3f20e84-f6a0-43f5-8b11-5dd741fc0a99 (d0, q0, r1, m0, 'indi-allsky/longitude', ... (5 bytes))
1682369129: Sending PUBLISH to mosqsub|21753-astroberr (d0, q0, r0, m0, 'indi-allsky/longitude', ... (5 bytes))
Client mosqsub|21753-astroberr received PUBLISH (d0, q0, r0, m0, 'indi-allsky/longitude', ... (5 bytes))
-84.0
1682369129: Received PUBLISH from a3f20e84-f6a0-43f5-8b11-5dd741fc0a99 (d0, q0, r1, m0, 'indi-allsky/sidereal_time', ... (10 bytes))
1682369129: Received PUBLISH from a3f20e84-f6a0-43f5-8b11-5dd741fc0a99 (d0, q0, r1, m0, 'indi-allsky/cpu/user', ... (3 bytes))
1682369129: Received PUBLISH from a3f20e84-f6a0-43f5-8b11-5dd741fc0a99 (d0, q0, r1, m0, 'indi-allsky/cpu/system', ... (3 bytes))
1682369129: Received PUBLISH from a3f20e84-f6a0-43f5-8b11-5dd741fc0a99 (d0, q0, r1, m0, 'indi-allsky/cpu/nice', ... (3 bytes))
1682369129: Received PUBLISH from a3f20e84-f6a0-43f5-8b11-5dd741fc0a99 (d0, q0, r1, m0, 'indi-allsky/cpu/iowait', ... (3 bytes))
1682369129: Received PUBLISH from a3f20e84-f6a0-43f5-8b11-5dd741fc0a99 (d0, q0, r1, m0, 'indi-allsky/cpu/total', ... (3 bytes))
1682369129: Received PUBLISH from a3f20e84-f6a0-43f5-8b11-5dd741fc0a99 (d0, q0, r1, m0, 'indi-allsky/memory/user', ... (3 bytes))
1682369129: Received PUBLISH from a3f20e84-f6a0-43f5-8b11-5dd741fc0a99 (d0, q0, r1, m0, 'indi-allsky/memory/cached', ... (4 bytes))
1682369129: Received PUBLISH from a3f20e84-f6a0-43f5-8b11-5dd741fc0a99 (d0, q0, r1, m0, 'indi-allsky/memory/total', ... (4 bytes))
1682369129: Received PUBLISH from a3f20e84-f6a0-43f5-8b11-5dd741fc0a99 (d0, q0, r1, m0, 'indi-allsky/disk/root', ... (4 bytes))
1682369129: Received PUBLISH from a3f20e84-f6a0-43f5-8b11-5dd741fc0a99 (d0, q0, r1, m0, 'indi-allsky/disk/boot', ... (4 bytes))
1682369129: Received PUBLISH from a3f20e84-f6a0-43f5-8b11-5dd741fc0a99 (d0, q0, r1, m0, 'indi-allsky/temp/cpu_thermal/0', ... (6 bytes))
1682369129: Received DISCONNECT from a3f20e84-f6a0-43f5-8b11-5dd741fc0a99
1682369129: Client a3f20e84-f6a0-43f5-8b11-5dd741fc0a99 disconnected.
^C1682369138: Socket error on client mosqsub|21753-astroberr, disconnecting.

aaronwmorris, 2023-04-24:
I see... interesting.  What distribution and version are you using?

ispybadguys, 2023-04-24:
I built it today because I was trying to solve this problem. Perhaps you were right and it was not running earlier when I tried the different configurations. I want to try and measure cloud cover so I can open the roof when it clears. I thought I would use the star count to estimate this. This is a great program Aaron!  

Kurt

aaronwmorris, 2023-04-25:
Sure thing, let me know if you figure it out.

closed 2023-07-26 by aaronwmorris