aaronwmorris/indi-allsky #713: add support for cloud and rain detection, notifications

uotw, 2023-04-25:
Hey guys, love this project. I'm building a RPi-based weather station with the RPi-HQ camera, small OLED screen, a rain sensor and a few temp sensors. I'm wanting to add some code to this project for cloud and rain detection. Would be nice to include these stats on the main dashboard page and sql logging as well. I can certainly write code for my specific hardware, but on initial inspection I had trouble finding the main program loop. Would love to contribute any way I can.

Was also planning on using Pushover to send me critical alerts when I need to haul my gear in (>80% cloud or first drop rain). Pushover is great. It allows web-based API calls to send alerts with different levels of panic, complete with text/links and pics. Would be fairly easy integration to include a config field for a user's API key and then using curl to send the alerts based on some trigger value.

Last, and perhaps more ambitious would be to include some ML to help teach someone's rig to detect the level of cloudiness. I need to play around with the Jupyter Notebook created for ML, but it seems like the perfect ML problem.

aaronwmorris, 2023-04-25:
indi-allsky utilizes python multiprocessing, so there are actually a few main program loops.

The main process starts in `indi_allsky/allsky.py` at the `run` method.  The process primarily manages the camera and takes exposures, along with a few other housekeeping activities.  It also starts the ImageWorker, VideoWorker, and UploaderWorker

There is the ImageWorker process in `indi_allsky/image.py` which processes the images from the camera.

The VideoWorker process in `indi_allsky/video.py` creates the timelapses, keograms, and startrails, and a few other activities.

The FileUploader process in `indi_allsky/uploader.py` performs file transfers.  There can be multiple file transfer workers running.

A combination of multiprocessing queues and the database are used for inter-process communication.

uotw, 2023-04-25:
Perfect, thanks for the leads here. I'll dive in and see what I can add. If it's decent I'll create pull request.