aaronwmorris/indi-allsky #715: Two possible enhancements?

vastmeridian, 2023-04-25:
Hi.

I have three possible enhancements I'd like to be considered:

1. A hook to an Aurora alert system that could be used to alert the watcher of a high probability of aurora activity overnight (in the UK we use Lancaster University's Aurora Watch UK - https://aurorawatch.lancs.ac.uk/ )
2. A moon mode that decreases the gain in proportion to the phase of the moon rather than a hard switch into 'Moon Mode'
3. The option of a fixed-space or monotype font in the image text drop-down 

This software is really developing very nicely now.  I've persuaded two others in my astronomy group to use it too. 

Thanks

aaronwmorris, 2023-04-25:
> This software is really developing very nicely now. I've persuaded two others in my astronomy group to use it too.

That is awesome!  I am glad people are liking the software.

> A hook to an Aurora alert system that could be used to alert the watcher of a high probability of aurora activity overnight (in the UK we use Lancaster University's Aurora Watch UK - https://aurorawatch.lancs.ac.uk/ )

I will see investigate this.

> A moon mode that decreases the gain in proportion to the phase of the moon rather than a hard switch into 'Moon Mode'

I will think about this.  I am kind of reticent to do this because it would mean more dark frames that have to be gathered.

> The option of a fixed-space or monotype font in the image text drop-down

I will consider this as a longer term goal.  Currently, I am using OpenCV to draw the labels on the image.  By default OpenCV only provides the 8 default fonts.  Unfortunately, not all Python wheels of OpenCV include the TrueType engine, so it is not always possible to load custom fonts.

Pillow might support TrueType fonts, but I would have to replace or extend the text labeling.

vastmeridian, 2023-04-26:
> I will think about this. I am kind of reticent to do this because it would mean more dark frames that have to be gathered.
Understood.  Perhaps use 'bands' of gain rather than a sliding scale?  I agree it needs thought.

aaronwmorris, 2023-04-26:
Merged #718 to support Pillow image labels using TrueType fonts.  Mono spaced fonts are included currently.

aaronwmorris, 2023-04-27:
Also merged #723 to permit loading a custom TrueType font.  indi-allsky includes some GPL licensed fonts and I can add more, but I have to ensure they are properly licensed and attributed.

Now you can just add your own.

vastmeridian, 2023-04-28:
Excellent.  Thank you.

I'm currently rebuilding my system to incorproate fan cooling for the processor board and upgrading the SSD to 128G.  All being well, the system should be back online tonight, so I'll try things out after I know it's stable.

Incidentally, my system with the Altair Hypercam 178C has been rock steady these past few weeks.  I don't know if you want to add this to the Wiki as a supported camera.  Happy to share the config I'm using to make things easier for other Altair 178C users. Shall I raise an issue with the code in as a 'new camera' feature?

aaronwmorris, 2023-04-28:
Awesome!  I will get it updated. This is the non GPCAM version, right?

vastmeridian, 2023-04-28:
The camera is a non-GPCam version.  It's the Altair Hypercam 178C

As I'm offline cos of the rebuild I can't add the camera config right now, but I'll do that shortly.

vastmeridian, 2023-04-28:
Altair Hypercam indi config:

```
{
    "PROPERTIES": {
        "CCD_CONTROLS": {
            "BandWidth": 40
        }
    },
    "SWITCHES": {
        "CCD_RESOLUTION": {
            "on": [
                "3040 x 2048"
            ]
        },
        "CCD_CAPTURE_FORMAT": {
            "on": [
                "INDI_RAW"
            ]
        }
    }
}
```

vastmeridian, 2023-04-30:
I'm back online, having transferred the old database so I am now in a position to start testing the TrueType fonts.  

Having switched from OpenCV to Pillow, the font didn't update for me with a simple 'reload on save' even after five minutes.  
So I tried again and after a few minutes I was informed that the camera may have hung.  I'm wondering if this was my setup, so I stopped and restarted the indi-allsky service.  This failed to deactivate, but the camera restarted (and the font system was incorporated) when indi-allsky started up again.

It's late over here, so I will carry out some better tests as soon as I can and capture some logs in case it happens again.

vastmeridian, 2023-05-01:
I've had a good go at tinkering this afternoon and I've switched from Pillow to OpenCV to Pillow; changed fonts; changed text; tinkered with settings and no instabilities.  Each config change was saved with the 'Reload on Save' enabled.  All good.  No instability.
The only tiny comment is that the Y offset seems to work differently between OpenCV and Pillow.   I wonder if Pillow's start point is the top left of the character and the OpenCV start point is the lower left?

OpenCV example text:
[image]

Pillow example text:
[image]

Pillow italic example text:
[image]

And finally the settings used for the setup:
[image]

vastmeridian, 2023-05-01:
Last step was to try a custom TTF font.  I downloaded Mistral.ttf from ttfonts.net.

[image]

All good. No instabilities noted.  Thank you.

aaronwmorris, 2023-05-01:
Looks good.  I did notice the offset for Pillow is different, but you can fix that with the offset settings in the config.  You can set the X offset to 30 (default 15) and it is corrected.

It is kind of subtle, but I separated the OpenCV and Pillow settings from the generic Text offset settings.

The text of keograms should also work.  The offsets are hard coded there, but the result should be relatively similar.

vastmeridian, 2023-05-01:
`Looks good. I did notice the offset for Pillow is different, but you can fix that with the offset settings in the config. You can set the X offset to 30 (default 15) and it is corrected.`
Yeah - it wasn't a criticism, just a comment.  
I'm busy coding a quick script to add weather info to the image from OpenWeather's API now.

vastmeridian, 2023-07-04:
Of the three enhancements I raised:
- The Aurora alert system is still being considered
- The moon mode enhancement could be merged with #689
- The enhanced font functionality is done.  

How do you want to progress this?  I can happily close this issue and raise a new one specific for the Aurora alert system and add my two cents worth to #689

aaronwmorris, 2023-07-08:
Merged #826 which includes Aurora and Kp-index data from NOAA.

vastmeridian, 2023-07-08:
Wow!  You've been busy!  I'll update today and provide some feedback. 👌

vastmeridian, 2023-07-08:
That's fantastic.  Thank you.
Is the aurora information location-aware (in terms of the likelyhood over the UK) or is it a general figure?

aaronwmorris, 2023-07-08:
The Kp-index is global, but the Aurora percentage is location specific.  The NOAA Ovation data has values for every latitude/longitude combination, with a resolution of 1 degree.  indi-allsky will use data from a 7x7 degree grid around your location (~500 mile radius) and create an aggregate score.

aaronwmorris, 2023-07-08:
A little more context on the Ovation model.  The percentage for every latitude/longitude pair is the chance that that the aurora will be visible directly overhead (90 degrees altitude) at that location.  My assumption is if aurora is directly overhead 500 miles away (in any direction), it should be visible to you, although maybe closer to the horizon.

vastmeridian, 2023-07-08:
That makes a lot of sense.  Might be worth adding to the MQTT export so that one can trigger an alert from it.

This site is thirty minutes away (and in southern England), so we need a warning to be able to get there in good time.  Earlier this year we had some rare visible pink aurora this far south.  It was all over in under two hours  It's a friend's picture, but you can see why we need the alerts:
[image]

vastmeridian, 2023-07-08:
If I move my moon enhancement request as I suggested above, I can then close this issue as complete.

Thank you.

aaronwmorris, 2023-07-08:
Merged #828 for adding aurora data to mqtt data.

vastmeridian, 2023-07-08:
You don't hang around, do you?? 
OK - updating now.  Thanks

vastmeridian, 2023-07-08:
... and Aurora now configured on my MQTT dashboard to read indi-allsky/ovation_max following your updated wiki page.

aaronwmorris, 2023-07-08:
I really do not like sitting on good ideas.  :-)

Vertikar, 2023-07-08:
> That makes a lot of sense.  Might be worth adding to the MQTT export so that one can trigger an alert from it.
> 
> This site is thirty minutes away (and in southern England), so we need a warning to be able to get there in good time.  Earlier this year we had some rare visible pink aurora this far south and a friend was able to take an excellent image.  It was all over in under two hours so you can see why we need the alerts:

If you want some really good aurora alerting, this website/app is one of my favourites:

Glendale App is not in any app store. Only available from the website. The world's most accurate aurora app. https://aurora-alerts.uk/

vastmeridian, 2023-07-08:
> If you want some really good aurora alerting, this website/app is one of my favourites:
> 
> https://aurora-alerts.uk/ Its built as a PWA though, so its not in the appstore, just need to install it from the website if you want alerting

I use aurora watch as I stated above, but the beauty of Aaron’s approach is that it’s global in reach and local in accuracy - and easily configurable in an external app (such as Home Assistant) to provide a wider range of notifications. 

It would also be interesting to compare the results from all the aurora apps over time too.

vastmeridian, 2023-09-03:
Seeing what you say regarding moon mode, I'l close this issue now as resolved and we'll see where the future takes us.  Thanks

closed 2023-09-03 by vastmeridian