aaronwmorris/indi-allsky #717: indi-allsky service keeps failing, status=UNKNOWN

milesnz, 2023-04-26:
Hi, from a newbie to both this forum and to sky photography.

I've very recently installed indi-allsky but my indi-allsky service keeps failing a short time after it gets started.
The web portal shows that the status is UNKNOWN
When checking the camera-info link it returns this error:
"The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application."

The camera is not being used by any other app at the time but I did recheck the libcamera commands and it is still working fine.

I also reran setup.sh and tried indi camera instead of libcamera - to no avail.
I also ran ./misc/rebuild_pyindi-client.sh and selected 1.9.7 but I think it went back to 1.9.1 ??

Any assistance would be great please.

 

Raspberrypi 4b 1.1
Raspbian Linux raspberrypi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
Arducam libcamera_64mp_Hawkeye (works great with libcamera commands)
I've also tried with a Raspberry camera v3 

 

Apr 25 18:17:53 raspberrypi [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/indi-allsky/indi_allsky/camera/indi.py", line 1199, in get_control\n    raise TimeOutException(\'Timeout finding control {0}\'.format(name))\n'
Apr 25 18:17:53 raspberrypi [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/indi-allsky/indi_allsky/camera/indi.py", line 1208, in set_number\n    c = self.get_control(device, name, \'number\')\n'
Apr 25 18:17:53 raspberrypi [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/indi-allsky/indi_allsky/camera/indi.py", line 696, in configureDevice\n    self.set_number(device, k, v)\n'
Apr 25 18:17:53 raspberrypi [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/indi-allsky/indi_allsky/camera/indi.py", line 737, in configureGpsDevice\n    self.configureDevice(self._gps_device, *args, **kwargs)\n'
Apr 25 18:17:53 raspberrypi [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/indi-allsky/indi_allsky/allsky.py", line 516, in _initialize\n    self.indiclient.configureGpsDevice(gps_config)\n'
Apr 25 18:17:53 raspberrypi [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/indi-allsky/indi_allsky/allsky.py", line 1019, in run\n    self._initialize()\n'
Apr 25 18:17:53 raspberrypi [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/indi-allsky/allsky.py", line 119, in <module>\n    action_func(*args_list, **kwargs_dict)\n'
Apr 25 18:17:53 raspberrypi [ERROR] MainProcess allsky.unhandled_exception() #36: Value: Timeout finding control GPS_REFRESH_PERIOD
Apr 25 18:17:53 raspberrypi [ERROR] MainProcess allsky.unhandled_exception() #35: Type: <class 'indi_allsky.exceptions.TimeOutException'>
Apr 25 18:17:53 raspberrypi [ERROR] MainProcess allsky.unhandled_exception() #34: An uncaught exception occurred:
Apr 25 18:17:43 raspberrypi [INFO] MainProcess indi.configureDevice() #695: Setting property (number) GPS_REFRESH_PERIOD
Apr 25 18:17:43 raspberrypi [INFO] MainProcess indi.newMessage() #324: new Message 2023-04-25T06:17:43: [WARNING] No GPSD running.
Apr 25 18:17:43 raspberrypi [WARNING] MainProcess allsky._initialize() #490: Connecting to GPS device GPSD
Apr 25 18:17:43 raspberrypi [WARNING] MainProcess allsky._initialize() #486: Connecting to Telescope device Telescope Simulator
Apr 25 18:17:43 raspberrypi [WARNING] MainProcess allsky._initialize() #482: Connecting to CCD device libcamera_64mp_hawkeye
Apr 25 18:17:43 raspberrypi [INFO] MainProcess indi.findGps() #662: Found 1 GPSs
Apr 25 18:17:43 raspberrypi [INFO] MainProcess indi._findGpss() #653:  Detected GPSD
Apr 25 18:17:43 raspberrypi [INFO] MainProcess indi._findGpss() #642: Searching for available GPS interfaces
Apr 25 18:17:43 raspberrypi [INFO] MainProcess indi.findTelescope() #629: Found 1 Telescopess
Apr 25 18:17:43 raspberrypi [INFO] MainProcess indi._findTelescopes() #620:  Detected Telescope Simulator
Apr 25 18:17:43 raspberrypi [INFO] MainProcess indi._findTelescopes() #609: Searching for available telescopes/mounts
Apr 25 18:17:35 raspberrypi [INFO] MainProcess indi.newDevice() #197: new device Telescope Simulator
Apr 25 18:17:35 raspberrypi [INFO] MainProcess indi.newDevice() #197: new device GPSD
Apr 25 18:17:35 raspberrypi [INFO] MainProcess indi.newDevice() #197: new device CCD Simulator
Apr 25 18:17:35 raspberrypi [INFO] MainProcess indi.serverConnected() #327: Server connected (localhost:7624)
Apr 25 18:17:35 raspberrypi [INFO] MainProcess allsky._initialize() #444: Connecting to indiserver
Apr 25 18:17:35 raspberrypi [INFO] MainProcess indi.__init__() #133: PyIndi version: 1.9.7
Apr 25 18:17:35 raspberrypi [INFO] MainProcess indi.__init__() #126: creating an instance of IndiClient

 

 

Additional info:

When I use URL  "https://localhost/indi-allsky/"  (with the trailing '/' then I get an error:

Internal Server Error
The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application.)
[indi-allsky.log.txt](https://github.com/aaronwmorris/indi-allsky/files/11331314/indi-allsky.log.txt)

 

  but when I remove the trailing '/' ie, URL "https://localhost/indi-allsky" then the indi-allsky web app opens.  
  I don't know if that's important or not.
[indi-allsky.log.txt](https://github.com/aaronwmorris/indi-allsky/files/11331319/indi-allsky.log.txt)
[webapp-indi-allsky.log.txt](https://github.com/aaronwmorris/indi-allsky/files/11331320/webapp-indi-allsky.log.txt)

aaronwmorris, 2023-04-26:
I think the problem is you selected a GPS device.  Try rerunning setup.sh and select "None" for the GPS device.

If you do have a GPS device, you will need to run `./misc/setup_gpsd.sh` and make sure GPSd is properly configured.

milesnz, 2023-04-26:
awesome, thank you - problem resolved

closed 2023-04-30 by aaronwmorris