aaronwmorris/indi-allsky #722: all images are black

milesnz, 2023-04-27:
Hi,
Now that my server is RUNNING - after your good advice - I need some more advice please.
I think my camera config might be wrong as my images are black.
The driver below concerns me - maybe I've done something wrong somewhere?

Name libcamera_64mp_hawkeye
Driver indi_fake_ccd
Resolution 9152 x 6944 (11488 diag)
Dimensions 7.3mm x 5.6mm (9.2mm diag)
Pixels 0.80µm
CFA RGGB
Bits 16
Gain 1-22
Exposure 0.001000-200s

aaronwmorris, 2023-04-27:
Try setting a default exposure of 1.0s and restarting.  Also, pull down the latest code, I recently fixed a bug that might be affecting you.

The driver is correct.  indi-allsky was designed to operate with INDI.  I designed an interface that emulates the behavior of the indi interface, but it is running the libcamera-still command instead.  I do probably need to change that driver name.

aaronwmorris, 2023-04-27:
Merged #724 which changes the driver name for libcamera cameras.

milesnz, 2023-04-28:
Hi,
I've set the exposure to 1, and then to 5 - no difference seen.
I also applied the latest git pull, thanks for updating the driver name now.
I'm still getting no images but I've noticed imageworker errors and when I watched the logs I noticed these errors about the database - database disk image is malformed.

Also, do I need to set anything here under the INDI Camera Configuration?
{
    "SWITCHES": {},
    "PROPERTIES": {},
    "TEXT": {}
}

Apr 28 15:48:19 raspberrypi [INFO] MainProcess allsky._startImageWorker() #775: Starting ImageWorker process 19
Apr 28 15:48:19 raspberrypi [ERROR] MainProcess allsky._startImageWorker() #768: Image worker exception:
Apr 28 15:48:19 raspberrypi [ERROR] MainProcess allsky._startImageWorker() #768: Image worker exception: (Background on this error at: https://sqlalche.me/e/20/4xp6)
Apr 28 15:48:19 raspberrypi [ERROR] MainProcess allsky._startImageWorker() #768: Image worker exception: [parameters: ('/var/www/html/allsky/images/ccd_2b68749a-5017-425c-b088-c2a8fea55b23/20230428/day/28_15/ccd1_20230428_154811.jpg', None, None, '2023-04-28 15:48:11.711963', '2023-04-28', 1.0, 14.609703540802002, 2.020246982574463, 1, 1, -273.1499938964844, 0, 65.10917580626734, 1, 0, None, 0, 2214.3979003072136, 0, None, 0, 0, 1)]
Apr 28 15:48:19 raspberrypi [ERROR] MainProcess allsky._startImageWorker() #768: Image worker exception: [SQL: INSERT INTO image (filename, remote_url, s3_key, "createDate", "dayDate", exposure, exp_elapsed, process_elapsed, gain, binmode, "temp", night, adu, stable, moonmode, moonphase, adu_roi, sqm, stars, sync_id, calibrated, detections, camera_id) VALUES (?, ?, ?, ?, ?, ?, ?, ?, ?, ?, ?, ?, ?, ?, ?, ?, ?, ?, ?, ?, ?, ?, ?)]
Apr 28 15:48:19 raspberrypi [ERROR] MainProcess allsky._startImageWorker() #768: Image worker exception: sqlalchemy.exc.DatabaseError: (sqlite3.DatabaseError) database disk image is malformed
Apr 28 15:48:19 raspberrypi [ERROR] MainProcess allsky._startImageWorker() #768: Image worker exception:     cursor.execute(statement, parameters)
Apr 28 15:48:19 raspberrypi [ERROR] MainProcess allsky._startImageWorker() #768: Image worker exception:   File "/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/sqlalchemy/engine/default.py", line 918, in do_execute
Apr 28 15:48:19 raspberrypi [ERROR] MainProcess allsky._startImageWorker() #768: Image worker exception:     self.dialect.do_execute(
Apr 28 15:48:19 raspberrypi [ERROR] MainProcess allsky._startImageWorker() #768: Image worker exception:   File "/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/sqlalchemy/engine/base.py", line 1963, in _exec_single_context
Apr 28 15:48:19 raspberrypi [ERROR] MainProcess allsky._startImageWorker() #768: Image worker exception:     raise sqlalchemy_exception.with_traceback(exc_info[2]) from e
Apr 28 15:48:19 raspberrypi [ERROR] MainProcess allsky._startImageWorker() #768: Image worker exception:   File "/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/sqlalchemy/engine/base.py", line 2339, in _handle_dbapi_exception
Apr 28 15:48:19 raspberrypi [ERROR] MainProcess allsky._startImageWorker() #768: Image worker exception:     self._handle_dbapi_exception(
Apr 28 15:48:19 raspberrypi [ERROR] MainProcess allsky._startImageWorker() #768: Image worker exception:   File "/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/sqlalchemy/engine/base.py", line 1982, in _exec_single_context
Apr 28 15:48:19 raspberrypi [ERROR] MainProcess allsky._startImageWorker() #768: Image worker exception:     return self._exec_single_context(
Apr 28 15:48:19 raspberrypi [ERROR] MainProcess allsky._startImageWorker() #768: Image worker exception:   File "/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/sqlalchemy/engine/base.py", line 1841, in _execute_context
Apr 28 15:48:19 raspberrypi [ERROR] MainProcess allsky._startImageWorker() #768: Image worker exception:     ret = self._execute_context(
Apr 28 15:48:19 raspberrypi [ERROR] MainProcess allsky._startImageWorker() #768: Image worker exception:   File "/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/sqlalchemy/engine/base.py", line 1637, in _execute_clauseelement
Apr 28 15:48:19 raspberrypi [ERROR] MainProcess allsky._startImageWorker() #768: Image worker exception:     return connection._execute_clauseelement(
Apr 28 15:48:19 raspberrypi [ERROR] MainProcess allsky._startImageWorker() #768: Image worker exception:   File "/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/sqlalchemy/sql/elements.py", line 483, in _execute_on_connection

aaronwmorris, 2023-04-28:
The database is corrupted.  See if an integrity check will fix it....

```
sqlite3 /var/lib/indi-allsky/indi-allsky.sqlite "PRAGMA integrity_check;"
```

milesnz, 2023-04-28:
Hi,
***@***.***:/indi-allsky $ sqlite3
/var/lib/indi-allsky/indi-allsky.sqlite "PRAGMA integrity_check;"
*** in database main ***
Page 141: btreeInitPage() returns error code 11
Page 140: btreeInitPage() returns error code 11
Page 139: btreeInitPage() returns error code 11
Error: database disk image is malformed

Is there a simple way of deleting and rebuilding the database, or should I
wipe and rebuild the whole system now that I understand it all a bit better?

regards

aaronwmorris, 2023-04-28:
These are the instructions for rebuilding the database.

https://github.com/aaronwmorris/indi-allsky/wiki/Rebuilding-the-database

You may want also pull down the latest code, I merged #725 to hopefully fix this.  There is an entire history behind this that is probably too much to go into here.

milesnz, 2023-05-01:
Hi,
I figured if the database was corrupt then other things might also be
damaged so I rebuilt the OS completely and it now works.
thank you

closed 2023-07-25 by aaronwmorris