aaronwmorris/indi-allsky #726: Problems running usertools.py

vastmeridian, 2023-04-28:
Running usertools.py from the command line gives me the following error:

`user@allsky:~/indi-allsky/misc$ ./usertool.py deleteuser -u viewer
Traceback (most recent call last):
  File "./usertool.py", line 8, in <module>
    from passlib.hash import argon2
ModuleNotFoundError: No module named 'passlib'
user@allsky:~/indi-allsky/misc$
`
Looks like I'm going to have to start from scratch.

aaronwmorris, 2023-04-28:
You just need to first activate the virtualenv...

```
source virtualenv/indi-allsky/bin/activate

./usertool.py deleteuser -u viewer
```

vastmeridian, 2023-04-28:
Thank you.  Issue closed.
(Maybe a mention on the wiki - or did I miss it?)

closed 2023-04-28 by vastmeridian