aaronwmorris/indi-allsky #727: setup.py fails at python virtual env setup of scikit-image

uotw, 2023-04-30:
Hey, just trying to set this up on RPi 3B+ running 32-bit bullseye. Compiled indi-allsky without issue. `setup.py` fails at python virtual env setup of scikit-image with the following error code. 

>  Collecting scikit-image (from ccdproc->-r /home/ben/indi-allsky/requirements_debian11_32.txt (line 11))
>   Using cached scikit_image-0.20.0.tar.gz (22.9 MB)
>   Installing build dependencies ... done
>   Getting requirements to build wheel ... done
>   Installing backend dependencies ... done
>   Preparing metadata (pyproject.toml) ... error
>   error: subprocess-exited-with-error
>   
>   × Preparing metadata (pyproject.toml) did not run successfully.
>   │ exit code: 1
>   ╰─> [26 lines of output]
>       + meson setup /tmp/pip-install-a_mn_x5l/scikit-image_239720f080ff4a62960b278d2db5bbf0 /tmp/pip-install-a_mn_x5l/scikit-image_239720f080ff4a62960b278d2db5bbf0/.mesonpy-tnulso0a/build -Dbuildtype=release -Db_ndebug=if-release -Db_vscrt=md --native-file=/tmp/pip-install-a_mn_x5l/scikit-image_239720f080ff4a62960b278d2db5bbf0/.mesonpy-tnulso0a/build/meson-python-native-file.ini
>       The Meson build system
>       Version: 1.1.0
>       Source dir: /tmp/pip-install-a_mn_x5l/scikit-image_239720f080ff4a62960b278d2db5bbf0
>       Build dir: /tmp/pip-install-a_mn_x5l/scikit-image_239720f080ff4a62960b278d2db5bbf0/.mesonpy-tnulso0a/build
>       Build type: native build
>       Project name: scikit-image
>       Project version: 0.20.0
>       C compiler for the host machine: cc (gcc 10.2.1 "cc (Raspbian 10.2.1-6+rpi1) 10.2.1 20210110")
>       C linker for the host machine: cc ld.bfd 2.35.2
>       C++ compiler for the host machine: c++ (gcc 10.2.1 "c++ (Raspbian 10.2.1-6+rpi1) 10.2.1 20210110")
>       C++ linker for the host machine: c++ ld.bfd 2.35.2
>       Host machine cpu family: arm
>       Host machine cpu: armv7l
>       Compiler for C supports arguments -Wno-unused-function: YES
>       Library m found: YES
>       Checking if "-Wl,--version-script" : links: YES
>       Program cython found: YES (/tmp/pip-build-env-ge5fvc5e/overlay/bin/cython)
>       Program pythran found: YES (/tmp/pip-build-env-ge5fvc5e/overlay/bin/pythran)
>       Program cp found: YES (/usr/bin/cp)
>       Program python found: YES (/home/ben/indi-allsky/virtualenv/indi-allsky/bin/python)
>       Found pkg-config: /usr/bin/pkg-config (0.29.2)
>       
>       ../../skimage/meson.build:54:17: ERROR: Command `/home/ben/indi-allsky/virtualenv/indi-allsky/bin/python -c 'import os; os.chdir(".."); import pythran; print(os.path.dirname(pythran.__file__));'` failed with status 1.
>       
>       A full log can be found at /tmp/pip-install-a_mn_x5l/scikit-image_239720f080ff4a62960b278d2db5bbf0/.mesonpy-tnulso0a/build/meson-logs/meson-log.txt
>       [end of output]
>   
>   note: This error originates from a subprocess, and is likely not a problem with pip.
> error: metadata-generation-failed
> 
> × Encountered error while generating package metadata.
> ╰─> See above for output.
> 
> note: This is an issue with the package mentioned above, not pip.
> hint: See above for details.
> 
> 
> The script exited abnormally, please try to run again...
>

aaronwmorris, 2023-04-30:
I think the build script for scikit-image is not including the pythran module.

Merged #729 to lock scikit-image to 0.19.3 which is the last available wheel.

uotw, 2023-04-30:
Boom, fixed.  Thanks.

closed 2023-05-10 by aaronwmorris