aaronwmorris/indi-allsky #728: Enhancement Request - Backup/Restore of settings

vastmeridian, 2023-04-30:
Having moved to a new (larger) ssd, I had to go through the pain of a database migration from an earlier build of indi-allsky to a newer build just to transfer all the settings from one instance of indi-alllsky to another.

```
**** Setup DB ****
**** Backup DB prior to migration ****
ERROR [flask_migrate] Error: Can't locate revision identified by 'ab5f1e700820'

The script exited abnormally, please try to run again...
```
Is there a way to add a indi-allsky settings backup or export to (say) the Utilities page that could then be used as an import on the new instance?  Perhaps in a json format or similar?

aaronwmorris, 2023-04-30:
These are the flask database migrations.

You can try these steps from https://github.com/aaronwmorris/indi-allsky/wiki/Database-migrations#fixing-broken-migrations

```
sqlite3 /var/lib/indi-allsky/indi-allsky.sqlite "DELETE FROM alembic_version;"

# from the indi-allsky folder
rm /var/lib/indi-allsky/migrations/versions/*.py

# now rerun setup.sh
```

vastmeridian, 2023-04-30:
I got to the same point in a slightly different way.    I reconnected the old disk, updated indi-allsky and re-run the export.  That has now worked and I'm up and running.  In retrospect, your method would have been slightly quicker. :)

Nevertheless, if you're interested, my request for the enhancement is still a valid one (he asked hopefully!)

aaronwmorris, 2023-04-30:
Your method was going to be my next suggestion, which is the best way.

I will write up a backup/restore procedure.

vastmeridian, 2023-05-03:
I did run into the same **Error: Can't locate revision** error when I next updated the new system from the GIT master - even after updating using 'my method', so I still had to use the `rm /var/lib/indi-allsky/migrations/versions/*.py` to recover the system.

aaronwmorris, 2023-05-03:
It is only a problem if the last database schema does not reflect what is in the DB.  If you upgraded before moving systems, you should be fine.

vastmeridian, 2023-06-01:
Hi Aaron
I'm just being tidy.  Have you written the backup/restore procedure. mentioned above?  If so, I'll test then close this issue.

aaronwmorris, 2023-06-06:
Not yet.  I will get to it very soon.

aaronwmorris, 2023-06-27:
This is my first pass at an informational page on backing up indi-allsky:

https://github.com/aaronwmorris/indi-allsky/wiki/Backup-and-Recovery

I am also planning on writing a script to create a unified tar file of all of the files necessary to backup.

vastmeridian, 2023-07-04:
That new page looks good, Aaron.  Script would be good (and I'd be happy to test - at least to a VM and back)
One (small) point:

> Backup and recovery of indi-allsky involves backing up 4 areas
> 
>     Database
>     Database migrations
>     Flask config
>     indi-allsky environment (optional)
>     Images and videos

I count 5 areas, not 4  (OK - so one is optional) :)

closed 2024-09-14 by aaronwmorris