aaronwmorris/indi-allsky #731: blank image raspbian 32-bit + RPi HQ camera

uotw, 2023-05-01:
Hey, I've just gotten this up and running and I'm having trouble getting the camera to work.  Using raspbian 32-bit + RPi HQ camera. First I attempted to use indi-allsky with legacy camera support, and that threw an error in the log with no image rendered. Next I disabled legacy support, and added these lines to my config.txt:

```
camera_auto_detect=1
dtoverlay=imx477
gpu_mem=128
```

indi-allsky configured to use libcamera IMX477.

What I'm seeing now is blank (black) images.

Here's the log:

> Apr 30 21:14:04 raspberrypi [INFO] MainProcess allsky._queueManualTasks() #2426: Checking for manually submitted tasks
> Apr 30 21:14:04 raspberrypi [INFO] MainProcess indi.getTelescopeRaDec() #833: Telescope Coord: RA 10.22, Dec 33.00
> Apr 30 21:14:04 raspberrypi [INFO] MainProcess allsky.detectMoonMode() #1608: Moon altitude: 62:27:56.8, phase 78.6%
> Apr 30 21:14:04 raspberrypi [INFO] MainProcess allsky.detectNight() #1591: Sun altitude: -11:49:04.6
> Apr 30 21:14:04 raspberrypi [INFO] MainProcess allsky.run() #1043: Exposure state: BUSY
> Apr 30 21:14:04 raspberrypi [INFO] MainProcess allsky.run() #1042: Camera last ready: 14.2s
> Apr 30 21:13:59 raspberrypi [INFO] MainProcess allsky.run() #1276: Total time since last exposure 15.0440 s
> Apr 30 21:13:59 raspberrypi [INFO] MainProcess libcamera.setCcdExposure() #183: image command: libcamera-still --immediate --nopreview --raw --denoise off --awbgains 1,1 --gain 22 --shutter 500000 --metadata /tmp/tmp4aosr27v.json --metadata-format json --output /tmp/tmp7ac4mj12.dng
> Apr 30 21:13:59 raspberrypi [INFO] MainProcess allsky.shoot() #1828: Taking 0.50000000 s exposure (gain 22)
> Apr 30 21:13:53 raspberrypi [INFO] MainProcess allsky._queueManualTasks() #2426: Checking for manually submitted tasks
> Apr 30 21:13:53 raspberrypi [INFO] MainProcess indi.getTelescopeRaDec() #833: Telescope Coord: RA 10.21, Dec 33.00
> Apr 30 21:13:53 raspberrypi [INFO] MainProcess allsky.detectMoonMode() #1608: Moon altitude: 62:26:42.9, phase 78.6%
> Apr 30 21:13:53 raspberrypi [INFO] MainProcess allsky.detectNight() #1591: Sun altitude: -11:46:59.0
> Apr 30 21:13:53 raspberrypi [WARNING] MainProcess image._load_detection_mask() #1507: No detection mask defined
> Apr 30 21:13:53 raspberrypi [INFO] MainProcess allsky._startImageWorker() #775: Starting ImageWorker process 31
> Apr 30 21:13:53 raspberrypi [ERROR] MainProcess allsky._startImageWorker() #768: Image worker exception: 
> Apr 30 21:13:53 raspberrypi [ERROR] MainProcess allsky._startImageWorker() #768: Image worker exception: Exception: *** rawpy module not available ***
> Apr 30 21:13:53 raspberrypi [ERROR] MainProcess allsky._startImageWorker() #768: Image worker exception:     raise Exception('*** rawpy module not available ***')
> Apr 30 21:13:53 raspberrypi [ERROR] MainProcess allsky._startImageWorker() #768: Image worker exception:   File "/home/ben/indi-allsky/indi_allsky/image.py", line 1750, in add
> Apr 30 21:13:53 raspberrypi [ERROR] MainProcess allsky._startImageWorker() #768: Image worker exception:     self.image_processor.add(filename_p, exposure, exp_date, exp_elapsed, camera)
> Apr 30 21:13:53 raspberrypi [ERROR] MainProcess allsky._startImageWorker() #768: Image worker exception:   File "/home/ben/indi-allsky/indi_allsky/image.py", line 328, in processImage
> Apr 30 21:13:53 raspberrypi [ERROR] MainProcess allsky._startImageWorker() #768: Image worker exception:     self.processImage(i_dict)
> Apr 30 21:13:53 raspberrypi [ERROR] MainProcess allsky._startImageWorker() #768: Image worker exception:   File "/home/ben/indi-allsky/indi_allsky/image.py", line 258, in saferun
> Apr 30 21:13:53 raspberrypi [ERROR] MainProcess allsky._startImageWorker() #768: Image worker exception:     self.saferun()
> Apr 30 21:13:53 raspberrypi [ERROR] MainProcess allsky._startImageWorker() #768: Image worker exception:   File "/home/ben/indi-allsky/indi_allsky/image.py", line 229, in run
> Apr 30 21:13:53 raspberrypi [ERROR] MainProcess allsky._startImageWorker() #768: Image worker exception: Traceback (most recent call last):
> Apr 30 21:13:53 raspberrypi [INFO] MainProcess allsky.run() #1043: Exposure state: READY
> Apr 30 21:13:53 raspberrypi [INFO] MainProcess allsky.run() #1042: Camera last ready: 3.0s

uotw, 2023-05-01:
Update: when I change to the INDI Camera Interface I get the same result, even after turning up the gain and exposure time.  Here's an example.

[image]

And here's the log

> May  1 15:26:39 raspberrypi [INFO] MainProcess allsky._queueManualTasks() #2426: Checking for manually submitted tasks
> May  1 15:26:39 raspberrypi [INFO] MainProcess indi.getTelescopeRaDec() #833: Telescope Coord: RA 4.48, Dec 33.00
> May  1 15:26:39 raspberrypi [INFO] MainProcess indi.getCcdTemperature() #848: Sensor temperature: 0.0
> May  1 15:26:39 raspberrypi [INFO] MainProcess allsky.detectMoonMode() #1608: Moon altitude: -13:49:01.6, phase 84.3%
> May  1 15:26:39 raspberrypi [INFO] MainProcess allsky.detectNight() #1591: Sun altitude: 58:43:52.0
> May  1 15:26:39 raspberrypi [INFO] MainProcess allsky.run() #1043: Exposure state: BUSY
> May  1 15:26:39 raspberrypi [INFO] MainProcess allsky.run() #1042: Camera last ready: 3.0s
> May  1 15:26:36 raspberrypi [INFO] ImageWorker001 miscDb.addImage() #171: Adding image /var/www/html/allsky/images/ccd_ac31317b-9524-437d-af7c-88f2e3c375a5/20230501/day/01_15/ccd2_20230501_152636.jpg to DB
> May  1 15:26:36 raspberrypi [INFO] ImageWorker001 image.write_img() #1236: Image compressed in 0.0836 s
> May  1 15:26:36 raspberrypi [INFO] MainProcess indi.newMessage() #324: new Message 2023-05-01T19:26:36: [ERROR] Got no stars, is gsc installed with appropriate environment variables set ?? 
> May  1 15:26:36 raspberrypi [INFO] ImageWorker001 image.processImage() #550: Image processed in 0.1235 s
> May  1 15:26:36 raspberrypi [INFO] MainProcess allsky.run() #1276: Total time since last exposure 15.4769 s
> May  1 15:26:36 raspberrypi [INFO] MainProcess indi.set_number() #1230: Setting Duration (s) = 15.0
> May  1 15:26:36 raspberrypi [WARNING] ImageWorker001 image.recalculate_exposure() #1464: New calculated exposure: 15.000000
> May  1 15:26:36 raspberrypi [INFO] ImageWorker001 image.calculate_histogram() #1368: Brightness average: 4.51
> May  1 15:26:36 raspberrypi [INFO] ImageWorker001 image._convert_16bit_to_8bit() #2268: Resampling image from 16 to 8 bits
> May  1 15:26:36 raspberrypi [INFO] ImageWorker001 image.processImage() #352: Image: 1280 x 1024
> May  1 15:26:36 raspberrypi [ERROR] ImageWorker001 image.debayer() #2203: No bayer pattern detected
> May  1 15:26:36 raspberrypi [INFO] ImageWorker001 sqm.calculate() #42: Weighted SQM average: 27.01
> May  1 15:26:36 raspberrypi [INFO] ImageWorker001 sqm.calculate() #37: Raw SQM average: 4.50
> May  1 15:26:36 raspberrypi [INFO] ImageWorker001 sqm.calculate() #21: Exposure: 15.000000, gain: 50
> May  1 15:26:36 raspberrypi [WARNING] ImageWorker001 image._calibrate() #2031: Dark not found: ccd2 16bit 15.0000000s gain 50 bin 1 0.00c
> May  1 15:26:36 raspberrypi [WARNING] ImageWorker001 image._calibrate() #2013: Temperature matched dark not found: 0.00c
> May  1 15:26:36 raspberrypi [INFO] ImageWorker001 image._calibrate() #1996: Searching for dark frame: gain 50, exposure >= 15.0, temp >= 0.0c
> May  1 15:26:36 raspberrypi [WARNING] ImageWorker001 image._calibrate() #1984: Bad Pixel Map not found: ccd2 16bit 15.0000000s gain 50 bin 1 0.00c
> May  1 15:26:36 raspberrypi [WARNING] ImageWorker001 image._calibrate() #1966: Temperature matched bad pixel map not found: 0.00c
> May  1 15:26:36 raspberrypi [INFO] ImageWorker001 image._calibrate() #1949: Searching for bad pixel map: gain 50, exposure >= 15.0, temp >= 0.0c
> May  1 15:26:36 raspberrypi [INFO] ImageWorker001 image._detectBitDepth() #1884: Image max value: 9
> May  1 15:26:36 raspberrypi [INFO] ImageWorker001 image.add() #1831: Image bits: 16, cfa: None
> May  1 15:26:36 raspberrypi [INFO] MainProcess allsky.shoot() #1828: Taking 15.00000000 s exposure (gain 50)
> May  1 15:26:36 raspberrypi [INFO] MainProcess allsky.run() #1184: Exposure received in 15.4769 s (0.4769)
> May  1 15:26:28 raspberrypi [INFO] MainProcess allsky._queueManualTasks() #2426: Checking for manually submitted tasks

aaronwmorris, 2023-05-01:
By switching to INDI, you are using the indi_ccd_simulator.  If you do not have the gsc package installed, it cannot display any simulated stars, so the black screen is expected.

I will do some investigation about the rawpy module on the 32-bit platform.

uotw, 2023-05-01:
Thanks, that does help. So it looks like the INDI driver for libcamera (indi_pylibcamera) isn't being used. I see on the indi_pylibcamera page they recommend starting the indi server with `indiserver -v ./indi_pylibcamer.py` Does this get automatically added by indi-allsky?

aaronwmorris, 2023-05-01:
You can run the install.sh script in indi_pylibcamera and it will install the server in a standard place.

Then you can run `indiserver -v indi_pylibcamera` (without the py).

aaronwmorris, 2023-05-01:
Yes, you can stop it with `systemctl --user stop indiserver`.

You should be able to re-run the indi-allsky setup.sh and setup the indi_pylibcamera server from there as well.

uotw, 2023-05-02:
Thanks. Works well like this.

aaronwmorris, 2023-05-04:
I have fixed the rawpy issue for Debian 11 32-bit in #737 and #738

uotw, 2023-05-06:
Thanks. Is there any advantage to using the libcamera driver vs the indi_pylibcamera driver? Now that I have it working with indi_pylibcamera I'm a afraid I'll break it by switching.

aaronwmorris, 2023-05-06:
I think the only advantage to using libcamera is you can disable the Defective Pixel Correction, ie the Sony star-eater algorithm.  I still need to open an issue in indi_pylibcamera with to find out how to disable DPC.  But I think this is a very minor issue for most people.

If indi_pylibcamera works for you, I think that is good enough.

aaronwmorris, 2023-06-22:
Closing this issue for now.  Feel free to reopen if this issue is not resolved.

closed 2023-06-22 by aaronwmorris