aaronwmorris/indi-allsky #736: Canon EOS 2000D

jkoenig72, 2023-05-04:
Hi Aaron, 

I think it would be quite interesting to use one older DSLR like Canon EOS 2000D with a fisheye lens and test this setup as an Allskycam,

I have one free to try including a good lens for that purpose.

When I try to configure Canon I detect this message in the logs:  

"Please update the CCD Information in the Image Info section before proceeding. The camera resolution shall be updated after the first exposure is complete." - I guess that's coming from the driver, as I can not find a place where I have to enter those values.

Thanks!
Joerg

jkoenig72, 2023-05-04:
Maybe I can add this to this "INDI Camera Configuration" as some properties?

jkoenig72, 2023-05-04:
I see there is an example for a canon - I will look into this...

https://github.com/aaronwmorris/indi-allsky/tree/main/examples/properties)/Canon_DSLR_EOS_1300D.txt

jkoenig72, 2023-05-04:
I see... 

{
    "PROPERTIES" : {
        "CCD_INFO" : {
            "CCD_MAX_X" : 5184,
            "CCD_MAX_Y" : 3456,
            "CCD_PIXEL_SIZE" : 4.3,
            "CCD_PIXEL_SIZE_X" : 4.3,
            "CCD_PIXEL_SIZE_Y" : 4.3,
            "CCD_BITSPERPIXEL" : 16
        }
    },
    "SWITCHES": {
        "CCD_CAPTURE_FORMAT": {
            "on": [
                "FORMAT_9"
            ]
        }
    }
}

aaronwmorris, 2023-05-04:
Yeah, give that a shot.

The CCD_CAPTURE_FORMAT could be different for your camera to set it to RAW mode.  In the indi-allsky log, it should print the setting label when it sets it.

jkoenig72, 2023-05-04:
Looking at it -

This could be the mother of all allsky cams ... ;-)

jkoenig72, 2023-05-05:
This seems to work for the 2000d:

{
    "PROPERTIES": {
        "CCD_INFO": {
            "CCD_MAX_X": 6000,
            "CCD_MAX_Y": 4000,
            "CCD_PIXEL_SIZE": 3.71,
            "CCD_PIXEL_SIZE_X": 3.71,
            "CCD_PIXEL_SIZE_Y": 3.71,
            "CCD_BITSPERPIXEL": 16
        }
    },
    "SWITCHES": {
        "CCD_CAPTURE_FORMAT": {
            "on": [
                "FORMAT_9"
            ]
        }
    }
}

jkoenig72, 2023-05-05:
[image]

jkoenig72, 2023-05-05:
Interesting... ;-)

aaronwmorris, 2023-05-05:
I recently worked with someone else on the Canon support, so it is fresh on my mind.
https://github.com/aaronwmorris/indi-allsky/discussions/681

closed 2023-07-26 by aaronwmorris