aaronwmorris/indi-allsky #757: Know-how question: How to use the multi-cam feature?

jkoenig72, 2023-05-14:
Hi Aaron,

I could not find any description of how to use multiple cams with one instance of indi-allsky,

Maybe I have missed it and it's there but how does it work?

Do I simply need to enter a different indi cam in the cam section of the config and then it duplicates all config settings for this new cam? Then I can select from the main UI? As simple as that?

Regards
Joerg

aaronwmorris, 2023-05-14:
It is probably better to say that the indi-allsky web interface and database supports multiple cameras.  You can only have one camera at a time actively taking pictures on a single system.

Where the multiple camera support becomes useful is setting up a remote web site, similar to TJ's allsky-website.  Using the SyncAPI, you can feed the images and videos from one installation of indi-allsky to another.  You could feed the data from 8 instances of indi-allsky to a single remote instance of indi-allsky.  The indi-allsky web interface is its own remote web site.

Gord Tulloch's installation is probably the best example of this:  https://www.gordtulloch.com/indi-allsky/  His camera runs from his home (I assume) and is syncing images using the SyncAPI to a virtual server on AWS.

You can even have hybrid setups where you have two standalone cameras and one system syncs images to the other where you would have both a local and remote data on a single instance.

jkoenig72, 2023-05-14:
Thanks. Understood. 

I run indi-allsky in multiple instances in lxd on an Intel nuc where the indi server is remote on small orange pi’s. Works nicely. Just wondering if there is a different way. 

I really enjoy your work. It’s a fantastic project and give lots of fun.

jkoenig72, 2023-05-14:
How is the docker support? I see some file in the repo. Can it be used?

jkoenig72, 2023-05-14:
I see there is some work ongoing. I will have a look…

aaronwmorris, 2023-05-15:
The docker stuff is not quite ready, maybe at 80%, but I can work on that tonight.

aaronwmorris, 2023-05-19:
I have working docker support as of #767

jkoenig72, 2023-05-19:
Thanks - I will have a look and test, most likely over the weekend.

aaronwmorris, 2023-10-31:
Closing this for now.  Let me know if you need any more help.

closed 2023-10-31 by aaronwmorris

jkoenig72, 2023-11-01:
I managed. It was very slow - more than 2 days, but worked at the end.

Thanks!