aaronwmorris/indi-allsky #759: Config menu on web interface maybe broken?

jkoenig72, 2023-05-15:
Hi Aaron,

I installed a fresh version and now I can not config indi-allsky.

Looks like the config tabs are somehow broken. Everything apart from camera-tab is mostly empty. I saw you applied some changes yesterday. maybe its worth to have a look 

Joerg

jkoenig72, 2023-05-15:
When I copy the indi-allsky folder from a machine I installed yesterday, it works. 

 scp -r indi_allsky/ xyz@192.168.10.3:/home/xyz/indi-allsky/

mk1blacklimited, 2023-05-15:
I have the same problem.

mk1blacklimited, 2023-05-15:
Tried running setup a couple of times. :) Also tried removing the installation completely and reinstalling, but the problem persists.

aaronwmorris, 2023-05-15:
Can you check web application the logs at `/var/log/indi-allsky/webapp-indi-allsky.log` for any recent exceptions?

aaronwmorris, 2023-05-15:
Confirmed.  Merged #761 to fix the config form.

I made a change for an element that was on the first tab.  I never clicked on the other tabs to test.

closed 2023-06-22 by aaronwmorris