aaronwmorris/indi-allsky #762: Know how question: what your experience with cam reflections inside the dome / pictures?

jkoenig72, 2023-05-16:
Just wondering if someone has found a clever solution for this. I see quite clear cam reflections in my allsky cams. 

It looks to me that it's worse if the cam lens is exposed more "out" of the case to the dome if the cam lens is relatively flat with the case, it does not seem that bad. 

Maybe someone found a good and simple way to minimize reflections.  

Thanks!
Joerg

aaronwmorris, 2023-05-16:
I have pretty bad reflections, as well.  The new stretching algorithm exacerbates the problem by really bringing out the reflections.  

I think the only real fix to this is a black, matte paint that is designed to be in the sun.

bluthen, 2023-05-16:
I do the opposite, I try to put the camera/lens as close to the top of the dome as I can. Go through less of the dome and get less distortion and better focus. But I'm probably using different materials, lenses, and have different points of reflection. We got a different setup, so to each their own to experiment with.