aaronwmorris/indi-allsky #763: New installation (RPI Model 4B 4GB, HQcam) - lots of errors

WirthmU, 2023-05-16:
I am running the RPI with the following version:
```
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
```

Installation went smoothly but the camera is not recognized:
```
May 16 12:02:38 allsky [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/usr/lib/python3.7/subprocess.py", line 1522, in _execute_child\n    raise child_exception_type(errno_num, err_msg, err_filename)\n'
May 16 12:02:38 allsky [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/usr/lib/python3.7/subprocess.py", line 775, in __init__\n    restore_signals, start_new_session)\n'
May 16 12:02:38 allsky [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/home/pi/indi-allsky/indi_allsky/camera/libcamera.py", line 191, in setCcdExposure\n    stderr=subprocess.STDOUT,\n'
May 16 12:02:38 allsky [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/home/pi/indi-allsky/indi_allsky/allsky.py", line 1830, in shoot\n    self.indiclient.setCcdExposure(exposure, sync=sync, timeout=timeout)\n'
May 16 12:02:38 allsky [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "/home/pi/indi-allsky/indi_allsky/allsky.py", line 1263, in run\n    self.shoot(self.exposure_v.value, sync=False)\n'
May 16 12:02:38 allsky [ERROR] MainProcess allsky.unhandled_exception() #41: '  File "allsky.py", line 119, in <module>\n    action_func(*args_list, **kwargs_dict)\n'
May 16 12:02:38 allsky [ERROR] MainProcess allsky.unhandled_exception() #36: Value: [Errno 2] No such file or directory: 'libcamera-still': 'libcamera-still'
May 16 12:02:38 allsky [ERROR] MainProcess allsky.unhandled_exception() #35: Type: <class 'FileNotFoundError'>
May 16 12:02:38 allsky [ERROR] MainProcess allsky.unhandled_exception() #34: An uncaught exception occurred:
May 16 12:02:38 allsky [INFO] MainProcess libcamera.setCcdExposure() #183: image command: libcamera-still --immediate --nopreview --raw --denoise off --awbgains 1,1 --gain 1 --shutter 1000 --metadata /tmp/tmpdlllbghq.json --metadata-format json --output /tmp/tmphly_dyqq.dng
May 16 12:02:38 allsky [INFO] MainProcess allsky.shoot() #1828: Taking 0.00100001 s exposure (gain 1)
May 16 12:02:38 allsky [WARNING] MainProcess allsky.reconfigureCcd() #1566: Change to day
May 16 12:02:38 allsky [INFO] MainProcess allsky._queueManualTasks() #2426: Checking for manually submitted tasks
May 16 12:02:38 allsky [INFO] MainProcess indi.getTelescopeRaDec() #833: Telescope Coord: RA 1.14, Dec 46.94
May 16 12:02:38 allsky [INFO] VideoWorker001 video.systemHealthCheck() #954: Performing system health check
May 16 12:02:37 allsky [INFO] MainProcess allsky.detectMoonMode() #1608: Moon altitude: 46:41:44.5, phase 11.9%
May 16 12:02:37 allsky [INFO] MainProcess allsky.detectNight() #1591: Sun altitude: 49:36:28.6
May 16 12:02:37 allsky [INFO] MainProcess allsky._fileUploadWorkerStart() #903: Starting FileUploader process 1
May 16 12:02:37 allsky [WARNING] MainProcess video._load_detection_mask() #1199: No detection mask defined
May 16 12:02:37 allsky [INFO] MainProcess allsky._startVideoWorker() #840: Starting VideoWorker process 1
May 16 12:02:37 allsky [WARNING] MainProcess image._load_detection_mask() #1491: No detection mask defined
May 16 12:02:37 allsky [INFO] MainProcess allsky._startImageWorker() #775: Starting ImageWorker process 1
May 16 12:02:37 allsky [INFO] MainProcess allsky.run() #1043: Exposure state: unset
May 16 12:02:37 allsky [INFO] MainProcess allsky.run() #1042: Camera last ready: 0.0s
May 16 12:02:34 allsky [ERROR] MainProcess allsky._initialize() #723: CCD day gain below minimum, changing to 1
May 16 12:02:31 allsky [ERROR] MainProcess allsky._initialize() #718: CCD moon mode gain above maximum, changing to 22
May 16 12:02:30 allsky [INFO] MainProcess indi.newMessage() #324: new Message 2023-05-16T10:02:30: [INFO] Telescope slew is complete. Parked 
```
Any help to this?

aaronwmorris, 2023-05-16:
I do not have any support for libcamera in Raspbian 10 Legacy.

In order to use libcamera, it requires Raspbian 11 (32 or 64-bit).

WirthmU, 2023-05-17:
Thank you for your quick reply.

Now I have installed Raspbian 11 31-bit.

However neither indiserver nor indi-allsky seem to run:
```
pi@allsky:~/indi-allsky $ systemctl --user status indiserver
● indiserver.service - Indi Server
     Loaded: loaded (/home/pi/.config/systemd/user/indiserver.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Wed 2023-05-17 12:19:39 CEST; 18s ago
    Process: 29983 ExecStart=/usr/bin/indiserver -p 7624 indi_simulator_telescope indi_simulator_ccd (code=exited, status=203/EXEC)
   Main PID: 29983 (code=exited, status=203/EXEC)
        CPU: 3ms

Mai 17 12:19:39 allsky systemd[776]: Started Indi Server.
Mai 17 12:19:39 allsky systemd[29983]: indiserver.service: Failed to locate executable /usr/bin/indiserver: No such file or directory
Mai 17 12:19:39 allsky systemd[29983]: indiserver.service: Failed at step EXEC spawning /usr/bin/indiserver: No such file or directory
Mai 17 12:19:39 allsky systemd[776]: indiserver.service: Main process exited, code=exited, status=203/EXEC
Mai 17 12:19:39 allsky systemd[776]: indiserver.service: Failed with result 'exit-code'.
```

```
pi@allsky:~/indi-allsky $ systemctl --user status indi-allsky
● indi-allsky.service - Indi AllSky Service
     Loaded: loaded (/home/pi/.config/systemd/user/indi-allsky.service; disabled; vendor preset: enabled)
     Active: failed (Result: signal) since Wed 2023-05-17 12:19:33 CEST; 57s ago
TriggeredBy: ● indi-allsky.timer
    Process: 29979 ExecStart=/home/pi/indi-allsky/virtualenv/indi-allsky/bin/python3 allsky.py --log syslog run (code=killed, signal=ABRT)
   Main PID: 29979 (code=killed, signal=ABRT)
        CPU: 30ms

Mai 17 12:19:33 allsky systemd[776]: Started Indi AllSky Service.
Mai 17 12:19:33 allsky python3[29979]: Could not find platform independent libraries <prefix>
Mai 17 12:19:33 allsky python3[29979]: Could not find platform dependent libraries <exec_prefix>
Mai 17 12:19:33 allsky python3[29979]: Consider setting $PYTHONHOME to <prefix>[:<exec_prefix>]
Mai 17 12:19:33 allsky python3[29979]: Fatal Python error: initfsencoding: Unable to get the locale encoding
Mai 17 12:19:33 allsky python3[29979]: ModuleNotFoundError: No module named 'encodings'
Mai 17 12:19:33 allsky python3[29979]: Current thread 0xb6f9bbd0 (most recent call first):
Mai 17 12:19:33 allsky systemd[776]: indi-allsky.service: Main process exited, code=killed, status=6/ABRT
Mai 17 12:19:33 allsky systemd[776]: indi-allsky.service: Failed with result 'signal'.
```

aaronwmorris, 2023-05-17:
Did you start from scratch?

aaronwmorris, 2023-05-17:
I would fully reimage your SDcard with the Raspbian 11 image.

closed 2023-07-26 by aaronwmorris