aaronwmorris/indi-allsky #777: Stretch and Text Overlay

HxPI, 2023-05-24:
It appears the text overlay is affected by the new stretch image function. When stretch is enabled, the text is washed out and unreadable. When disabled, the text is clear and legible. Please verify whether the stretch is affecting the text overlay. I would assume the text overlay would take place after the image stretch. Also, it appears stretch does not affect the raw image output, i.e. tiff or fits; is this correct? Thank you.

aaronwmorris, 2023-05-24:
That is odd.  The text label is applied well after the stretch.  Can you post some screenshots?

HxPI, 2023-05-24:
Nevermind. I opened the images directly from storage and the text overlay is fine. The text overlay on the web interface become readable from time to time. Must be the way the webpage is displaying them.

closed 2023-05-24 by HxPI