aaronwmorris/indi-allsky #780: Not getting Images after recent update.

Sedim, 2023-05-28:
Thank you for this comprehensive capture software. Was working great. Updated to latest version. And then it stopped working. "Latest" does not display any images. Here is a sampling of the initial log with errors after updating again: 

May 27 23:37:53 allsky [INFO] MainProcess allsky.run() #1043: Exposure state: OK
May 27 23:37:53 allsky [INFO] MainProcess allsky.run() #1042: Camera last ready: 1.7s
May 27 23:37:52 allsky [INFO] MainProcess allsky.run() #1184: Exposure received in 0.5569 s (0.5569)
May 27 23:37:51 allsky [INFO] MainProcess allsky.run() #1276: Total time since last exposure 10.0214 s
May 27 23:37:51 allsky [INFO] MainProcess indi.set_number() #1230: Setting Duration (s) = 6.400999700417742e-05
May 27 23:37:51 allsky [INFO] MainProcess allsky.shoot() #1828: Taking 0.00006401 s exposure (gain 50)
May 27 23:37:42 allsky [INFO] MainProcess allsky._queueManualTasks() #2426: Checking for manually submitted tasks
May 27 23:37:42 allsky [INFO] MainProcess indi.getTelescopeRaDec() #833: Telescope Coord: RA 15.05, Dec 40.37
May 27 23:37:42 allsky [INFO] MainProcess indi.getCcdTemperature() #848: Sensor temperature: 27.5
May 27 23:37:42 allsky [INFO] ImageWorker002 image.processImage() #352: Image: 1280 x 960
May 27 23:37:42 allsky [ERROR] ImageWorker002 image.debayer() #2237: No bayer pattern detected
May 27 23:37:42 allsky [INFO] ImageWorker002 sqm.calculate() #42: Weighted SQM average: 14862.51
May 27 23:37:42 allsky [INFO] ImageWorker002 sqm.calculate() #37: Raw SQM average: 990.84
May 27 23:37:42 allsky [WARNING] ImageWorker002 sqm._generateSqmMask() #63: Using central 20% ROI for SQM calculations
May 27 23:37:42 allsky [INFO] ImageWorker002 sqm._generateSqmMask() #48: Generating mask based on SQM_ROI
May 27 23:37:42 allsky [INFO] ImageWorker002 sqm.calculate() #21: Exposure: 0.000064, gain: 50
May 27 23:37:42 allsky [WARNING] ImageWorker002 image._calibrate() #2029: Dark not found: ccd1 16bit 0.0000640s gain 50 bin 1 29.70c
May 27 23:37:42 allsky [WARNING] ImageWorker002 image._calibrate() #2011: Temperature matched dark not found: 29.70c
May 27 23:37:42 allsky [INFO] ImageWorker002 image._calibrate() #1994: Searching for dark frame: gain 50, exposure >= 0.0, temp >= 29.7c
May 27 23:37:42 allsky [WARNING] ImageWorker002 image._calibrate() #1982: Bad Pixel Map not found: ccd1 16bit 0.0000640s gain 50 bin 1 29.70c
May 27 23:37:42 allsky [WARNING] ImageWorker002 image._calibrate() #1964: Temperature matched bad pixel map not found: 29.70c
May 27 23:37:42 allsky [INFO] ImageWorker002 image._calibrate() #1947: Searching for bad pixel map: gain 50, exposure >= 0.0, temp >= 29.7c
May 27 23:37:42 allsky [WARNING] ImageWorker002 image._detectBitDepth() #1910: Updated default bit depth: 16
May 27 23:37:42 allsky [INFO] ImageWorker002 image._detectBitDepth() #1882: Image max value: 65520
May 27 23:37:42 allsky [INFO] ImageWorker002 image.add() #1827: Image bits: 16, cfa: None
May 27 23:37:42 allsky [INFO] MainProcess allsky.detectMoonMode() #1612: Moon Mode conditions detected
May 27 23:37:42 allsky [INFO] MainProcess allsky.detectMoonMode() #1608: Moon altitude: 28:25:00.7, phase 55.0%
May 27 23:37:42 allsky [INFO] MainProcess allsky.detectNight() #1591: Sun altitude: -25:45:51.4
May 27 23:37:42 allsky [INFO] MainProcess allsky._startImageWorker() #775: Starting ImageWorker process 2
May 27 23:37:42 allsky [ERROR] MainProcess allsky._startImageWorker() #768: Image worker exception: 
May 27 23:37:42 allsky [ERROR] MainProcess allsky._startImageWorker() #768: Image worker exception: 
May 27 23:37:42 allsky [ERROR] MainProcess allsky._startImageWorker() #768: Image worker exception: cv2.error: OpenCV(4.7.0) /io/opencv/modules/core/src/matrix_iterator.cpp:71: error: (-215:Assertion failed) A.size == arrays[i0]->size in function 'init'
May 27 23:37:42 allsky [ERROR] MainProcess allsky._startImageWorker() #768: Image worker exception:     adu = cv2.mean(src=self.image, mask=self._adu_mask)[0]
May 27 23:37:42 allsky [ERROR] MainProcess allsky._startImageWorker() #768: Image worker exception:   File "/home/pi/indi-allsky/indi_allsky/image.py", line 2098, in _calculate_8bit_adu
May 27 23:37:42 allsky [ERROR] MainProcess allsky._startImageWorker() #768: Image worker exception:     return self._calculate_8bit_adu(i_ref)
May 27 23:37:42 allsky [ERROR] MainProcess allsky._startImageWorker() #768: Image worker exception:   File "/home/pi/indi-allsky/indi_allsky/image.py", line 2087, in calculate_8bit_adu
May 27 23:37:42 allsky [ERROR] MainProcess allsky._startImageWorker() #768: Image worker exception:     adu = self.image_processor.calculate_8bit_adu()
May 27 23:37:42 allsky [ERROR] MainProcess allsky._startImageWorker() #768: Image worker exception:   File "/home/pi/indi-allsky/indi_allsky/image.py", line 457, in processImage
May 27 23:37:42 allsky [ERROR] MainProcess allsky._startImageWorker() #768: Image worker exception:     self.processImage(i_dict)
May 27 23:37:42 allsky [ERROR] MainProcess allsky._startImageWorker() #768: Image worker exception:   File "/home/pi/indi-allsky/indi_allsky/image.py", line 256, in saferun
May 27 23:37:42 allsky [ERROR] MainProcess allsky._startImageWorker() #768: Image worker exception:     self.saferun()
May 27 23:37:42 allsky [ERROR] MainProcess allsky._startImageWorker() #768: Image worker exception:   File "/home/pi/indi-allsky/indi_allsky/image.py", line 227, in run
May 27 23:37:42 allsky [ERROR] MainProcess allsky._startImageWorker() #768: Image worker exception: Traceback (most recent call last):

aaronwmorris, 2023-05-28:
What kind of camera are you using?  Do you have any masks defined in your configuration?

Sedim, 2023-05-28:
Yes, I have a mask defined. The Camera is ZWO ASI224MC 1.2 MP

Sedim, 2023-05-28:
So I removed the reference to the mask and I am getting pictures now! Is there something special about the mask that is causing these errors? I attached the mask above.

Sedim, 2023-05-28:
Update: Rebooted and put back the reference to the very same mask and it is now working!!!!. This is very strange because before removing the mask, I was getting errors. I removed the reference and it worked (No errors). I put back the reference and it is still working (No errors). I also enabled the on-screen mask view and that works fine. So... strange. 

At this point I will close this but will update if I get this again. Thank you for your quick response.

closed 2023-05-28 by Sedim

aaronwmorris, 2023-05-28:
Very strange.  I am guessing something was causing the image or mask size to be different.   If the mask is a different size from the image, you can get errors like this.  `Assertion failed) A.size == arrays[i0]->size` seems to indicate that.

Rebooting was probably the right thing to do.