aaronwmorris/indi-allsky #783: Bayer transformation problems

tonycolo, 2023-05-28:
I'm experimenting with the QHY168c and have tried all 5 patterns and none render colors correctly.
QHY says it should be RGGB per:
Appendix B:  https://www.qhyccd.com/astronomical-camera-qhy168-qhy247-qhy128-qhy367/
Versions:
indi-3rdparty 2.0.1
indi-allsky 6.3
Raspberry PI OS 6.1 64-bit
Raspberry PI 4 Raspberry PI 

I was using the Occulus All-Sky Pro successfully, but that's black and white.

Thanks for any suggestions.

aaronwmorris, 2023-05-29:
Can you describe what is wrong with the color rendering?

Can you enable saving FITS and upload one here so I can inspect it?

tonycolo, 2023-05-29:
Previously, I ran indoor tests at night and the coloring never looked anything close to correct; rggb was far too red/orange.  But...

I ran the update procedure (using main) for indi-allsky--but did not update indi--and let it run for a while indoor during the day.  
Here are the [png](https://drive.google.com/file/d/1tafqVJu1pfolp7h2l2Zdq5ar5DwNPItj/view?usp=share_link) and [fit](https://drive.google.com/file/d/1YHPxJTeI0ylTSZ6UMomR5Fc0iPt6Hvj9/view?usp=share_link). The color of the wood panelling looks ok in the png, but the sky is weird, perhaps it's due to too much variation in lighting?

I'll rerun at night outside with stars and report back.

aaronwmorris, 2023-05-29:
I tested debayering using the `./testing/debayer.py` script in the repo.  RGGB is definitely the correct pattern, but I do not see any of the purple splotches, the colors look fairly normal.  Some colors may be off if you do not have an UV/IR cut filter with your camera.

The purple splotches are probably an integer overflow in the blue and green channels during some processing.  Did you enable any other extra processing options in indi-allsky?

tonycolo, 2023-05-29:
Thank you for pointing out that script--I did not know you were using OpenCV.  I obtain reasonable results when running it directly against that same .fit file.   The resulting greenish cast is a [known artifact of debayering](https://www.cloudynights.com/topic/547023-green-hue-after-debayer/) as you know.
The only other processing turned on is SCNR and horizontal and vertical flipping.

More testing forthcoming.   I should probably close this and carry on in the indilib forum.

aaronwmorris, 2023-05-29:
Its not a problem.  I wrote that script to specifically test different debayering options.  I have had one instance where the automatically detected debayering algorithm was wrong.

Now that you mention SCNR, are you using `Average Neutral`?  If so, this can cause the magenta bias.  Use `Maximum Neutral` for a better image.

closed 2023-08-24 by aaronwmorris