aaronwmorris/indi-allsky #789: libcamera-still error: b"ERROR: *** unrecognised option '--metadata' ***\n"

milesnz, 2023-06-04:
Hi,
Below I have pasted the log messages that I'm getting when the indi-allsky attempts to take a photo every 15 secs.
It seems like the libcamera-still command doesn't accept the --metadata parameter.
If I cut 'n paste the command from the log into a cli command, and remove the --metadata parameters (all them) then the command works and captures the photo.
Have I missed a step that I can check on?
Also - my camera is a 5mp OV5647 camera but I don't know which IMX overlay to match it up to when configuring indi-allsky?

Raspberry PI4b, 4gb, loaded on 32gb USB3.1 flash drive.

astroberry@astroberry:/tmp $ lsb_release -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 10 (buster)
Release:	10
Codename:	buster

I first installed astroberry and then loaded libcamera onto it.
Then I installed indi-allsky.
I do not have a telescope nor a gps unit.

any help would be appreciated please.
regards

Jun 4 18:54:04 astroberry [ERROR] MainProcess libcamera._processMetadata() #385: libcamera sensor black level key not found 
Jun 4 18:54:04 astroberry [ERROR] MainProcess libcamera._processMetadata() #306: libcamera sensor temperature key not found 
Jun 4 18:54:04 astroberry [ERROR] MainProcess libcamera._processMetadata() #283: Metadata file not found: [Errno 2] No such file or directory: '/tmp/tmpc2fxvozh.json' 
Jun 4 18:54:04 astroberry [ERROR] MainProcess libcamera.getCcdExposureStatus() #263: libcamera-still error: b"ERROR: *** unrecognised option '--metadata' ***\n" 
Jun 4 18:54:04 astroberry [INFO] MainProcess allsky.run() #1281: Total time since last exposure 15.0082 s 
Jun 4 18:54:04 astroberry [INFO] MainProcess libcamera.setCcdExposure() #212: image command: libcamera-still --immediate --nopreview --encoding jpg --quality 95 --denoise off --gain 1 --shutter 100 --metadata /tmp/tmpc2fxvozh.json --metadata-format json --awbgains 1,1 --vflip 1 --hflip 1 --metadata / --output /tmp/tmpj4t4u5jk.jpg

aaronwmorris, 2023-06-04:
I did not support Debian 10/Astroberry for libcamera.  The libcamera support is very stale.

I would recommend Raspbian 11 64-bit.  There is no selection for the ov5647, but you can select any of the imx models to test.  The Bayer pattern may be wrong, but it will work.

I can research adding the model when I am back from vacation.

milesnz, 2023-06-04:
Thanks for the fast response.
Enjoy your vacation

closed 2023-06-22 by aaronwmorris