aaronwmorris/indi-allsky #795: Error on config page: Exposure cannot be greater than night period / Exposure Period (Night)

jkoenig72, 2023-06-17:
Hi Aaron,

got a strange issue: new installation.

Want to configure Error on config page: Exposure cannot be greater than night period / Exposure Period (Night)

Use 60 / 15 - before that works.

Now I got a checking error saying 

Exposure cannot be greater than night period

Period is less than maximuum exposure

Even if in enter 65 or 70 in Exposure Period (Night) it does not allow me to save config.

Guess that's a validation checking error you introduced lately

/jk

jkoenig72, 2023-06-17:
If I put in values for Exposure Period (Night) and Exposure Period (Day)  greater than Max Exposure

then it allows me to save.

its in forms.py on line 2106 ff:

        # exposure checking
        if self.CCD_EXPOSURE_MAX.data > self.EXPOSURE_PERIOD.data:
            self.CCD_EXPOSURE_MAX.errors.append('Exposure cannot be greater than night period')
            self.EXPOSURE_PERIOD.errors.append('Period is less than maximuum exposure')
            result = False

        if self.CCD_EXPOSURE_MAX.data > self.EXPOSURE_PERIOD_DAY.data:
            self.CCD_EXPOSURE_MAX.errors.append('Exposure cannot be greater than day period')
            self.EXPOSURE_PERIOD.errors.append('Period is less than maximuum exposure')
            result = False

Sure that correct?

aaronwmorris, 2023-06-17:
The "period" is the time between the beginning of one exposure to the beginning of the next exposure.  If the exposure is longer than the period, then it just extends the period.

This is the reason I added the validation.  The exposure needs to fit within the period window.

I am willing to be wrong here, but I cannot see a valid case where you would want an exposure to be longer than the period.

jkoenig72, 2023-06-18:
[image]

Let me post a screenshot. Another fresh install, it does not allow me to save. 15 should be ok, nor?

jkoenig72, 2023-06-18:
Let me show you - OOTB box fresh install:

Let me save only if Exposure Period (Day) ist also set to 60. Find that strange. I want more pics during the daytime. 

[image]

aaronwmorris, 2023-06-18:
I understand now.  I have removed those checks from the form validation.