aaronwmorris/indi-allsky #802: Star trail images not being generated

xsnrg, 2023-06-21:
The last couple of nights have been perfectly clear, but it seems the star detection is starting a little too early or something, and the star trails are not getting generated. The star trail video is not getting generated either, and the image looks like this: 
[image]

Would it help to define a region of interest for the ADU, or is there another setting that would help?

I have tried removing the timelapses and re-generating them. I get the same result.

aaronwmorris, 2023-06-21:
There is a setting in the config:  `Startrail Max ADU` that could be too aggressive (low) for your setup.  This number probably needs to be increased.

There is also a hidden view you can use to help determine the appropriate ADU cutoff:  https://hostname/indi-allsky/adu that will show you the adu averages in 15 minute blocks.  I just merged #803 to increase the number of days reported to 7.

xsnrg, 2023-06-21:
In looking at the ADUs, most of the images that would have been used to create the star trail are under the setting for Startrail Max ADU. Is there a threshold that would abort the startrail creation if the images at the start are higher ADU?

xsnrg, 2023-06-21:
`Jun 21 09:51:30 rpi4 [INFO] MainProcess allsky._startVideoWorker() #840: Starting VideoWorker process 8
Jun 21 09:51:53 rpi4 [INFO] VideoWorker008 video.generateVideo() #279: Found 809 images for timelapse
Jun 21 09:51:54 rpi4 [INFO] VideoWorker008 miscDb.addVideo() #367: Adding video /var/www/html/allsky/images/ccd_986d2ff8-6b06-4d9a-971e-9837262561c4/20230620/allsky-timelapse_ccd3_20230620_night.mp4 to DB
Jun 21 09:54:06 rpi4 [INFO] VideoWorker008 timelapse.generate() #81: Timelapse generated in 132.6851 s
Jun 21 09:54:06 rpi4 [INFO] VideoWorker008 timelapse.generate() #83: FFMPEG output: b'[swscaler @ 0xaaaae7c11db0] [warning] deprecated pixel format used, make sure you did set range correctly\n'
Jun 21 09:54:07 rpi4 [INFO] VideoWorker008 video.generateKeogramStarTrails() #543: Found 809 images for keogram/star trails
Jun 21 09:54:07 rpi4 [INFO] VideoWorker008 miscDb.addKeogram() #413: Adding keogram /var/www/html/allsky/images/ccd_986d2ff8-6b06-4d9a-971e-9837262561c4/20230620/allsky-keogram_ccd3_20230620_night.jpg to DB
Jun 21 09:54:07 rpi4 [INFO] VideoWorker008 miscDb.addStarTrail() #461: Adding star trail /var/www/html/allsky/images/ccd_986d2ff8-6b06-4d9a-971e-9837262561c4/20230620/allsky-startrail_ccd3_20230620_night.jpg to DB
Jun 21 09:54:07 rpi4 [INFO] VideoWorker008 video.generateKeogramStarTrails() #611: Processed 0 of 809 images
Jun 21 09:54:07 rpi4 [INFO] VideoWorker008 keogram.processImage() #128: New Shape: (1080, 1, 3)
Jun 21 09:54:07 rpi4 [INFO] VideoWorker008 keogram.processImage() #131: New dtype: uint8
Jun 21 09:54:07 rpi4 [INFO] VideoWorker008 starTrails._generateSqmMask() #346: Generating mask based on SQM_ROI
Jun 21 09:54:15 rpi4 [INFO] VideoWorker008 video.generateKeogramStarTrails() #611: Processed 100 of 809 images
Jun 21 09:54:22 rpi4 [INFO] VideoWorker008 video.generateKeogramStarTrails() #611: Processed 200 of 809 images
Jun 21 09:54:30 rpi4 [INFO] VideoWorker008 video.generateKeogramStarTrails() #611: Processed 300 of 809 images
Jun 21 09:54:37 rpi4 [INFO] VideoWorker008 video.generateKeogramStarTrails() #611: Processed 400 of 809 images
Jun 21 09:54:44 rpi4 [INFO] VideoWorker008 video.generateKeogramStarTrails() #611: Processed 500 of 809 images
Jun 21 09:54:51 rpi4 [INFO] VideoWorker008 video.generateKeogramStarTrails() #611: Processed 600 of 809 images
Jun 21 09:54:59 rpi4 [INFO] VideoWorker008 video.generateKeogramStarTrails() #611: Processed 700 of 809 images
Jun 21 09:55:08 rpi4 [INFO] VideoWorker008 video.generateKeogramStarTrails() #611: Processed 800 of 809 images
Jun 21 09:55:09 rpi4 [INFO] VideoWorker008 keogram.finalize() #143: Images processed for keogram in 16.0 s
Jun 21 09:55:09 rpi4 [INFO] VideoWorker008 keogram.trimEdges() #285: Switch angle: 60.64
Jun 21 09:55:09 rpi4 [INFO] VideoWorker008 keogram.trimEdges() #303: Trim angle: 0
Jun 21 09:55:09 rpi4 [INFO] VideoWorker008 keogram.trimEdges() #306: Keogram dimensions: 810 x 1080
Jun 21 09:55:09 rpi4 [INFO] VideoWorker008 keogram.trimEdges() #307: Original image dimensions: 1920 x 1080
Jun 21 09:55:09 rpi4 [INFO] VideoWorker008 keogram.trimEdges() #308: Original rotated image dimensions: 1920 x 1080
Jun 21 09:55:09 rpi4 [INFO] VideoWorker008 keogram.trimEdges() #320: Trim height: 18
Jun 21 09:55:09 rpi4 [INFO] VideoWorker008 keogram.trimEdges() #328: Calculated trimmed area: (0, 18) (810, 1062)
Jun 21 09:55:09 rpi4 [INFO] VideoWorker008 keogram.trimEdges() #335: New trimmed image: 810 x 1044
Jun 21 09:55:09 rpi4 [WARNING] VideoWorker008 keogram.finalize() #225: Creating keogram: /var/www/html/allsky/images/ccd_986d2ff8-6b06-4d9a-971e-9837262561c4/20230620/allsky-keogram_ccd3_20230620_night.jpg
Jun 21 09:55:09 rpi4 [INFO] VideoWorker008 keogram.finalize() #245: Image compressed in 0.0075 s
Jun 21 09:55:09 rpi4 [WARNING] VideoWorker008 starTrails.finalize() #229: Star trails images processed in 0.4 s
Jun 21 09:55:09 rpi4 [WARNING] VideoWorker008 starTrails.finalize() #230: Excluded 802 images
Jun 21 09:55:09 rpi4 [WARNING] VideoWorker008 starTrails.finalize() #304: Creating star trail: /var/www/html/allsky/images/ccd_986d2ff8-6b06-4d9a-971e-9837262561c4/20230620/allsky-startrail_ccd3_20230620_night.jpg
Jun 21 09:55:09 rpi4 [INFO] VideoWorker008 starTrails.finalize() #324: Image compressed in 0.0444 s
Jun 21 09:55:09 rpi4 [ERROR] VideoWorker008 video.generateKeogramStarTrails() #675: Not enough frames to generate star trails timelapse: 7
Jun 21 09:55:09 rpi4 [WARNING] VideoWorker008 video.generateKeogramStarTrails() #680: Total keogram/star trail processing in 62.2 s
`

The last part about not enough frames to generate star trails tells me something is still off. I did raise the `Startrail Max ADU` to 80 from 74. Most of the images during the night are right around 70 ADU

xsnrg, 2023-06-21:
One more interesting thing, the ADU values start at 22:00 in the /adu link, but the image timestamp is 21:30.

aaronwmorris, 2023-06-21:
The ADU view only shows data between 22:00 and 04:00 to ensure it only shows night time data.

Also, ensure you reload (or restart) the indi-allsky service when you change the parameters.  Otherwise, they will not become active.

xsnrg, 2023-06-21:
I restarted the services, I also rebooted the entire system, cleaned the timelapses out, ran setup to get things up to release 7.0, and ran again with the same result.
```Jun 21 10:31:31 rpi4 [WARNING] VideoWorker001 starTrails.finalize() #230: Excluded 802 images

Jun 21 10:31:31 rpi4 [ERROR] VideoWorker001 video.generateKeogramStarTrails() #570: Not enough frames to generate star trails timelapse: 7
```

xsnrg, 2023-06-21:
[image]
[image]

aaronwmorris, 2023-06-21:
That last view is weird.  It appears to *NOT* show the night time stats.

Anyway, you may need to enable some debug lines in `./indi_allsky/starTrails.py` to log the reason the images are being excluded.

An image can be excluded for two reasons:
1.  The average ADU exceeds the cutoff
2. The Pixel Threshold (percentage, default 1.0) of pixels above the Mask Threshold (default 190) is exceeded.

xsnrg, 2023-06-21:
The last view is a little confusing, but it is from the bottom up, so it stops logging ADU at 4:59 AM and starts again at 22:00, which seems reasonable, given your previous comment.

I will see if I can set some debugging in starTrails.py and let you know.

xsnrg, 2023-06-21:
The answer is both.  

```Jun 21 17:18:38 rpi4 [WARNING] VideoWorker001 starTrails.processImage() #178:  Excluding image due to pixel cutoff: 133872
Jun 21 17:18:38 rpi4 [WARNING] VideoWorker001 starTrails.processImage() #178:  Excluding image due to pixel cutoff: 104159
Jun 21 17:18:38 rpi4 [WARNING] VideoWorker001 starTrails.processImage() #178:  Excluding image due to pixel cutoff: 111501
```
and
```
Jun 21 17:18:39 rpi4 [WARNING] VideoWorker001 starTrails.processImage() #170:  Excluding image due to brightness: 107.55
Jun 21 17:18:39 rpi4 [WARNING] VideoWorker001 starTrails.processImage() #170:  Excluding image due to brightness: 107.46
Jun 21 17:18:39 rpi4 [WARNING] VideoWorker001 starTrails.processImage() #170:  Excluding image due to brightness: 106.81
Jun 21 17:18:39 rpi4 [WARNING] VideoWorker001 starTrails.processImage() #170:  Excluding image due to brightness: 94.79
Jun 21 17:18:39 rpi4 [WARNING] VideoWorker001 starTrails.processImage() #170:  Excluding image due to brightness: 83.99
```

I set the ADU up to 110, and ran it again. There are far fewer of the brightness errors, and now many more of the pixel cutoff errors. The odd part is nothing changed with the configuration until these adjustments of trying to get it to work again.

```Jun 21 17:34:11 rpi4 [WARNING] VideoWorker003 starTrails.processImage() #178:  Excluding image due to pixel cutoff: 242108
Jun 21 17:34:11 rpi4 [WARNING] VideoWorker003 starTrails.processImage() #178:  Excluding image due to pixel cutoff: 242075
Jun 21 17:34:11 rpi4 [WARNING] VideoWorker003 starTrails.processImage() #178:  Excluding image due to pixel cutoff: 240854
Jun 21 17:34:11 rpi4 [WARNING] VideoWorker003 starTrails.processImage() #178:  Excluding image due to pixel cutoff: 192257
Jun 21 17:34:11 rpi4 [WARNING] VideoWorker003 starTrails.processImage() #178:  Excluding image due to pixel cutoff: 151052
```

All of the cutoffs seem to be between 70k and 250k. The timelapse for the actual sky looks good, if not a bit washed out due to light pollution, but you can even see the galaxy in it.

aaronwmorris, 2023-06-21:
You can increase the Mask Threshold to 255 to effectively disable that threshold.  No pixels can be above 255 in value.

xsnrg, 2023-06-22:
The mask threshold certainly seems to be making a difference, but it seems more like for my lattitude (41.7N) that the startrail timelapse is starting too early and ending too late. If I set the threshold to 255, not rejecting any frames, the sides where the sun set and the sun rises just take over and wash the image out. If I make a very small adjustment away from 255 to 252, I get this image: 
[image]
It makes me wonder if being the solstice, and the very expanded twilight hours are wreaking havoc with the algorithm? I am going to try leaving the threshold at 252 and taking the ADU down to 100, and will post an image. 

It does seem, at this point, that it is a tuning thing, not a code problem. I do wonder though if there is something that could be done to auto-tune around this. I don't want to make seasonal adjustments to the parameters if it can be helped.

aaronwmorris, 2023-06-22:
I am guessing you are using the IMX477?  Are you using darks?

xsnrg, 2023-06-22:
Sony IMX662. I have not gotten darks set up again since switching sensors from the 290. Getting darks is on my short list. Here is a timelapse from last night: https://youtu.be/KfjseaXX9ec

aaronwmorris, 2023-06-22:
Nice!  Is it an ASI662 or some other module?

The startrails are all about excluding images and I have the two methods for excluding.  I think I need to add a third factor... sun altitude.  I can work on excluding all images where the sun is above a configurable level, like -12 or -18 degrees.  This would exclude the images with the bright edges.

xsnrg, 2023-06-22:
It is the ZWO ASI implementation. I bought it on specs alone, never having seen one before. My ASI290 is getting a lot of dead pixels and is showing its age. So far, I am very impressed with the new camera.

Sun altitude would be an excellent addition I think. It should squeeze out the twilight on both ends.

aaronwmorris, 2023-06-22:
It looks really good for not having darks.

I have merged #808 which adds the threshold for the sun altitude.  The default is -15 degrees, but this is configurable.

xsnrg, 2023-06-22:
That certainly gave me something to play with. Would it be hard to make the sunrise and sunset altitudes independent? Not a big deal here, but I could see it being handy for mountain/valley situations. It looks like I could benefit from keeping the morning away a little longer too. Odd things with the oblique spheroid we live on :)

[image]

aaronwmorris, 2023-06-22:
I do not really have much for the star trails docuementation, I will get around to writing this up eventually.

The ADU average threshold is for the average ADU of the central region of interest of the image.  This is dynamically calculated as the central 2/3s of the image (to cut off the edges).  You can play with this by adjusting the `SQM/Star Region of Interest` value.  If you defined a detection mask, the mask area is used instead.

You can read about detection masks here:  https://github.com/aaronwmorris/indi-allsky/wiki/Detection-Masks

The pixel threshold is just the percentage of pixels in the ROI that exceed a certain ADU.  The default is 1% of pixels exceeding 190.  You could raise the percentage value and/or increase the ADU threshold in your case.

The values I have designated as the default are just the defaults because they worked well on my cameras (SV305, ASI290MM, ASI385MC, etc).  But these values certainly do not always work very well with all cameras, especially higher resolution cameras.

aaronwmorris, 2023-06-24:
Merged #813 with a moon altitude limit, although it is set to 91 which is effectively disabled by default.

xsnrg, 2023-06-24:
Nice, that will be fun to play with as well. Thank you. I wonder if it would also be possible to turn moon mode off based on altitude and phase. eg. settable days +/-  into a cycle, and altitude below settable  degrees, turn moon mode off.

aaronwmorris, 2023-06-24:
Done.  Merged #814 with moon phase limit for startrails.

aaronwmorris, 2023-06-24:
I realized I made a mistake in the logic here and I have fixed it in #815 .

I also added a checkbox to just re-use the moonmode parameters to exclude frames.

xsnrg, 2023-06-26:
You have provided options to tune around the problems I had been having. Thank you very much. Closing.

closed 2023-06-26 by xsnrg