aaronwmorris/indi-allsky #819: Problems with taking darks

jkoenig72, 2023-07-01:
Hi Aaron,

Have 3 cams running - all stable when running is capture mode.  

Now I want to create darks.

I have 2  ToupTek G3CMOS06300 one mono, one color and a Canon 2000D.

Now I want to add darks.

I have errors in both no 1. and  no 3.

##########################
no 1.): - not working 
##########################
(indi-allsky) fritz@zx02:~/indi-allsky$ ./darks.py sigmaclip
2023-07-01 17:50:40,318 [INFO] MainProcess indi.__init__() #129: creating an instance of IndiClient
2023-07-01 17:50:40,318 [INFO] MainProcess indi.__init__() #136: PyIndi version: 2.0.2
2023-07-01 17:50:40,318 [INFO] MainProcess darks._initialize() #191: Connecting to indiserver
INDI::BaseClient::connectServer: creating new connection...
2023-07-01 17:50:40,322 [INFO] MainProcess indi.serverConnected() #334: Server connected (192.168.10.41:7624)
2023-07-01 17:50:40,334 [INFO] MainProcess indi.newDevice() #204: new device Telescope Simulator
2023-07-01 17:50:40,388 [INFO] MainProcess indi.newDevice() #204: new device ToupTek G3CMOS06300KPA(USB2.0)
2023-07-01 17:50:48,331 [INFO] MainProcess indi._findCcds() #574: Searching for available cameras
2023-07-01 17:50:48,331 [INFO] MainProcess indi._findCcds() #585:  Detected ToupTek G3CMOS06300KPA(USB2.0)
2023-07-01 17:50:48,332 [INFO] MainProcess indi.findCcd() #594: Found 1 CCDs
2023-07-01 17:50:48,332 [WARNING] MainProcess darks._initialize() #214: Connecting to device ToupTek G3CMOS06300KPA(USB2.0)
2023-07-01 17:50:48,534 [INFO] MainProcess indi.getCcdInfo() #519: Detecting bayer pattern
2023-07-01 17:50:48,950 [INFO] MainProcess miscDb.addCamera() #81: Camera DB ID: 1
2023-07-01 17:50:48,950 [INFO] MainProcess indi.configureDevice() #697: Setting switch DEBUG
2023-07-01 17:50:49,051 [INFO] MainProcess indi.set_switch() #1265: Enabling Disable (DISABLE)
2023-07-01 17:50:50,053 [INFO] MainProcess indi.updateCcdBlobMode() #396: Set BLOB mode
2023-07-01 17:50:51,055 [INFO] MainProcess indi.configureDevice() #697: Setting switch CCD_FRAME_TYPE
2023-07-01 17:50:51,156 [INFO] MainProcess indi.set_switch() #1265: Enabling Dark (FRAME_DARK)
2023-07-01 17:50:51,735 [INFO] MainProcess indi.newMessage() #331: new Message 2023-07-01T15:50:51: [WARNING] The CCD does not have a shutter. Cover the camera in order to take a dark frame.
2023-07-01 17:50:52,158 [WARNING] MainProcess indi.disableCcdCooler() #882: Disabling CCD cooling
2023-07-01 17:50:54,167 [WARNING] MainProcess indi.disableCcdCooler() #887: Cooling not supported
2023-07-01 17:50:54,168 [WARNING] MainProcess darks._run() #626: ****** IF THE CCD COOLER WAS ENABLED, YOU MAY CONSIDER STOPPING THIS UNTIL THE SENSOR HAS WARMED ******
2023-07-01 17:51:02,172 [WARNING] MainProcess indi.setCcdGain() #1062: Setting CCD gain to 100
2023-07-01 17:51:02,172 [INFO] MainProcess indi.configureDevice() #702: Setting property (number) CCD_CONTROLS
2023-07-01 17:51:02,273 [INFO] MainProcess indi.set_number() #1237: Setting Gain = 100
2023-07-01 17:51:02,284 [INFO] MainProcess indi.newMessage() #331: new Message 2023-07-01T15:51:02: [INFO] Limiting frame rate to 63 FPS
2023-07-01 17:51:03,426 [WARNING] MainProcess indi.setCcdBinning() #1148: Setting CCD binning to (1, 1)
2023-07-01 17:51:03,426 [INFO] MainProcess indi.configureDevice() #702: Setting property (number) CCD_BINNING
2023-07-01 17:51:03,527 [INFO] MainProcess indi.set_number() #1237: Setting X = 1
2023-07-01 17:51:03,527 [INFO] MainProcess indi.set_number() #1237: Setting Y = 1
2023-07-01 17:51:04,681 [INFO] MainProcess darks._take_exposures() #675: Temp folder: /tmp/tmpliizf7ro
2023-07-01 17:51:04,682 [INFO] MainProcess darks.shoot() #323: Taking 60.00000000 s exposure (gain 100)
2023-07-01 17:51:04,782 [INFO] MainProcess indi.set_number() #1237: Setting Duration (s) = 60.0
2023-07-01 17:52:09,769 [INFO] MainProcess darks._take_exposures() #690: Exposure received in 65.0872 s
--- Logging error ---
Traceback (most recent call last):
  File "/usr/lib/python3.10/logging/__init__.py", line 1100, in emit
    msg = self.format(record)
  File "/usr/lib/python3.10/logging/__init__.py", line 943, in format
    return fmt.format(record)
  File "/usr/lib/python3.10/logging/__init__.py", line 678, in format
    record.message = record.getMessage()
  File "/usr/lib/python3.10/logging/__init__.py", line 368, in getMessage
    msg = msg % self.args
TypeError: only size-1 arrays can be converted to Python scalars
Call stack:
  File "/home/fritz/indi-allsky/./darks.py", line 90, in <module>
    action_func()
  File "/home/fritz/indi-allsky/indi_allsky/darks.py", line 472, in sigmaclip
    self._sigmaclip()
  File "/home/fritz/indi-allsky/indi_allsky/darks.py", line 479, in _sigmaclip
    self._run(IndiAllSkyDarksSigmaClip)
  File "/home/fritz/indi-allsky/indi_allsky/darks.py", line 638, in _run
    self._take_exposures(exposure, dark_filename_t, bpm_filename_t, ccd_bits, stacking_class)
[... 129 lines cut ...]
2023-07-01 17:57:37,455 [WARNING] MainProcess indi.setCcdBinning() #1141: Setting CCD binning to (1, 1)
2023-07-01 17:57:37,455 [WARNING] MainProcess indi.setCcdBinning() #1173: indi_canon_ccd does not support bin settings
2023-07-01 17:57:37,457 [INFO] MainProcess darks._take_exposures() #672: Temp folder: /tmp/tmpzw1khwud
2023-07-01 17:57:37,457 [INFO] MainProcess darks.shoot() #320: Taking 60.00000000 s exposure (gain 100)
2023-07-01 17:57:37,558 [INFO] MainProcess indi.set_number() #1230: Setting Duration (s) = 60.0
2023-07-01 17:57:37,610 [INFO] MainProcess indi.newMessage() #324: new Message 2023-07-01T15:57:37: [INFO] Starting 60 seconds exposure.
2023-07-01 17:58:39,577 [INFO] MainProcess indi.newMessage() #324: new Message 2023-07-01T15:58:37: [WARNING] Could not wait for event.
2023-07-01 17:58:39,577 [INFO] MainProcess indi.newMessage() #324: new Message 2023-07-01T15:58:37: [WARNING] Could not wait for event.
2023-07-01 17:58:39,577 [INFO] MainProcess indi.newMessage() #324: new Message 2023-07-01T15:58:37: [WARNING] Could not wait for event.
2023-07-01 17:58:39,578 [INFO] MainProcess indi.newMessage() #324: new Message 2023-07-01T15:58:37: [WARNING] Could not wait for event.
2023-07-01 17:58:39,578 [INFO] MainProcess indi.newMessage() #324: new Message 2023-07-01T15:58:37: [WARNING] Could not wait for event.
2023-07-01 17:58:39,578 [INFO] MainProcess indi.newMessage() #324: new Message 2023-07-01T15:58:37: [WARNING] Could not wait for event.
2023-07-01 17:58:39,578 [INFO] MainProcess indi.newMessage() #324: new Message 2023-07-01T15:58:37: [WARNING] Could not wait for event.
2023-07-01 17:58:39,578 [INFO] MainProcess indi.newMessage() #324: new Message 2023-07-01T15:58:37: [WARNING] Could not wait for event.
2023-07-01 17:58:39,578 [INFO] MainProcess indi.newMessage() #324: new Message 2023-07-01T15:58:37: [WARNING] Could not wait for event.
2023-07-01 17:58:39,578 [INFO] MainProcess indi.newMessage() #324: new Message 2023-07-01T15:58:37: [WARNING] Could not wait for event.
2023-07-01 17:58:39,578 [INFO] MainProcess indi.newMessage() #324: new Message 2023-07-01T15:58:37: [ERROR] Exposure failed to save image... Unspecified error

Any idea what could be wrong?

2 looks ok, 1 and 3 not.

the strange temp reading in 1 is not the issue, as same on 2 and there it is ok.

3 looks like an error in saving the image. This is a Canon 2000d

aaronwmorris, 2023-07-01:
I am assuming the Touptek camera that is generating the exception is the color camera.  It is possible the camera is set to INDI_RGB mode instead of INDI_RAW mode.  This is also the default for Altair cameras which are related to Touptek.

Try adding the following setting to set it to RAW mode....

```
{
    "PROPERTIES": {},
    "SWITCHES": {
        "CCD_CAPTURE_FORMAT": {
            "on": [
                "INDI_RAW"
            ]
        }
    }
}
```

As for the Canon, I am not sure.  Is the Camera set to manual mode?

jkoenig72, 2023-07-01:
Cool will try that.

About the Canon - seems it get a bit wet inside the case. I see error when
I manually take a photo and the screen is not working.

Time to open a cam! 😎

jkoenig72, 2023-07-01:
Works, now taking darks... Thanks!

What came to my mind: would it not be a nice feature to add temperature via
the command line when creating darks? You can check the temp via an
external sensor and then create the matching dark. By doing so over time, a
good dark library could be created.

Then feed in the external temp via your existing feature via a script and
the right dark is selected. Will not as good as a real built-in sensor but
could be an improvement for those cams.

?

jkoenig72, 2023-07-01:
Looks like you have something like this already- 

        # query external temperature if camera does not return temperature
        if temp_val < -100.0 and self.config.get('CCD_TEMP_SCRIPT'):

Cool.

aaronwmorris, 2023-07-01:
You can certainly use the temperature script feature.  There are some example scripts (shell and python) in the misc/ folder.

aaronwmorris, 2023-07-11:
@interplanetarychris What version of INDI are you running on your indiserver?

You can usually query this with `pkg-config --modversion libindi`.  If you are using a remote indiserver, make sure this is from the remote server.

interplanetarychris, 2023-07-11:
pkg-config --modversion libindi

1.9.7

Linux allskypi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

This is on a Raspberry Pi 4B with 8GB RAM.

aaronwmorris, 2023-07-11:
Looks like you are using the astroberry repo.  The timeout is probably a firmware issue with your camera.  The only way to upgrade this is to install a newer version of INDI.

You can clone the indi-allsky on your raspberry pi and run `./misc/build_indi.sh` to build the latest version on your Pi.  You will have to uninstall the existing version to proceed.  Compiling usually takes 20-30 minutes on a Pi4.

https://github.com/aaronwmorris/indi-allsky/wiki/Uninstall-indi-repository

aaronwmorris, 2023-07-11:
Also, fully reboot your Pi once you have a new version of INDI installed.

interplanetarychris, 2023-07-12:
The build of indi solved the problem (but took 4 hours on my Raspberry Pi 4b).  I submitted https://github.com/aaronwmorris/indi-allsky/pull/834 with some refinements to show progress in collecting darks.

aaronwmorris, 2023-07-12:
The compile of INDI should have taken no longer than about 40 minutes on a Pi4.  Do you have a heat sink and/or fan installed?  Throttling would explain that level of performance.

aaronwmorris, 2023-08-24:
Closing this for now.  Feel free to re-open if necessary.

closed 2023-08-24 by aaronwmorris