aaronwmorris/indi-allsky #833: Disk use suddenly increasing following update to #826 or #828

vastmeridian, 2023-07-10:
I'm suddenly seeing my disk space being used much more rapidly since updating the system.

[image]

I'll start looking round to see what has triggered the change, but if you have already found the iissue, I would be grateful if you could let me know.

aaronwmorris, 2023-07-10:
It sounds like the images might have stopped processing.  I have not changed anything recently that would affect space like that.

vastmeridian, 2023-07-10:
And a reboot has sorted everything out - including removing the .fit files.

I'll keep an eye on things over the next few days and if it's all sorted, I'll close the ticket.

vastmeridian, 2023-07-12:
The camera has been left untouched for a day or so and the same thing is happening:
[image]
I've manually cleared the directory of the .fit files but something is upsetting indi-allsky.  During the day it is now saying Daytime Timelapse is Disables and no picture is visible.

Looking for ERROR in the logs gives me
[ERROR examples.txt](https://github.com/aaronwmorris/indi-allsky/files/12032972/ERROR.examples.txt)

The leadup to 04:21:04 gives 
[Leadup ToERROR.txt](https://github.com/aaronwmorris/indi-allsky/files/12032973/Leadup.ToERROR.txt)

Can you shed any light on this at all?  Is there any info I can supply you with?

aaronwmorris, 2023-07-12:
This is helpful, let me research this.

aaronwmorris, 2023-07-12:
The problem is with the metadata upload.  I am adding a fix but it may be a few days before I merge my code.  Disabling metadata uploading will correct the issue for now.

aaronwmorris, 2023-07-12:
Nevermind.  Merged #835 to fix.  Let me know if that helps.

vastmeridian, 2023-07-13:
Thanks Aaron
I've updated, rebooted and we're running again.  Currently no changes to config, so I'll leave it going but keep an eye on disk space.  Judging by the disk usage sawtooth over the past few days we won't need to wait long.