aaronwmorris/indi-allsky #846: Web Browsers

mshetzer, 2023-07-19:
I am trying to access Allsky images from an older ipad to display the imagery.  The site loads, however under latest, I receive 'No Image for 15 Minutes', when I other browsers display images.  Any recommendations or setup changes I can make to have this work in the older browsers?

Also, is there anyway to change the web link so it is not labeled "Not Secure"

Thank you,
Matt

aaronwmorris, 2023-07-19:
Does this older iPad happen to have an internet connection via a SIM card?

mshetzer, 2023-07-19:
On wifi local.

aaronwmorris, 2023-07-19:
What version of indi-allsky are you running?

aaronwmorris, 2023-07-19:
That helps.  I am assuming this is a problem with jquery.

Supposedly, jquery is supposed to support Safari all the way back to iOS 7.  The only iPad that does not support at least this level is the first gen iPad.  What iPad model and iOS are you running that has this problem?

mshetzer, 2023-07-19:
IOS version 12.5.7

Ipad Mini 2 model MF084LL/A

 

I’ve tried Safari, and Chrome, however I can not download the lastest due to the IOS not upgradable any more.  Glad to use another browser if that works.

 

Matt

aaronwmorris, 2023-07-20:
I am still investigating this.  You may try updating indi-allsky to the latest.  I did a little work to the index page since 7.0, however, I do not think this will correct your issue.

The javascript for loading the images is fairly simple.  Just an AJAX GET request to find the latest image.

As for the "Not secure" issue, this is due to indi-allsky using a self-signed certificate.  The browser will not report "secure" unless a certificate signed by a trusted certificate authority is used.  There are ways to do this, even for free, but that is outside the scope of what I can provide.

It is possible the first issue is tied to the "not secure" problem.  The Safari browser might be blocking the AJAX requests because the certificate is not trusted.  Once you manually approve browsing the site, this should **not** be the case, but it would not surprise me if it is.

mshetzer, 2023-07-24:
Thanks for looking into it.  I do get through the 'Not secure' message, however the screen shows everything normal, just not the images.

aaronwmorris, 2023-07-24:
You might see if you can find a way to add the self-signed certificate to the trust store on the iPad.  It is possible that will fix the issue.

It should be safe to do this.  The certificate on your indi-allsky was generated during the install and is unique to you, so there is really no danger in doing this.

aaronwmorris, 2023-10-31:
I am going to close this for now.  I am not sure there is anything I can do here, but this is good information if someone needs it.  Please re-open if anything else pops up.

closed 2023-10-31 by aaronwmorris