aaronwmorris/indi-allsky #862: Installing Indi-allsky error message

Startupa, 2023-08-02:
I'Ive tried installling Indi_allsky on a RPI4 a couple of times but continue to get the message below when trying to do ./setup.sh. First of all the pi responds saying I need to use ./misc/build_indi.sh then when I type in that I get:

"INDI 3rd party drivers will not build properly on Raspbian 11 32-bit
You must add the following parameter to /boot/config.text and reboot:
arm 64bit=0"

I'm completely new with linux and I don't know how to do this.

In your instructions, it refers to having to compile using build-indi.sh, should I be doing that first?

Thanks
Anne

aaronwmorris, 2023-08-02:
Yes, this error is specific to the 32-bit version of Raspbian.  You would not get this error on the 64-bit install.

The background on this is the newest versions of 32-bit Raspbian are running a 64-bit kernel with all of the applications compiled as 32-bit.  This disparity causes the indi compile to fail.  The config above switches the kernel to the 32-bit version.

Startupa, 2023-08-03:
Good morning
I've run the commands as stated above.  Indi-allsky seems to have set itself up but I can't run the two systemctl commands. Bash tells me the services can't be found. I then tried stopping the service, which said the service wasn't loaded, then followed through the manual operation section. No such file or directory when I tried the vertualenv bit.  The allsky.py run command brought the message below up.

File "/home/pi/indi-allsky/./allsky.py", line 3, in <module>
    import indi_allsky
  File "/home/pi/indi-allsky/indi_allsky/__init__.py", line 3, in <module>
    from .allsky import IndiAllSky
  File "/home/pi/indi-allsky/indi_allsky/allsky.py", line 7, in <module>
    import psutil
ModuleNotFoundError: No module named 'psutil'

The pi is running a headless version of the 32bit OS and I'm using Putty to connect to it.

aaronwmorris, 2023-08-03:
You should just need to activate the virtualenv from the git checkout folder.

```
cd indi-allsky

source virtualenv/indi-allsky/bin/activate

./allsky.py run
```

Startupa, 2023-08-03:
It's still not running the source virtualenv command, error message -bash: virtualenv/indi-allsky/bin/activate: No such file or directory

Sorry to keep pestering you.

aaronwmorris, 2023-08-03:
It is no problem.  I will help as long as I can.

When you are in the indi-allsky folder, what do you see.  Can you show me the out put of the following commands:

```
ls -l
ls -l virtualenv
```

aaronwmorris, 2023-08-03:
Ah, I think you just need to run the `./setup.sh` script until it gets to completion.  That will setup the virtualenv for you.

Startupa, 2023-08-03:
Ok, I'll try in the morning. It's getting a bit late now.

Startupa, 2023-08-04:
Hi
I've got indi-allsky installed and I can access the web interface using firefox. The camera isn't showing though. It's also not on the list of your supported Altair cameras, possibly because they discontinued it a while ago and replaced it with a newer model. 

It's a GPCAM130 mono, a version 1. It works fine with the same driver as the rest of their cameras. It's not the directshow driver model, though I've also got one of those.) It works fine in Windows apps using the Altair driver and under Ascom in other software. I selected 1, indi, as the camera driver and I've put the camera make in the camer but there's no option to tell it which  model it is. I'm pretty sure it doesn't allow binning or changing the default gain setting.

Also, I assume, because I can't see the camera in your software, that's why I can't find any live view controls at all.

Anne

aaronwmorris, 2023-08-04:
I think there is a good chance of the camera being supported.

The camera has to be plugged in when the indiserver starts with the indi_altair_ccd driver.  You may try leaving everything plugged in and fully rebooting.  The indiserver starts first and then the indi-allsky capture process will start 2 minutes after boot.

You can check the logs at `/var/log/indi-allsky/indi-allsky.log`

Startupa, 2023-08-04:
Hi again. I didn't have the camera plugged in when I started the rpi this morning but I did have it plugged in when I rebooted later on. I've also turned it on from scratch with the camera plugged after your email. I've screengrabbed the log from your app, I'll try to paste it in in a moment but, scanning through the log, Indi-allsky appears to be find the GPCAM, then losing it again immediately.
[image]

Anne

aaronwmorris, 2023-08-04:
Interesting... I am investigating

aaronwmorris, 2023-08-04:
I setup my [newer] Altair camera just to run a test and everything seems to be fine there.

I will admit that the error above does not make any sense.  The error should not be possible.  Obviously, it is possible :-) so I assume it there might be other exceptions indicating a problem.

If you could shutdown indi-allsky, run from the CLI for a few minutes, and then upload the logs here.

```
source virtualenv/indi-allsky/bin/activate

systemctl --user stop indi-allsky

systemctl --user restart indiserver

./allsky.py run
```

I could also SSH into your system if you are comfortable setting up a port forward.  If not, do not worry about it.

Startupa, 2023-08-05:
You won't believe it, I've connected everything up and it worked perfectly! I was going to test it with my Hypercam camera but thought I'd try a usb3 socket first. Then I powered it all down and tried the original usb2 socket. Worked there too! I can't believe I didn't plug the usb in properly, particularly as I tried several times! I did uncoil the usb cable this time though.

A couple of other questions:

How can I change the wifi network? When I'm away from home I use a travel router which I can plug my phone in if I need internet. I'm off to Portugal and may want to use it there.

I notice on one screen there are options to shut various things down, how do these work? Will it turn off the pi too?

Anne

aaronwmorris, 2023-08-05:
Great news!  I am glad it is now working for you.

I do not currently have a way to change the wifi network, so you would have to change it using raspi-config on the CLI.

As far as the shutdown, indi-allsky uses the programmatic interface for systemd to manage services, similar to using the systemctl command.  If you press the shutdown button, it will issue the shutdown command to the operating system.  After that, you can safely unplug the pi.  If you keep the LEDs enabled, you can see when it is ready to be powered off when the activity LED flashes at a regular rate.

Startupa, 2023-08-05:
I have alot to learn re RPIs! I've been using the PUTTY terminal to issue the shutdown command. The one in indi-allsky didn't seem to do anything. But I could have missed something.

If it ever stops raining tonight I was going to give it a try outside, if now I may just stick it in the conservatory and leave it on all night. One final question, if I leave everything now as set in indi-allsky, ie all the auto settings, will it just work? It seems to save images by date etc, I assume it's putting them somewhere!

Thank you for your help.
Anne

aaronwmorris, 2023-08-05:
Just make sure you check the two checkboxes by the shutdown button.  Those are there to prevent accidental clicks.  It should say "Job submitted" at the bottom if it works.

Everything should "just work" without any additional settings.  indi-allsky is expected to run automatically at startup (within 2 minutes).  Images and videos are stored in `/var/www/html/allsky/images/`.

Startupa, 2023-08-06:
Everything worked fine last night and I had quite a few hours of images this morning. Much to my surprise as I couldn't connect to the Pi as it was out of range of the router. So I had to trust it. I had clouds, more clouds, hazy moon and rain! Occasional stars.

I've also set up the 128gb sd card which works but I haven't used it with the camera yet as that was drying off after the rain this morning. It's also on the wifi extender so should be seen when it's outside.
Thanks for your help.
Anne

aaronwmorris, 2023-10-31:
Closing this for now.  Let me know if I can help with anything else.

closed 2023-10-31 by aaronwmorris