aaronwmorris/indi-allsky #863: New install Error - Astroberry apt-key is deprecated

dxfiles-uk, 2023-08-02:
When installing on a new Pi4 (8gb with 128Gb Card), I receive the following error.

###############################################
### Welcome to the indi-allsky setup script ###
###############################################

Distribution: Debian
Release: 11
Arch: aarch64
Bits: 64

INDI_DRIVER_PATH: /usr/bin
INDISERVER_SERVICE_NAME: indiserver
ALLSKY_SERVICE_NAME: indi-allsky
GUNICORN_SERVICE_NAME: gunicorn-indi-allsky
ALLSKY_ETC: /etc/indi-allsky
HTDOCS_FOLDER: /var/www/html/allsky
DB_FOLDER: /var/lib/indi-allsky
DB_FILE: /var/lib/indi-allsky/indi-allsky.sqlite
INSTALL_INDI: true
HTTP_PORT: 80
HTTPS_PORT: 443

Setup proceeding in 10 seconds... (control-c to cancel)

indi-allsky supports the following camera interfaces.

indi: For astro/planetary cameras normally connected via USB
libcamera: supports cameras connected via CSI interface on Raspberry Pi SoCs
indi_passive: Connect a second instance of indi-allsky to an existing indi-allsky indiserver

1) indi
2) libcamera
3) indi_passive
Select a camera interface: 1

Fixing git checkout permissions
**** Installing packages... ****

Current default time zone: 'Europe/London'
Local time is now:      Wed Aug  2 19:17:08 BST 2023.
Universal Time is now:  Wed Aug  2 18:17:08 UTC 2023.

Installing INDI via Astroberry repository
--2023-08-02 19:17:08--  https://www.astroberry.io/repo/key
Resolving www.astroberry.io (www.astroberry.io)... 46.105.204.23
Connecting to www.astroberry.io (www.astroberry.io)|46.105.204.23|:443... connected.
Warning: apt-key is deprecated. Manage keyring files in trusted.gpg.d instead (see apt-key(8)).
ERROR: The certificate of ‘www.astroberry.io’ is not trusted.
ERROR: The certificate of ‘www.astroberry.io’ has expired.
The certificate has expired
gpg: no valid OpenPGP data found.

The script exited abnormally, please try to run again...

aaronwmorris, 2023-08-02:
Merged #864 to remove support for the Astroberry repository.  The Astroberry project is basically abandoned so this was inevitable.

Remove the astroberry repo config:
```
sudo rm -f /etc/apt/sources.list.d/astroberry.list
```

Pull down the latest code and remove the astroberry repo config from your system and re-run setup.sh.  You will have to build INDI using the `./misc/build_indi.sh`.  It takes about 40 minutes on a Raspi 4.

dxfiles-uk, 2023-08-02:
> Merged #864 to remove support for the Astroberry repository. The Astroberry project is basically abandoned so this was inevitable.
>
> Remove the astroberry repo config:
>
> ```
> sudo rm -f [...]

The instructions worked great - issue resolved.

dxfiles-uk, 2023-08-02:
Aaron's instructions worked perfectly. (Thanks)
I think the next build will need the astroberry repository parts removed

closed 2023-08-02 by dxfiles-uk