aaronwmorris/indi-allsky #872: RPI HQ grey image

mschrock, 2023-08-08:
Fresh install. 7.4
Raspberry Pi 3 Model B Rev 1.2
Raspbian Lite 11 64bit
Rpi HQ camera, libcamera_imx477 driver.
Completely default config, all I get is gray image, not sure what I am doing wrong.

[image]

trying to figure out if there is log that I can pull.

mschrock, 2023-08-09:
Well, adding the following two lines to /boot/config.txt:
dtoverlay=imx477
gpu_mem=128

and a restart not I have images.   Sorry for being the noob.

aaronwmorris, 2023-08-09:
I just wanted to reopen this.  You should be able to safely set the GPU memory to 32MB for the imx477.

The reason for the gray image is due to image scaling algorithm in indi-allsky.  This is from the FAQ:

https://github.com/aaronwmorris/indi-allsky/wiki/FAQ

Why is my image gray?

This can happen with certain camera models at very low exposures. indi-allsky starts taking exposures at the lowest possible exposure settings by default. The data received by indi-allsky is normally RAW data and may be in 8, 10, 12, 14, or 16 bit format--the bit depth must be detected in real time. Newer cameras with better noise control and hot pixel suppression can confuse the scaling algorithm into thinking higher bit depths are lower and over-scales the image. This will result in a gray image.

The fix is setting the Default exposure for your camera to 0.01 or 0.001

mschrock, 2023-08-09:
Ok,  I commented out the gpu_mem=128 line in config.txt and made the suggested change to the exposure. It is working.  LOL, every megabyte of ram matters on these little computers. Thinking back I do remember it was trying to use a very low exposure number (lots of zeros), and I assume the image darkened and the exposure time increased coincidently as my edits.  I didn't think to look in the wiki, I did search the issues here, my apologies. This can be closed.

Probably not the place or time: I really like the modular implimentation of this allsky software, especially having the ability to split the operation between two computers.   Let the Pi do the just capturing, and I spun up a VM (safe in the basement) to do the images/timelapse/etc.   No need to put all that processing on the pi on the roof, YAY! full res timelapses.

aaronwmorris, 2023-08-09:
I am glad things are working for you.  I just happened to recognize the problem.

The wiki has grown to the point where it is difficult to find things sometimes.  I believe the default GPU memory is 64MB, so you can set `gpu_mem=32` and save an extra 32MB of RAM.

closed 2023-08-11 by aaronwmorris