aaronwmorris/indi-allsky #877: Exposure period values don't make sense

milesnz, 2023-08-11:
Hi Aaron,
When using the following settings I get the error as shown in the screenshot which don't make sense to me.
If this is not a bug would you mind explaining the logic to me please.
I was expecting to be able to set my day period different to the night period, and also to be able to make them less than the maximum exposure?

regards
Miles

Max Exposure = 60
        error "Exposure period cannot be greater than the night period"
Exposure period night = 30
        error "Period is less than maximum exposure"
Exposure period day = 30

 
[image]

aaronwmorris, 2023-08-11:
Update to the latest code and those checks are no longer in place.

jkoenig72 convinced me to remove those in https://github.com/aaronwmorris/indi-allsky/issues/795

milesnz, 2023-08-11:
awesome, thank you

closed 2023-08-24 by aaronwmorris