aaronwmorris/indi-allsky #878: Arducam IMX462 - Image quality not great

Vertikar, 2023-08-11:
### Issue 
Hi Aaron, have recently grabbed a second camera and Pi, but the images aren't looking the best, especially at night.
Suspect it might be because the Waveshare IMX462 has an onboard ISP, and maybe the Arducam doesnt?
This is the Arducam I've got, the B0423 model - https://www.arducam.com/product/arducam-2mp-ultra-low-light-starvis-imx462-motorized-ir-cut-camera-for-raspberry-pi/

### Version
7.4 - updated this morning
### Screenshots
**Daytime:**
From libcamera-still with no options
[image]
From indi-allsky
[image]

**Nighttime**
from libcamera-still with shutter set to 30, gain 0
[image]
from libcamera-still with shutter set to 30, gain 16 (the lens had fogged over by this stage)
[image]
From indi-allsky
[image]

aaronwmorris, 2023-08-11:
I've got an IMX462 module I purchased about a month ago and I have not even taken it out of the box.  Let me go find it and run some tests.

aaronwmorris, 2023-08-12:
I tested my IMX462 sensor and I saw basically the same results.  The non-AWB images look terrible.

There appears to be a problem taking long exposures with AWB disabled.  I noticed my 15s exposures were being received in less than 2 seconds.  I will do some more research on this.

If you enable AWB, the camera appears to function fairly normally, although, you will have to deal with the exposures taking a long time to complete at night.

Vertikar, 2023-08-12:
> There appears to be a problem taking long exposures with AWB disabled. I noticed my 15s exposures were being received in less than 2 seconds. I will do some more research on this.

Even with 7 second exposure times testing in a darker spot next to my desk I'm seeing a similar issue:

```
Aug 13 00:00:45 allsky2 [INFO] Capture-1/MainThread capture.saferun() #325: Exposure received in 1.5375 s (-5.6630)
Aug 13 00:00:45 allsky2 [ERROR] Capture-1/MainThread capture.saferun() #329: 7.2005s EXPOSURE RECEIVED IN 1.5375s.  POSSIBLE CAMERA PROBLEM.
Aug 13 00:00:45 allsky2 [WARNING] Capture-1/MainThread miscDb.addNotification() #705: Not adding existing notification
```
```
Aug 13 00:01:00 allsky2 [INFO] Capture-1/MainThread capture.saferun() #325: Exposure received in 1.4878 s (-5.7126)
Aug 13 00:01:00 allsky2 [ERROR] Capture-1/MainThread capture.saferun() #329: 7.2005s EXPOSURE RECEIVED IN 1.4878s.  POSSIBLE CAMERA PROBLEM.
Aug 13 00:01:00 allsky2 [WARNING] Capture-1/MainThread miscDb.addNotification() #705: Not adding existing notification
```

aaronwmorris, 2023-08-13:
I think your only option right now is to enable AWB.  It looks like the interface for the imx462 is just not currently stable in non-AWB mode.

Vertikar, 2023-08-14:
> I think your only option right now is to enable AWB. It looks like the interface for the imx462 is just not currently stable in non-AWB mode.

Ah that's a shame, as the camera has got such good potential.

Guessing that switching to indi, and using either https://github.com/indilib/indi-3rdparty/tree/master/indi-libcamera or https://github.com/scriptorron/indi_pylibcamera/tree/main wouldn't help either as they're still using the underlying libcamera interface?

aaronwmorris, 2023-08-14:
Yes, I would expect to see the same issue with both of those.  My guess is this is some sort of firmware or dtoverlay kernel issue related to the camera.  Some register is probably being set incorrectly.

There are actually quite a few posts I found regarding exposure issues with this camera.  The driver for this camera is actually related/based on the imx290.  I wonder if that camera has the same issue.

I have tested a number of the other modules, but this is the only time I have run across this.  The camera module v3 (imx708) is actually quite good.

Vertikar, 2023-08-14:
> Something else I have noticed is the camera has bad issues with horizontal banding with long exposures.

I'm seeing similar even on shorter exposures.. example from last night

[image]

> The driver for this camera is actually related/based on the imx290. I wonder if that camera has the same issue.

Yeah i did notice that when you run --list-cameras it just shows up as the 290:

```
karl@allsky2:~ $ libcamera-still --list-cameras
Available cameras
-----------------
0 : imx290 [1920x1080] (/base/soc/i2c0mux/i2c@1/imx290@1a)
    Modes: 'SRGGB10_CSI2P' : 1280x720 [60.00 fps - (320, 180)/1280x720 crop]
                             1920x1080 [60.00 fps - (0, 0)/1920x1080 crop]
           'SRGGB12_CSI2P' : 1280x720 [60.00 fps - (320, 180)/1280x720 crop]
                             1920x1080 [60.00 fps - (0, 0)/1920x1080 crop]
```

Vertikar, 2023-08-15:
Playing around some more as I'm not willing to give up yet, looks like raw mode could be a winner...
[image]

Will try it tonight and see how it goes

Vertikar, 2023-08-17:
Haven't had a chance to put it outside, but I've had some success with [indi_pylibcamera](https://github.com/scriptorron/indi_pylibcamera/tree/main), still not perfect, but better than libcamera on indi-allsky

daytime
[image]

nighttime
[image]

Vertikar, 2023-08-23:
So it seems I'm having the same issue with pylibcam via a remote indi server, so I'm guessing this is more of an issue in libcamera, or the camera itself.
@aaronwmorris any particular debugging that you know of I can run with libcamera to provide more info to raise an issue with?

Vertikar, 2023-08-25:
> Opened [raspberrypi/libcamera-apps#559](https://github.com/raspberrypi/libcamera-apps/issues/559)

Great, thanks, looks like they have a fix already too!

aaronwmorris, 2023-08-25:
I compiled libcamera and libcamera-apps from source with the patch does appear to fix the problem with long exposures.

Vertikar, 2023-08-29:
> Something else I have noticed is the camera has bad issues with horizontal banding with long exposures.

So now it looks like the exposure time issue looks to be sorted, want to see if we can tackle the banding issue with high gain or long exposures, prime example just now:

[image]

This is what indi-server reports back in its logs, not sure if that helps at all

```
2023-08-29T08:48:52: Driver indi_pylibcamera: root-INFO- exposure settings: CameraSettings: FastExposure=False, DoRaw=True, ProcSize=(1945, 1097), ` +        `RawMode={'label': '1920x1080 RGGB 10bit', 'size': (1920, 1080), 'true_size': (1920, 1080), 'camera_format': 'SRGGB10', 'bit_depth': 10, 'FITS_format': 'RGGB', 'binning': (1, 1)}, CameraControls={'ExposureTime': 97, 'AnalogueGain': 15.0, 'AeEnable': False, 'AeConstraintMode': <AeConstraintModeEnum.Normal: 0>, 'AeExposureMode': <AeExposureModeEnum.Normal: 0>, 'AeMeteringMode': <AeMeteringModeEnum.CentreWeighted: 0>, 'AwbEnable': False, 'AwbMode': <AwbModeEnum.Auto: 0>, 'Brightness': 0.0, 'ColourGains': (2.0, 2.0), 'Contrast': 1.0, 'ExposureValue': 0.0, 'NoiseReductionMode': <NoiseReductionModeEnum.Off: 0>, 'Saturation': 1.0, 'Sharpness': 0.0}
2023-08-29T08:48:52: Driver indi_pylibcamera: picamera2.picamera2-INFO- Camera started
2023-08-29T08:48:52: Driver indi_pylibcamera: root-INFO- camera started
2023-08-29T08:48:52: Driver indi_pylibcamera: root-INFO- got exposed frame
2023-08-29T08:48:52: Driver indi_pylibcamera: picamera2.picamera2-INFO- Camera stopped
2023-08-29T08:48:53: Driver indi_pylibcamera: root-INFO- Finished collecting snooped data.
2023-08-29T08:48:53: Driver indi_pylibcamera: root-INFO- preparing frame as BLOB: 4152960 bytes
2023-08-29T08:48:53: Driver indi_pylibcamera: root-INFO- sending BLOB
```

Do you think this is more of a libcamera, or camera driver issue?

aaronwmorris, 2023-08-30:
This may partly be indi-allsky's fault.  Not the banding itself, but maybe the severity is exaggerated a bit.

Normally, indi-allsky detects the bit depth of the camera by detecting the maximum value of the image data.  Using this method, noise and hot pixels actually help in determining this value.  Newer cameras and software have better noise control and can actually confuse the indi-allsky bit scaling algorithm.  If the camera is 12-bit, but it thinks the data is 10 bits, the data when scaled to 8 bits ends up being much brighter than it should be.

In PR #921, I have introduced a config option to explicitly configure a fixed bit depth instead of trying to automatically detect.  The imx462 is a 12-bit camera, so give that a try.

Vertikar, 2023-08-31:
> This may partly be indi-allsky's fault. Not the banding itself, but maybe the severity is exaggerated a bit.

Yeah, I can see that in KStars, found that when you run an indiserver, you can have multiple clients attached and they'll all receive the image that was requested. I've found another interesting issue from that, but I'll have a dig in the code and see what I can find.

> The imx462 is a 12-bit camera, so give that a try.
I gave that a try and got some very strange results. Keep in mind I'm still using a remote indiserver so not sure if that has any impact on it. What's interesting is that in the indiserver logs, and KStars options is that it seems to default to 10 bits in the options there, yet indi-allsky seems to think its receiving a 16-bit image.

Here's the resulting images:
[image]
[image]

And logs from the capture process:

```
 karl@kasse  ~  docker logs docker-capture.indi.allsky-1 -f -n 50
2023-08-31 22:00:03,961 [INFO] Capture-2/MainThread indi.updateCcdBlobMode() #396: Set BLOB mode
2023-08-31 22:00:03,961 [INFO] Capture-2/MainThread indi.configureDevice() #697: Setting switch CCD_FRAME_TYPE
2023-08-31 22:00:04,062 [INFO] Capture-2/MainThread indi.set_switch() #1265: Enabling Light (FRAME_LIGHT)
2023-08-31 22:00:05,066 [INFO] Capture-2/MainThread capture._initialize() #678: Minimum CCD exposure: 0.00001401
2023-08-31 22:00:05,066 [INFO] Capture-2/MainThread capture._initialize() #699: Maximum CCD exposure: 30.00000000
2023-08-31 22:00:05,066 [INFO] Capture-2/MainThread capture._initialize() #743: Default CCD exposure: 0.00100000
2023-08-31 22:00:05,066 [ERROR] Capture-2/MainThread capture._initialize() #769: CCD day gain below minimum, changing to 1
2023-08-31 22:00:07,353 [INFO] MainProcess/MainThread allsky._queueManualTasks() #1114: Checking for manually submitted tasks
2023-08-31 22:00:08,095 [INFO] Capture-2/MainThread capture.saferun() #207: Camera last ready: 0.0s
2023-08-31 22:00:08,095 [INFO] Capture-2/MainThread capture.saferun() #208: Exposure state: unset
2023-08-31 22:00:08,096 [INFO] Capture-2/MainThread capture.detectNight() #1264: Sun altitude: -47:15:56.3
2023-08-31 22:00:08,096 [INFO] Capture-2/MainThread capture.detectMoonMode() #1281: Moon altitude: 42:16:02.7, phase 99.6%
2023-08-31 22:00:08,096 [INFO] Capture-2/MainThread capture.detectMoonMode() #1285: Moon Mode conditions detected
2023-08-31 22:00:08,197 [INFO] Capture-2/MainThread indi.getCcdTemperature() #855: Sensor temperature: 0.0
2023-08-31 22:00:08,248 [INFO] Capture-2/MainThread capture.shoot() #1429: Taking 0.35032496 s exposure (gain 15)
2023-08-31 22:00:08,349 [INFO] Capture-2/MainThread indi.set_number() #1237: Setting Duration (s) = 0.35032495856285095
2023-08-31 22:00:08,349 [INFO] Capture-2/MainThread capture.saferun() #414: Total time since last exposure 0.1527 s
2023-08-31 22:00:11,120 [INFO] Capture-2/MainThread capture.saferun() #326: Exposure received in 2.8726 s (2.5223)
2023-08-31 22:00:11,338 [INFO] Image-2/MainThread image.add() #1766: Image bits: 16, cfa: RGGB
2023-08-31 22:00:11,343 [INFO] Image-2/MainThread image._detectBitDepth() #1848: Image max value: 26944
2023-08-31 22:00:11,343 [WARNING] Image-2/MainThread image._detectBitDepth() #1879: *** DETECTED BIT DEPTH (15) IS DIFFERENT FROM CONFIGURED BIT DEPTH (12) ***
2023-08-31 22:00:11,343 [INFO] Image-2/MainThread image._detectBitDepth() #1881: Overriding bit depth to 12 bits
2023-08-31 22:00:11,344 [INFO] Image-2/MainThread image._calibrate() #1927: Searching for bad pixel map: gain 15, exposure >= 0.4, temp >= 0.0c
2023-08-31 22:00:11,349 [WARNING] Image-2/MainThread image._calibrate() #1944: Temperature matched bad pixel map not found: 0.00c
2023-08-31 22:00:11,351 [WARNING] Image-2/MainThread image._calibrate() #1962: Bad Pixel Map not found: ccd1 16bit 0.3503250s gain 15 bin 1 0.00c
2023-08-31 22:00:11,352 [INFO] Image-2/MainThread image._calibrate() #1974: Searching for dark frame: gain 15, exposure >= 0.4, temp >= 0.0c
2023-08-31 22:00:11,355 [WARNING] Image-2/MainThread image._calibrate() #1991: Temperature matched dark not found: 0.00c
2023-08-31 22:00:11,357 [WARNING] Image-2/MainThread image._calibrate() #2009: Dark not found: ccd1 16bit 0.3503250s gain 15 bin 1 0.00c
2023-08-31 22:00:11,357 [INFO] Image-2/MainThread sqm.calculate() #21: Exposure: 0.350325, gain: 15
2023-08-31 22:00:11,357 [INFO] Image-2/MainThread sqm._generateSqmMask() #48: Generating mask based on SQM_ROI
2023-08-31 22:00:11,357 [WARNING] Image-2/MainThread sqm._generateSqmMask() #63: Using central ROI for SQM calculations
2023-08-31 22:00:11,360 [INFO] Image-2/MainThread sqm.calculate() #37: Raw SQM average: 13590.67
2023-08-31 22:00:11,360 [INFO] Image-2/MainThread sqm.calculate() #42: Weighted SQM average: 134716.38
2023-08-31 22:00:11,372 [INFO] Image-2/MainThread image.processImage() #369: Image: 1920 x 1080
2023-08-31 22:00:11,373 [INFO] Image-2/MainThread image._generateAduMask() #3660: Generating mask based on ADU_ROI
2023-08-31 22:00:11,373 [WARNING] Image-2/MainThread image._generateAduMask() #3675: Using central ROI for ADU calculations
2023-08-31 22:00:11,378 [INFO] Image-2/MainThread image._calculate_8bit_adu() #2102: ADU average: 13965.1 (872)
2023-08-31 22:00:11,379 [INFO] Image-2/MainThread stretch._generateNumpyMask() #175: Generating mask based on SQM_ROI
2023-08-31 22:00:11,380 [WARNING] Image-2/MainThread stretch._generateNumpyMask() #189: Using central ROI for blob calculations
[... cut ...]
2023-08-31T12:00:23: Driver indi_pylibcamera: picamera2.picamera2-INFO- Camera started
2023-08-31T12:00:23: Driver indi_pylibcamera: root-INFO- camera started
2023-08-31T12:00:23: Driver indi_pylibcamera: root-INFO- got exposed frame
2023-08-31T12:00:23: Driver indi_pylibcamera: picamera2.picamera2-INFO- Camera stopped
2023-08-31T12:00:23: Driver indi_pylibcamera: root-INFO- Finished collecting snooped data.
2023-08-31T12:00:24: Driver indi_pylibcamera: root-INFO- preparing frame as BLOB: 4152960 bytes
2023-08-31T12:00:24: Driver indi_pylibcamera: root-INFO- sending BLOB
^C
```

aaronwmorris, 2023-08-31:
Ah, indi_pylibcamera or the indiserver may be scaling the data up to 16-bits.  My experience is with libcamera-still which outputs 12-bit data.

I am not seeing the horizontal lines like that with my camera.

Vertikar, 2023-08-31:
> Ah, indi_pylibcamera or the indiserver may be scaling the data up to 16-bits. My experience is with libcamera-still which outputs 12-bit data.

Interesting, I'll see if I can flick back to directly interfacing via libcamera and see how that goes.

Vertikar, 2023-10-11:
Going to close this off now as it looks like the issues are upstream, but I'll reference it if needed in the future

closed 2023-10-11 by Vertikar