aaronwmorris/indi-allsky #881: error installind indi-allsky armbian

pierrebilb, 2023-08-12:
Hi,
i tried to install indi-allsky on a orange pi pc+2e with armbian 23.05.1 jammy with kernel 6.1.30 (latest on armbian.com) . I could install  indi server and drivers without problem but i have the following error:
` Building wheel for shapely (pyproject.toml) ... error
  error: subprocess-exited-with-error

  × Building wheel for shapely (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [105 lines of output]
      <string>:8: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
      Could not find geos-config executable. Either append the path to geos-config to PATH or manually provide the include_dirs, library_dilibraries and other link args for compiling against a GEOS version >=3.5.
      /tmp/pip-build-env-_2xvegtg/overlay/lib/python3.10/site-packages/setuptools/config/pyprojecttoml.py:66: _BetaConfiguration: Support f[tool.setuptools]` in `pyproject.toml` is still *beta*.
        config = read_configuration(filepath, True, ignore_option_errors, dist)
      running bdist_wheel`

Any idea whatis going wrong?
Thanks for help
Pierre

aaronwmorris, 2023-08-12:
Looks like there is not a pre-compiled wheel for shapely for your platform.  Merged #882 to add the development packages to be able to build shapely from source.

Just pull down the latest code and re-run setup.sh

pierrebilb, 2023-08-13:
That's it!
Thank you very much!!
Pierre

closed 2023-08-13 by pierrebilb