aaronwmorris/indi-allsky #884: running but no image

pierrebilb, 2023-08-13:
I have a qhylii on opi+2e running but no image is appearing on the web page.
Running manually gives me this:
`File "/home/pierre/indi-allsky/virtualenv/indi-allsky/lib/python3.10/site-packages/PIL/ImageFont.py", line 48, in <module>
    from . import _imagingft as core
ImportError: cannot import name '_imagingft' from 'PIL' (/home/pierre/indi-allsky/virtualenv/indi-allsky/lib/python3.10/site-packages/PIL/__init__.py)
2023-08-13 11:39:16,921 [ERROR] MainProcess/MainThread allsky._startImageWorker() #349: Image worker exception: Traceback (most recent call last):
2023-08-13 11:39:16,922 [ERROR] MainProcess/MainThread allsky._startImageWorker() #349: Image worker exception:   File "/home/pierre/indi-allsky/indi_allsky/ima          ge.py", line 234, in run
2023-08-13 11:39:16,923 [ERROR] MainProcess/MainThread allsky._startImageWorker() #349: Image worker exception:     self.saferun()
2023-08-13 11:39:16,924 [ERROR] MainProcess/MainThread allsky._startImageWorker() #349: Image worker exception:   File "/home/pierre/indi-allsky/indi_allsky/ima          ge.py", line 263, in saferun
2023-08-13 11:39:16,925 [ERROR] MainProcess/MainThread allsky._startImageWorker() #349: Image worker exception:     self.processImage(i_dict)
2023-08-13 11:39:16,926 [ERROR] MainProcess/MainThread allsky._startImageWorker() #349: Image worker exception:   File "/home/pierre/indi-allsky/indi_allsky/ima          ge.py", line 569, in processImage
2023-08-13 11:39:16,927 [ERROR] MainProcess/MainThread allsky._startImageWorker() #349: Image worker exception:     self.image_processor.label_image()
2023-08-13 11:39:16,928 [ERROR] MainProcess/MainThread allsky._startImageWorker() #349: Image worker exception:   File "/home/pierre/indi-allsky/indi_allsky/ima          ge.py", line 2768, in label_image
2023-08-13 11:39:16,929 [ERROR] MainProcess/MainThread allsky._startImageWorker() #349: Image worker exception:     self._label_image_pillow(i_ref)
2023-08-13 11:39:16,930 [ERROR] MainProcess/MainThread allsky._startImageWorker() #349: Image worker exception:   File "/home/pierre/indi-allsky/indi_allsky/ima          ge.py", line 3284, in _label_image_pillow
2023-08-13 11:39:16,930 [ERROR] MainProcess/MainThread allsky._startImageWorker() #349: Image worker exception:     self.drawText_pillow(
2023-08-13 11:39:16,931 [ERROR] MainProcess/MainThread allsky._startImageWorker() #349: Image worker exception:   File "/home/pierre/indi-allsky/indi_allsky/ima          ge.py", line 3370, in drawText_pillow
2023-08-13 11:39:16,932 [ERROR] MainProcess/MainThread allsky._startImageWorker() #349: Image worker exception:     font = ImageFont.truetype(str(font_file), fo          nt_size)
2023-08-13 11:39:16,933 [ERROR] MainProcess/MainThread allsky._startImageWorker() #349: Image worker exception:   File "/home/pierre/indi-allsky/virtualenv/indi          -allsky/lib/python3.10/site-packages/PIL/ImageFont.py", line 791, in truetype
2023-08-13 11:39:16,934 [ERROR] MainProcess/MainThread allsky._startImageWorker() #349: Image worker exception:     return freetype(font)
2023-08-13 11:39:16,935 [ERROR] MainProcess/MainThread allsky._startImageWorker() #349: Image worker exception:   File "/home/pierre/indi-allsky/virtualenv/indi          -allsky/lib/python3.10/site-packages/PIL/ImageFont.py", line 788, in freetype
2023-08-13 11:39:16,936 [ERROR] MainProcess/MainThread allsky._startImageWorker() #349: Image worker exception:     return FreeTypeFont(font, size, index, encod          ing, layout_engine)
2023-08-13 11:39:16,937 [ERROR] MainProcess/MainThread allsky._startImageWorker() #349: Image worker exception:   File "/home/pierre/indi-allsky/virtualenv/indi          -allsky/lib/python3.10/site-packages/PIL/ImageFont.py", line 198, in __init__
2023-08-13 11:39:16,938 [ERROR] MainProcess/MainThread allsky._startImageWorker() #349: Image worker exception:     if core.HAVE_RAQM:
2023-08-13 11:39:16,938 [ERROR] MainProcess/MainThread allsky._startImageWorker() #349: Image worker exception:   File "/home/pierre/indi-allsky/virtualenv/indi          -allsky/lib/python3.10/site-packages/PIL/_util.py", line 19, in __getattr__
2023-08-13 11:39:16,939 [ERROR] MainProcess/MainThread allsky._startImageWorker() #349: Image worker exception:     raise self.ex
2023-08-13 11:39:16,940 [ERROR] MainProcess/MainThread allsky._startImageWorker() #349: Image worker exception:   File "/home/pierre/indi-allsky/virtualenv/indi          -allsky/lib/python3.10/site-packages/PIL/ImageFont.py", line 48, in <module>
2023-08-13 11:39:16,941 [ERROR] MainProcess/MainThread allsky._startImageWorker() #349: Image worker exception:     from . import _imagingft as core
2023-08-13 11:39:16,942 [ERROR] MainProcess/MainThread allsky._startImageWorker() #349: Image worker exception: ImportError: cannot import name '_imagingft' fro          m 'PIL' (/home/pierre/indi-allsky/virtualenv/indi-allsky/lib/python3.10/site-packages/PIL/__init__.py)
2023-08-13 11:39:16,943 [ERROR] MainProcess/MainThread allsky._startImageWorker() #349: Image worker exception:
2023-08-13 11:39:16,944 [INFO] MainProcess/MainThread allsky._startImageWorker() #356: Starting Image-11 worker
`
Any clue?
Thanks for help
Cheers
Pierre

aaronwmorris, 2023-08-13:
Can you activate the virtualenv and show the version of Pillow installed?

```
source virtualenv/indi-allsky/bin/activate

pip freeze | grep -i pil
```

pierrebilb, 2023-08-13:
the command gives:
`(indi-allsky) pierre@orangepiplus2e:~/indi-allsky$ pip freeze | grep -i pil
Pillow==10.0.0
`

aaronwmorris, 2023-08-13:
I was finally able to simulate this error.  It looks like Pillow was compiled from source, but does not have any of the image interfaces.  I am investigating.

aaronwmorris, 2023-08-13:
Merged #886 to install additional development packages.

Go ahead and re-run setup.sh to install the additional development packages.

Once that is complete, it will take some manual intervention to recompile pillow with the additional sources.

```
source virtualenv/indi-allsky/bin/activate

# remove all compiled wheels
rm -fR ~/.cache/pip/wheels/

pip uninstall Pillow

pip install --upgrade Pillow --no-binary :all:
```

Not sure how long the compile will take, but this should fix your issue.

pierrebilb, 2023-08-13:
Ok,
i will try tonight..
Thank you very much!!
Pierre

pierrebilb, 2023-08-13:
That made the trick; it's running with images!
Thanks again for your support and for all your efforts to improve your program!!
Best regards
Pierre

closed 2023-08-13 by pierrebilb