aaronwmorris/indi-allsky #897: Docker issues

Vertikar, 2023-08-17:
### Issue 
Unable to use docker images with remote indiserver

Running into a few issues here, looks like there's an issue with the database which is causing issues with the capture container, and maybe the same issue is causing issues with the gunicorn container.

I did have a question about INDIALLSKY_IMAGE_FOLDER - this looks to just be used internally by the docker images, and doesn't need to exist, as there's a volume defined for it.

.env file used is below:

```
### copy this file to .env ###

# shellcheck disable=SC2034

TZ=Australia/Melbourne

INDIALLSKY_INDI_CCD_DRIVER=indi
INDIALLSKY_INDI_GPS_DRIVER=

# This folder needs to be shared to the capture, gunicorn, and nginx containers
INDIALLSKY_IMAGE_FOLDER=/var/www/html/allsky/images

INDIALLSKY_FLASK_AUTH_ALL_VIEWS=false
INDIALLSKY_FLASK_SECRET_KEY=<removed>
INDIALLSKY_FLASK_PASSWORD_KEY=<removed>

# Folder for temporary files in capture process.  Uncomment to activate
#CAPTURE_TMPDIR=/tmp

INDIALLSKY_WEB_USER=karl
INDIALLSKY_WEB_PASS=<removed>
INDIALLSKY_WEB_NAME=Karl
INDIALLSKY_WEB_EMAIL=<removed>

INDIALLSKY_MOSQUITTO_USER=karl
INDIALLSKY_MOSQUITTO_PASS=<removed>

# remember to set back to false after first run
INDIALLSKY_WEB_GENERATE_APIKEY=false

# Only used on the debian build
#INDIALLSKY_INDI_VERSION=2.0.2

INDIALLSKY_MARIADB_HOST=mariadb.indi.allsky
INDIALLSKY_MARIADB_PORT=3306
INDIALLSKY_MARIADB_SSL=false

MARIADB_RANDOM_ROOT_PASSWORD=yes
MARIADB_DATABASE=indi_allsky
MARIADB_USER=indi_allsky_own
MARIADB_PASSWORD=<removed>
```

### Version
commit 6103f95a9426d99d9877d5f89b626453bd66e936 (HEAD -> main, origin/main, origin/HEAD)
Merge: 3cf98300 39c999b0

### log outputs

#### gunicorn container errors: ####

```
[2023-08-17 11:54:29 +1000] [193] [INFO] Starting gunicorn 21.2.0
[2023-08-17 11:54:29 +1000] [193] [INFO] Listening at: http://0.0.0.0:8000 (193)
[2023-08-17 11:54:29 +1000] [193] [INFO] Using worker: gthread
[2023-08-17 11:54:29 +1000] [195] [INFO] Booting worker with pid: 195
--- Logging error ---
Traceback (most recent call last):
  File "/home/allsky/venv/lib/python3.10/site-packages/mysql/connector/connection_cext.py", line 633, in cmd_query
    self._cmysql.query(
_mysql_connector.MySQLInterfaceError: Unknown collation: 'utf8mb4_0900_ai_ci'

The above exception was the direct cause of the following exception:
```
also seeing the below in the gunicorn container, which maybe related to the above:

```
  File "/home/allsky/indi-allsky/indi_allsky/flask/base_views.py", line 88, in getCameraById
    app.logger.warning('No cameras are defined')
Message: 'No cameras are defined'
Arguments: ()
```

#### capture container error: ####
```
2023-08-17 12:39:56,958 [WARNING] MainProcess/MainThread allsky._periodic_tasks() #1190: Periodic tasks triggered
2023-08-17 12:39:57,060 [INFO] Capture-208/MainThread indi.serverConnected() #334: Server connected (indiserver-imx462.kn-mck:7624)
2023-08-17 12:39:57,071 [INFO] Capture-208/Dummy-5 indi.newDevice() #204: new device indi_pylibcamera
2023-08-17 12:40:05,094 [INFO] Capture-208/MainThread indi._findCcds() #574: Searching for available cameras
2023-08-17 12:40:05,094 [INFO] Capture-208/MainThread indi._findCcds() #585:  Detected indi_pylibcamera
2023-08-17 12:40:05,094 [INFO] Capture-208/MainThread indi.findCcd() #594: Found 1 CCDs
2023-08-17 12:40:05,094 [INFO] Capture-208/MainThread indi._findTelescopes() #616: Searching for available telescopes/mounts
2023-08-17 12:40:05,095 [INFO] Capture-208/MainThread indi.findTelescope() #636: Found 0 Telescopess
2023-08-17 12:40:05,095 [ERROR] Capture-208/MainThread indi.findTelescope() #643: No telescopes found
2023-08-17 12:40:05,095 [INFO] Capture-208/MainThread indi._findGpss() #649: Searching for available GPS interfaces
2023-08-17 12:40:05,095 [INFO] Capture-208/MainThread indi.findGps() #669: Found 0 GPSs
2023-08-17 12:40:05,095 [WARNING] Capture-208/MainThread capture._initialize() #479: Connecting to CCD device indi_pylibcamera
Process Capture-208:
Traceback (most recent call last):
  File "/home/allsky/venv/lib/python3.10/site-packages/mysql/connector/connection_cext.py", line 633, in cmd_query
    self._cmysql.query(
_mysql_connector.MySQLInterfaceError: Unknown collation: 'utf8mb4_0900_ai_ci'

The above exception was the direct cause of the following exception:
<snip>
 File "/home/allsky/venv/lib/python3.10/site-packages/mysql/connector/connection_cext.py", line 641, in cmd_query
    raise get_mysql_exception(
mysql.connector.errors.DatabaseError: 1273 (HY000): Unknown collation: 'utf8mb4_0900_ai_ci'
```
### Screenshots:

#### System info:
[image]

#### Config page 1
[image]

#### Camera info page - returns 500
[image]

aaronwmorris, 2023-08-17:
Earlier today I merged #894 to fix the docker issue.  The problem is in the mysql-connector driver.  It recently set a default collation that is not supported in MariaDB.

There are a couple of new env variables that need to be defined and hopefully you will be good to go.

Vertikar, 2023-08-17:
Oh great, I'll let you know how it goes

Vertikar, 2023-08-17:
Looks way better! Still seeing some errors in gunicorn, looks like it might be related to the image read time tracking you added for my issues with the IMX462:

  File "/home/allsky/indi-allsky/indi_allsky/flask/views.py", line 610, in get_objects
    'chart_data' : self.getChartData(history_seconds),
  File "/home/allsky/indi-allsky/indi_allsky/flask/views.py", line 749, in getChartData
    app.logger.info('Image read in %0.4f s', image_elapsed_s)
Message: 'Image read in %0.4f s'
Arguments: (0.035051584243774414,)
--- Logging error ---

Want me to raise a separate issue for that?

aaronwmorris, 2023-08-17:
No, I am not quite sure how to fix this error, but it is benign anyway.  It is trying to log to the system /dev/log socket, but it does not exist in a container.

Vertikar, 2023-08-17:
No probs, will close this issue off then.
Thanks!

closed 2023-08-17 by Vertikar