aaronwmorris/indi-allsky #924: Enhancement Request - Overnight Image Management

vastmeridian, 2023-09-03:
Would it be possible to have a feature (only available after admin login) that could allow the images/videos from poor nights to be removed from storage and thus saving disk space?  I know there's an ability to automatically delete files after a certain age, but perhaps there are spectacular days that you want to keep irrespective of this?

I was thinking that say, on the Timelapse page, reviewing nights of poor (i.e. cloudy/rainy/bright moon) visibility and ticking a box besides the keogram then clicking something like **Delete Selected**

Or have I missed a trick and this is already available?  I'm running version 7.1.

aaronwmorris, 2023-09-03:
Currently, there is only the ability to delete timelapses and keograms on the generate page, but no way to just delete a day or night's individual images.  I did just add the ability delete all daytime data, but I can see a use for this.

aaronwmorris, 2023-09-10:
In #937 I added the ability to delete all images for a specific day or night in the Generator form.

vastmeridian, 2023-09-13:
Thanks.  Testing this out now.
I can see why you've done it this way, but you do need two windows open to do this- one to see the images to confirm the keogram is suitably bad and the other to set up the delete.  Alternatively you need to have reviewed and made a note of the 'dead days' and then switch to generator view and go through the list one by one to delete the files.
Pity you can't have a thumbnail of the keogram by the days in the generator view to see which days you are interested in for the actions you're trying to carry out.

vastmeridian, 2023-09-13:
I think your action 'Delete Images *DANGER*' should read Delete all images on this date' or similar.  

'Delete Images *DANGER* **could** imply all images on the server.

vastmeridian, 2023-09-13:
Having chosen to delete all files for 1st September, the startrail and keogram are still visible and the timelapse is playable.  Trying to delete these files then results in this message:
[image]

vastmeridian, 2023-09-13:
Final point for tonight - how do we know if the images have gone?  The Timelapse screen still shows the presence of the timelapse and star trail (even plays the deleted ones) which is a bit bizarre given I have just 'deleted' the images.

aaronwmorris, 2023-09-13:
I am running some tests, and I saw the same error one time, however the "Not a valid choice" error is not something generated by my code, but something internally by wtforms.  Very weird.

You may want to try restarting the gunicorn-indi-allsky service (or just rebooting).

```
systemctl --user restart gunicorn-indi-allsky
```

vastmeridian, 2023-09-15:
We're good, thanks
And we're MUCH better off with regard to disk space:
[image]
But I'm still seeing timelapses, keograms and star trails from days I have deleted.  Are these in the database and away from the filesystem?

aaronwmorris, 2023-09-16:
When you delete images, it will only delete the individual timelapse frames.  You still have to delete the timelapse/keogram/startrails separately.  I can add another option to delete all files for a day.

vastmeridian, 2023-09-18:
OK - I'll go back and check this.  Thanks.

A 'delete all files' for a gven day would be a useful addition.  If I'm deleting the image files because of rain etc. then the timelapse/keogram/star trail aren't going to be much use either.  Thanks.

I've noticed in this build that the image processing seems to have changed.  I'll open a new issue #950.

vastmeridian, 2023-12-29:
Again, I think this can be closed now.  There are still issues (for me) about which days I have deleted as these days are not easily seen from the table, but I can get round this.
(for me, an indication by the date that there are no files for that day would help)

closed 2023-12-29 by vastmeridian

aaronwmorris, 2023-12-29:
The `Generate` view shows how many images are available for day/night for each day.