aaronwmorris/indi-allsky #926: Updated older allsky to new version, not capturing images

Vertikar, 2023-09-05:
Hey Aaron, found another weird one.
Might be something I've done, but I dug up an older SD card, and upgraded indi-allsky and now images aren't being captured.
Suspect its due to me messing around, and I'm tempted to blow it all away as I pull the images out manually easily enough if I want

### Issue:
Unable to capture images, even in focus mode

### Steps to recreate:
upgraded from older version then blew away the config with config.py flush.
Tried config.py bootstrap, but same issue.
Did another config.py flush, then ./setup.sh, and same issue.

### Logs
#### Focus mode:

```
Sep  5 04:55:18 rpi4 [WARNING] Capture-2/MainThread capture._periodic_tasks() #995: Periodic tasks triggered
Sep  5 04:55:19 rpi4 [INFO] Capture-2/MainThread capture.shoot() #1429: Taking -1.00000000 s exposure (gain 0)
Sep  5 04:55:19 rpi4 [INFO] Capture-2/MainThread libcamera.setCcdExposure() #235: image command: libcamera-still --immediate --nopreview --encoding jpg --quality 95 --gain 0 --shutter -1000000 --metadata /tmp/tmp1o21jlen.json --metadata-format json --awbgains 1,1 --output /tmp/tmp38vkrkdh.jpg
Sep  5 04:55:19 rpi4 [WARNING] Capture-2/MainThread capture.saferun() #407: *** FOCUS MODE ENABLED ***
Sep  5 04:55:19 rpi4 [INFO] Capture-2/MainThread capture.saferun() #414: Total time since last exposure 5.0349 s
Sep  5 04:55:20 rpi4 [ERROR] Capture-2/MainThread libcamera._processMetadata() #329: libcamera sensor temperature key not found
Sep  5 04:55:20 rpi4 [INFO] Capture-2/MainThread capture.saferun() #326: Exposure received in 2.1640 s (3.1640)
Sep  5 04:55:23 rpi4 [INFO] Capture-2/MainThread capture.shoot() #1429: Taking -1.00000000 s exposure (gain 0)
Sep  5 04:55:23 rpi4 [INFO] Capture-2/MainThread libcamera.setCcdExposure() #235: image command: libcamera-still --immediate --nopreview --encoding jpg --quality 95 --gain 0 --shutter -1000000 --metadata /tmp/tmp4n9_80ob.json --metadata-format json --awbgains 1,1 --output /tmp/tmp_i7sbzaa.jpg
Sep  5 04:55:23 rpi4 [WARNING] Capture-2/MainThread capture.saferun() #407: *** FOCUS MODE ENABLED ***
Sep  5 04:55:23 rpi4 [INFO] Capture-2/MainThread capture.saferun() #414: Total time since last exposure 5.0303 s
Sep  5 04:55:24 rpi4 [ERROR] Capture-2/MainThread libcamera._processMetadata() #329: libcamera sensor temperature key not found
Sep  5 04:55:24 rpi4 [INFO] Capture-2/MainThread capture.saferun() #326: Exposure received in 1.0413 s (2.0413)
Sep  5 04:55:24 rpi4 [ERROR] MainProcess/MainThread allsky._startImageWorker() #348: Image worker exception: Traceback (most recent call last):
Sep  5 04:55:24 rpi4 [ERROR] MainProcess/MainThread allsky._startImageWorker() #348: Image worker exception:   File "/opt/indi-allsky/indi_allsky/image.py", line 241, in run
Sep  5 04:55:24 rpi4 [ERROR] MainProcess/MainThread allsky._startImageWorker() #348: Image worker exception:     self.saferun()
Sep  5 04:55:24 rpi4 [ERROR] MainProcess/MainThread allsky._startImageWorker() #348: Image worker exception:   File "/opt/indi-allsky/indi_allsky/image.py", line 270, in saferun
Sep  5 04:55:24 rpi4 [ERROR] MainProcess/MainThread allsky._startImageWorker() #348: Image worker exception:     self.processImage(i_dict)
Sep  5 04:55:24 rpi4 [ERROR] MainProcess/MainThread allsky._startImageWorker() #348: Image worker exception:   File "/opt/indi-allsky/indi_allsky/image.py", line 329, in processImage
Sep  5 04:55:24 rpi4 [ERROR] MainProcess/MainThread allsky._startImageWorker() #348: Image worker exception:     camera = IndiAllSkyDbCameraTable.query\                                                                                                                                                        Sep  5 04:55:24 rpi4 [ERROR] MainProcess/MainThread allsky._startImageWorker() #348: Image worker exception:   File "/opt/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/sqlalchemy/orm/query.py", line 2869, in one
Sep  5 04:55:24 rpi4 [ERROR] MainProcess/MainThread allsky._startImageWorker() #348: Image worker exception:     return self._iter().one()                                                                                                                                                                      Sep  5 04:55:24 rpi4 [ERROR] MainProcess/MainThread allsky._startImageWorker() #348: Image worker exception:   File "/opt/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/sqlalchemy/engine/result.py", line 1510, in one
Sep  5 04:55:24 rpi4 [ERROR] MainProcess/MainThread allsky._startImageWorker() #348: Image worker exception:     return self._only_one_row(                                                                                                                                                                     Sep  5 04:55:24 rpi4 [ERROR] MainProcess/MainThread allsky._startImageWorker() #348: Image worker exception:   File "/opt/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/sqlalchemy/engine/result.py", line 562, in _only_one_row
Sep  5 04:55:24 rpi4 [ERROR] MainProcess/MainThread allsky._startImageWorker() #348: Image worker exception:     raise exc.NoResultFound(
Sep  5 04:55:24 rpi4 [ERROR] MainProcess/MainThread allsky._startImageWorker() #348: Image worker exception: sqlalchemy.exc.NoResultFound: No row was found when one was required
Sep  5 04:55:24 rpi4 [ERROR] MainProcess/MainThread allsky._startImageWorker() #348: Image worker exception:                                                                                                                                                                                                    Sep  5 04:55:24 rpi4 [INFO] MainProcess/MainThread allsky._startImageWorker() #355: Starting Image-113 worker
Sep  5 04:55:24 rpi4 [WARNING] MainProcess/MainThread image._load_detection_mask() #3570: No detection mask defined                                                                                                                                                                                             Sep  5 04:55:24 rpi4 [INFO] MainProcess/MainThread allsky._queueManualTasks() #1115: Checking for manually submitted tasks
Sep  5 04:55:25 rpi4 [INFO] Capture-2/MainThread capture.saferun() #207: Camera last ready: 1.0s
Sep  5 04:55:25 rpi4 [INFO] Capture-2/MainThread capture.saferun() #208: Exposure state: READY
Sep  5 04:55:25 rpi4 [INFO] Capture-2/MainThread capture.detectNight() #1264: Sun altitude: 58:04:01.8
Sep  5 04:55:25 rpi4 [INFO] Capture-2/MainThread capture.detectMoonMode() #1281: Moon altitude: -27:06:17.6, phase 71.9%
Sep  5 04:55:28 rpi4 [INFO] Capture-2/MainThread capture.shoot() #1429: Taking -1.00000000 s exposure (gain 0)
Sep  5 04:55:28 rpi4 [INFO] Capture-2/MainThread libcamera.setCcdExposure() #235: image command: libcamera-still --immediate --nopreview --encoding jpg --quality 95 --gain 0 --shutter -1000000 --metadata /tmp/tmp8t6hcdea.json --metadata-format json --awbgains 1,1 --output /tmp/tmpk6b5wd80.jpg
Sep  5 04:55:28 rpi4 [WARNING] Capture-2/MainThread capture.saferun() #407: *** FOCUS MODE ENABLED ***
```

#### Normal mode

### Version:
Latest:
```
karl@rpi4:/opt/indi-allsky $ git log
commit 13e93af011b16274da3c505fb11b7220ad6ef295 (HEAD -> master, origin/main)
Merge: 3b0d5302 375ca1e7
Author: Aaron W Morris <aaron@aarmor.net>
Date:   Sun Sep 3 08:13:39 2023 -0400

    Merge pull request #923 from aaronwmorris/dev

    Refactor image and timelapse viewers for better responsiveness
```

Vertikar, 2023-09-05:
This looks similar to this issue - https://github.com/aaronwmorris/indi-allsky/issues/927 but the config dumped from config.py looks ok
It's now crashed the GUI too, so I can't recover it easily

[image]

Here's the webapp logs:

```
Sep  5 11:23:48 rpi4 [INFO] MainProcess glogging.info() #264: Worker reloading: flask.json modified
Sep  5 11:23:49 rpi4 [INFO] MainProcess glogging.info() #264: Worker exiting (pid: 7678)
Sep  5 11:23:49 rpi4 gunicorn.indi_allsky.wsgi.error: [7859] Booting worker with pid: 7859
^B1Sep  5 12:07:47 rpi4 [WARNING] MainProcess forms.__init__() #2567: No images found in DB
Sep  5 12:08:18 rpi4 [INFO] MainProcess forms.__init__() #2587: Dates processed in 0.7206 s
Sep  5 12:08:26 rpi4 [ERROR] MainProcess app.log_exception() #1414: Exception on /indi-allsky/imageviewer [GET]#012Traceback (most recent call last):#012  File "/opt/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/flask/app.py", line 2190, in wsgi_app#012    response = self.full_dispatch_request()#012  File "/opt/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/flask/app.py", line 1486, in full_dispatch_request#012    rv = self.handle_user_exception(e)#012  File "/opt/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/flask/app.py", line 1484, in full_dispatch_request#012    rv = self.dispatch_request()#012  File "/opt/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/flask/app.py", line 1469, in dispatch_request#012    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)#012  File "/opt/indi-allsky/indi_allsky/flask/misc.py", line 22, in decorated_view#012    return app.ensure_sync(func)(*args, **kwargs)#012  File "/opt/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/flask/views.py", line 109, in view#012    return current_app.ensure_sync(self.dispatch_request)(**kwargs)#012  File "/opt/indi-allsky/indi_allsky/flask/base_views.py", line 189, in dispatch_request#012    context = self.get_context()#012  File "/opt/indi-allsky/indi_allsky/flask/views.py", line 1651, in get_context#012    context = super(ImageViewerView, self).get_context()#012  File "/opt/indi-allsky/indi_allsky/flask/base_views.py", line 195, in get_context#012    'indi_allsky_status' : self.get_indi_allsky_status(),#012  File "/opt/indi-allsky/indi_allsky/flask/base_views.py", line 264, in get_indi_allsky_status#012    obs.lon = math.radians(self.camera.longitude)#012TypeError: must be real number, not NoneType
Sep  5 12:08:28 rpi4 [ERROR] MainProcess app.log_exception() #1414: Exception on /indi-allsky/ [GET]#012Traceback (most recent call last):#012  File "/opt/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/flask/app.py", line 2190, in wsgi_app#012    response = self.full_dispatch_request()#012  File "/opt/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/flask/app.py", line 1486, in full_dispatch_request#012    rv = self.handle_user_exception(e)#012  File "/opt/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/flask/app.py", line 1484, in full_dispatch_request#012    rv = self.dispatch_request()#012  File "/opt/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/flask/app.py", line 1469, in dispatch_request#012    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)#012  File "/opt/indi-allsky/indi_allsky/flask/misc.py", line 22, in decorated_view#012    return app.ensure_sync(func)(*args, **kwargs)#012  File "/opt/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/flask/views.py", line 109, in view#012    return current_app.ensure_sync(self.dispatch_request)(**kwargs)#012  File "/opt/indi-allsky/indi_allsky/flask/base_views.py", line 189, in dispatch_request#012    context = self.get_context()#012  File "/opt/indi-allsky/indi_allsky/flask/views.py", line 102, in get_context#012    context = super(IndexView, self).get_context()#012  File "/opt/indi-allsky/indi_allsky/flask/base_views.py", line 195, in get_context#012    'indi_allsky_status' : self.get_indi_allsky_status(),#012  File "/opt/indi-allsky/indi_allsky/flask/base_views.py", line 264, in get_indi_allsky_status#012    obs.lon = math.radians(self.camera.longitude)#012TypeError: must be real number, not NoneType
Sep  5 12:08:31 rpi4 [ERROR] MainProcess app.log_exception() #1414: Exception on /indi-allsky/ [GET]#012Traceback (most recent call last):#012  File "/opt/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/flask/app.py", line 2190, in wsgi_app#012    response = self.full_dispatch_request()#012  File "/opt/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/flask/app.py", line 1486, in full_dispatch_request#012    rv = self.handle_user_exception(e)#012  File "/opt/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/flask/app.py", line 1484, in full_dispatch_request#012    rv = self.dispatch_request()#012  File "/opt/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/flask/app.py", line 1469, in dispatch_request#012    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)#012  File "/opt/indi-allsky/indi_allsky/flask/misc.py", line 22, in decorated_view#012    return app.ensure_sync(func)(*args, **kwargs)#012  File "/opt/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/flask/views.py", line 109, in view#012    return current_app.ensure_sync(self.dispatch_request)(**kwargs)#012  File "/opt/indi-allsky/indi_allsky/flask/base_views.py", [... cut ...]
Sep  5 12:29:52 rpi4 [ERROR] MainProcess/MainThread allsky._startImageWorker() #348: Image worker exception: sqlalchemy.exc.NoResultFound: No row was found when one was required
Sep  5 12:29:52 rpi4 [ERROR] MainProcess/MainThread allsky._startImageWorker() #348: Image worker exception:
Sep  5 12:29:52 rpi4 [INFO] MainProcess/MainThread allsky._startImageWorker() #355: Starting Image-135 worker
Sep  5 12:29:52 rpi4 [WARNING] MainProcess/MainThread image._load_detection_mask() #3570: No detection mask defined

```

Vertikar, 2023-09-05:
Actually, I think this might be my fault, I don't think I ran setup.sh after pulling the latest version.
Happy to close it off as this is an unusual issue that most people wouldn't run into

aaronwmorris, 2023-09-05:
I have seen this happen with a few other people.  It really should not be possible, but obviously it is.  I think it is just a temporary issue when certain new features are deployed.

Vertikar, 2023-09-05:
Definitely seems to be a temporary issue, it has sorted itself out now.

closed 2023-09-09 by Vertikar