aaronwmorris/indi-allsky #927: Setting minimum exposure to a low value causes min. exposure to be set to -1 and

Vertikar, 2023-09-05:
### Issue
Setting minimum exposure to a low value causes min. exposure to be set to -1 and scientific notation in the config.
### Steps to recreate
Set min exposure to something low like 0.00002, save config and reload.

Config will save correctly, and show correctly in config.

Sometime after first image is captured with new config, the exposure will show in the logs as -1, and will show scientific notation in the config screen

Vertikar, 2023-09-05:
I think this issue is actually partially fixed by https://github.com/aaronwmorris/indi-allsky/commit/f8e59de97764a6ad7e91515c4494118e115d439c

I've only been able to partially re-create this on my relatively clean ASI224 setup with an RPI4.

I tried to re-create it, and the capture thread happily ran, but I can still see the min exposure being set to 2e-05 in the config screen.

```
Sep  5 13:16:23 allsky [INFO] Capture-5/MainThread indi.configureDevice() #697: Setting switch CCD_FRAME_TYPE
Sep  5 13:16:23 allsky [INFO] Capture-5/MainThread indi.set_switch() #1265: Enabling Light (FRAME_LIGHT)
Sep  5 13:16:24 allsky [WARNING] Capture-5/MainThread capture._initialize() #670: Minimum exposure 0.00002000 too low, increasing to 0.00003201
Sep  5 13:16:24 allsky [INFO] Capture-5/MainThread capture._initialize() #678: Minimum CCD exposure: 0.00003201
Sep  5 13:16:24 allsky [INFO] Capture-5/MainThread capture._initialize() #699: Maximum CCD exposure: 28.00000000
Sep  5 13:16:24 allsky [INFO] Capture-5/MainThread capture._initialize() #743: Default CCD exposure: 0.01000000
Sep  5 13:16:24 allsky [INFO] Capture-5/MainThread capture.saferun() #207: Camera last ready: 0.0s
Sep  5 13:16:24 allsky [INFO] Capture-5/MainThread capture.saferun() #208: Exposure state: unset
Sep  5 13:16:24 allsky [INFO] Capture-5/MainThread capture.detectNight() #1264: Sun altitude: 43:03:46.8
```

Vertikar, 2023-09-05:
Was able to also re-create this with a min exposure above the minimum of the CCD

Set to 0.00004, after first image captured the config screen shows 4e-05.

The logs don't reflect the change though, is a full restart of indi-allsky or indiserver required to change the min exposure?

```
Sep  5 13:24:37 allsky [INFO] Capture-6/MainThread indi.configureDevice() #697: Setting switch CCD_FRAME_TYPE
Sep  5 13:24:37 allsky [INFO] Capture-6/MainThread indi.set_switch() #1265: Enabling Light (FRAME_LIGHT)
Sep  5 13:24:38 allsky [INFO] Capture-6/MainThread capture._initialize() #678: Minimum CCD exposure: 0.00003201
Sep  5 13:24:38 allsky [INFO] Capture-6/MainThread capture._initialize() #699: Maximum CCD exposure: 28.00000000
Sep  5 13:24:38 allsky [WARNING] Capture-6/MainThread capture._initialize() #724: Reusing last stable exposure: 0.009332
Sep  5 13:24:38 allsky [INFO] Capture-6/MainThread capture._initialize() #743: Default CCD exposure: 0.00933202
Sep  5 13:24:38 allsky [INFO] Capture-6/MainThread capture.saferun() #207: Camera last ready: 0.0s
Sep  5 13:24:38 allsky [INFO] Capture-6/MainThread capture.saferun() #208: Exposure state: unset
```

aaronwmorris, 2023-09-05:
The scientific notation is just a thing that python does once there are enough decimal points.  I will include an update to force the minimum and default exposure values to 6 places of precision.

As for the issue with -1, are you sure you are running the latest code.  I think I fixed a bug related to this a couple of weeks ago.

Vertikar, 2023-09-06:
> The scientific notation is just a thing that python does once there are enough decimal points. I will include an update to force the minimum and default exposure values to 6 places of precision.
Just saw that in the dev branch, thanks!
> 
> As for the issue with -1, are you sure you are running the latest code? I think I fixed a bug related to this a couple of weeks ago.
I am, but I saw this issue a while ago and didn't get around to reporting it. I can't currently recreate it.
let me see if I took any notes

aaronwmorris, 2023-09-06:
From what I recall, the -1 exposure issue cropped up when a reload was performed.

Vertikar, 2023-10-11:
Haven't seen this one again for a while either, so will close it off

closed 2023-10-11 by Vertikar