aaronwmorris/indi-allsky #928: When using a remote indiserver, config options appear to not be retrieved and set if they are changed by another client

Vertikar, 2023-09-05:
### Issue
If another INDI client say KStars/EKOS changes the image mode, or enables AWB, indi-allsky does not seem to be aware of this and continues taking images.
Unsure if this is expected behaviour or if its just an edge case you haven't run into yet.

### Steps to re-create
Setup a remote camera on another device with indiserver.
Connect indi-allsky to it.
Let some images be captured.

Connect EKOS to the same remote indiserver.
Change a setting, such as format, or auto WB
Capture an image from EKOS

Setting change will be reflected in indiserver logs.

Wait for indi-allsky to capture another image, and check indiserver logs, you will see the changed setting will stay the same and indi-allsky will not change it.

aaronwmorris, 2023-09-05:
This is basically working as designed.  indi-allsky assumes it is the only client making configuration changes on the indiserver.

Vertikar, 2023-09-05:
Ok, I had a feeling that was the case. I'll close this off now

closed 2023-09-05 by Vertikar