aaronwmorris/indi-allsky #931: Orb Display Mode not sufficiently documented

interplanetarychris, 2023-09-07:
I've tried to understand the Orb Display Mode options, and the (potentially) related Location->Attitude/Azimuth configuration options, but what is displayed still makes no sense to me.

Please provide clearer documentation for what is expected in this configuration variables, including Attitude/Azimuth and the Hour Angle, Azimuth and Attitude Orb Modes.

aaronwmorris, 2023-09-07:
I will create a wiki page for the Orbs.

The altitude and azimuth config options under Location have no effect currently.  I added it for future functionality.

interplanetarychris, 2023-09-07:
Thanks - I had assumed altitude and azimuth config options were for describing the orientation of your camera -- while most point vertically, the rotation could be arbitrary, and if the intent of the Orbs is to display the angle to the sun and moon, the user would want to be able to adjust for camera mounting orientation.

aaronwmorris, 2023-09-07:
The orbs are not relative to the orientation of the camera, but relative to the location of the camera on Earth.  For Local Hour Angle and Azimuth, top center is 0 degrees in both measurements.  Altitude just renders the relative altitude relative to the sides of the image which represent +90 to -90 degrees.

The ticks on the side represent the relative measurement of 0, 6, 12, and 18 degrees of altitude of the Sun for each mode.

aaronwmorris, 2024-01-19:
I plan on adding some screenshots eventually, but I started this wiki page to document the orbs.

https://github.com/aaronwmorris/indi-allsky/wiki/Orbs

StarGeezerPhil, 2024-01-26:
It took me some figuring out too - I thought that our azimuth in location would orientate the orbs as well. 

When I realised it was fixed I rotated the image instead (but this took a lot of tweaking):
[image]

If it'd be possible for the orbs to be plotted based on the camera orientation it'd really make tweaking this really easy.

afjkeese, 2024-04-21:
So looking at the image, it is the expected behavior to have the orbs for sun and moon always being opposite of where I actually see them on the image?
Tweaked for hours to get the sun dot be near where the sun is (whether visible or not) but failed ...

So is this the expected behavior?

aaronwmorris, 2024-04-21:
I think I finally have some time to work on this enhancement this week.

aaronwmorris, 2024-04-26:
I have taken some time to think about this and it is going to be much more difficult than I originally anticipated.

@afjkeese It is the expected behavior, but it it not necessarily intentional.  The difficulty I had not considered before is most allsky lenses flips the image upside down and backwards.  The default settings of indi-allsky re-flip the image again to hopefully orient it correctly, but the option remains to manually flip one or both axis.  Trying to account for the flipped image with the orbs makes this **very** complicated.

aaronwmorris, 2024-04-27:
I thought about the problem some more and I believe it was easier than I originally thought.  Flipping the image just results in reversing the orb motion.

With #1237 there are now options to offset the Azimuth of the orbs and reverse their motion.  The azimuth offset is separate from the main camera Azimuth because I think there is opportunity for a flipped image to require a different setting from the system.  That assumption might be wrong, time will tell.

afjkeese, 2024-04-27:
that approach makes clearly a lot of sense to just reverse the motion on a flipped image. Thanks for fixing it Aaron!

> I thought about the problem some more and I believe it was easier than I originally thought. Flipping the image just results in reversing the orb motion.
>
> With #1237 there are now options to offset [...]

closed 2024-08-10 by aaronwmorris