aaronwmorris/indi-allsky #934: How to display 'extra text' fields

sifank, 2023-09-08:
Is there something else I need to do besides just fill out the Extra Text Image field?  Before I used to do that and it showed up.  But now with the field formatting in the Label Template field, this doesn't seem to work.  Do I need to add something in the Label Template field or how to get the Extra Text file (plan ascii) displayed?

aaronwmorris, 2023-09-08:
With the new label location options, you just need to make sure there is enough space on the image to write the extra text.  The extra text is appended at the end of the normal label, so if that is at the bottom of the image, it might be written outside of the image.

Make sure to check the log to make sure there are no errors regarding reading the file.

sifank, 2023-09-08:
All of the images in the log file says" adding extra text file", no errors
or warnings.  I did find a couple of images last night that had the extra
text in them, but more than 98% do not.  So it fits.  I didn't modify the
formatting, so it should come out in the upper left under "Stacking", which
those that worked, did, and it fit ok.

Is there something else I could check?

Sifan
Carpe Nocturnum ...

aaronwmorris, 2023-09-09:
Is the extra text file being dynamically generated?  If so, is it possible that while it is being written, there is a lot of lag between opening the file for writing and data being flushed to disk?

sifank, 2023-09-10:
It was dynamically generated.  So I created a 4 line static file.  It is
still not working, although the log file says "#3428: Adding extra text
from /opt/KOBS/allskyscripts/allsky2.txt" contents are;
DpDep:        2.8 F
Temp:        60.3 F
Hum:         89.9 %
DP:          57.5 F

Permissions are 644

The bottom of the Label Template is:
# xy:15,15 (Upper Left)
# color:0,150,0
Exposure {exposure:0.6f}
# color:150,50,0
Gain {gain:d}
# color:50,50,150
Temp {temp:0.1f}{temp_unit:s}
# color:150,0,150
Stretch {stretch:s}
Stacking {stack_method:s}
# color:200,200,200 (default color)
# additional labels will be added here

Is there any special formatting required?

Sifan
Carpe Nocturnum ...

sifank, 2023-09-10:
This might be related:  every 3rd or 4th image has this dark band across
the top (see image attached).

This is also the same when using the static file mentioned in my previous
email.

Sifan
Carpe Nocturnum ...

aaronwmorris, 2023-09-10:
I found the problem.  I had an incorrect variable name that my linter did not catch.  Merged #941 to fix.

sifank, 2023-09-10:
Again, thanks for all you do and your responsiveness - much appreciated!

Sifan
Carpe Nocturnum ...

aaronwmorris, 2023-09-10:
I never saw the attached image with the dark band.  I do not think it was added to the issue via email.

sifank, 2023-09-10:
[image]

Sorry, image attached via github now.

aaronwmorris, 2023-09-10:
That is very strange.  I do not know what would cause that.  I would have guessed a bad dark frame, but it looks like you have not taken dark frames yet.

sifank, 2023-09-10:
The dark band has not come back yet since I upgraded to 941!

I have two units, I upgraded both, and no dark band on either so far.  It's
looking good!

Sifan
Carpe Nocturnum ...

sifank, 2023-09-12:
Well, now that dark band is back, same as before.  Should I write a new
issue on that?

Sifan
Carpe Nocturnum ...

aaronwmorris, 2023-09-12:
I doubt I would be able to do much about that.  What kind of camera are you using?  I will try to research it.

sifank, 2023-09-14:
I'm using the ASI120MC.  The dark band is occurring about every other frame
now.  It's been like this for the last couple of days.  When I first
downloaded and updated to 941, it seemed to not do it, until perhaps a day
or so later.  I have two RPI 4's setup as allskycams, both using the ASI120
and both did the same thing.

Sifan
Carpe Nocturnum ...

sifank, 2023-09-19:
A bit more info:  I played the 4 hr loop from this morning and the dark
band was there during the night, but at 6:09am this morning all the images
were clear.

Sifan
Carpe Nocturnum ...

aaronwmorris, 2023-10-31:
Is the dark band still a problem for you?

sifank, 2023-11-01:
Hi Aaron, thanks for asking!

I have two setups, both RPI 4s and both using the ASI120mc-s.  Both are at
Pull Release 941.  Checking the time lapse for last night, one still had
the dark band across the top, maybe 4 or 5 times over the entire night -
not very much.  The other one didn't have any ...  Now other times, they
both have had it.

I will update to PR 1010 (latest I see on github) and let 'em run and let
you know in two days (nights) what I find.

Sifan
Carpe Nocturnum ...

aaronwmorris, 2023-11-01:
These suggestions are going to be generic, but I can only think of two possible solutions.  If we already discussed these, I apologize.  :-)

- Replace your USB cable
- Replace your power supply

sifank, 2023-11-02:
Hi Aaron, I've updated both of my allskycams to the latest pr and ran them
all night collecting the timelapses.  So far no dark bars.  I'll continue
to run them for a few more nights and let you know.

Both these units run on POE, so plenty of power for the RPI 4's (4gb).
Both USB cables are short (<6"), but one cable is older than the other.  If
one of these shows dark bands, I'll note which cable and change it and
continue testing.

Thanks!!

Sifan
Carpe Nocturnum ...

sifank, 2023-11-03:
Ok, here's what I found - both allskycams are doing the same thing:
This is for last night 11/02/2023 at sunset:
Cam1:  17:59 1st dark band  (it's getting darker outside)
            18:34  last dark band
            18:35  very bright image as I believe it is switching over to
night mode
            18:36 cam adjusted, no more dark bands

Cam2:  17:51 1st dark bands appear (this is inside the house so darker
earlier)
             18:34 last dark band
             18:35 very bright image - cam adjusting
             18:36 cam adjusted, no more dark bands

This did not happen in the mornings, just at night and just before the
camera 'bloomed' and adjusted back.  The only time the entire 24hr period
dark bands occurred was just before switchover to night mode.

Hope this helps.

Sifan
Carpe Nocturnum ...

aaronwmorris, 2024-10-11:
Hello @sifank .  I realize this is an old issue, but there was another occurrence of the "dark band" issue and adjusting the USB Bandwidth setting appears to resolve it.  #1559 

The following is the config to implement the fix.

```json
{
    "PROPERTIES": {
      "CCD_CONTROLS": {
        "BandWidth": 100
      }
    },
    "SWITCHES": {
      "CCD_VIDEO_FORMAT": {
        "on": [
          "ASI_IMG_RAW16"
        ]
      }
    }
}
```

sifank, 2024-10-15:
Thanks!

Sifan
Carpe Nocturnum ...